espressif/arduino-esp32 #4588: Interface SD card module with HSPI

PrateekGoyal18, 2020-11-28:
I am trying to integrate an sd card module with the HSPI part of ESP32, since I am using VSPI pins for some other purpose and facing 2 issues.
**First**, when I have all the pins (MISO, MOSI, SCK, CS) connected I am not able to upload code to ESP32 and the following error is there: 
```
esptool.py v2.6
Serial port COM35
Connecting........_____....._
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 24:6f:28:aa:aa:d0
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0xffffff, SizeID=0xff), defaulting to 4MB
Compressed 8192 bytes to 47...

A fatal error occurred: Timed out waiting for packet content
A fatal error occurred: Timed out waiting for packet content
```

**Second**, when I upload the code to ESP32 and after that do the connections with the sd card module, then the serial monitor starts printing some garbage values which are not readable. 

PS: I am using the below code snippet for HSPI communication
```
SPIClass SDSPI(HSPI);
void setup(){
  SDSPI.begin(sck, miso, mosi, -1);
  SD.begin(ss, SDSPI);
}
```

lbernstone, 2020-11-28:
You have a strapping pin, most likely gpio12, held in an incorrect state during boot.  Run the espefuse command in the link you posted, or else show what pins you are using for HSPI.

PrateekGoyal18, 2020-11-29:
@lbernstone where and how to run ``espfuse`` command. Can you help?
I am using GPIO13 (MOSI), GPIO12 (MISO), GPIO14 (CLK) and GPIO15 (CS) pins for HSPI.

PrateekGoyal18, 2020-12-16:
@lbernstone some help please ?

lbernstone, 2020-12-16:
https://github.com/espressif/esptool
If you are not capable of following those instructions, you could use the code provided in the first issue you opened about this.
https://github.com/espressif/arduino-esp32/issues/3621#issuecomment-572198892

closed 2021-06-28 by stale[bot]