espressif/arduino-esp32 #4589: ESP32 doesn't work with MAX30102 (based on Sparkfun MAX3010x Library)

theotherjenkutler, 2020-11-28:
### Hardware:
Board:							Sparkfun ESP32 Thing Plus
Core Installation version:			1.0.5-rc2
IDE name:						Arduino IDE 1.8.13
Flash Frequency:					Both 40Mhz and 80Mhz
PSRAM enabled:                                     no
Upload Speed:				       115200
Computer OS:                                        Windows 10

### Description:
ESP32 never initializes an instance of this I2C device. The light on the device doesn't turn on and it doesn't appear to communicate. When tested with the exact same sketch with a Teensy LC or Sparkfun Pro Micro, it functions perfectly.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>
/*
  MAX30105 Breakout: Output all the raw Red/IR/Green readings
  By: Nathan Seidle @ SparkFun Electronics
  Date: October 2nd, 2016
  https://github.com/sparkfun/MAX30105_Breakout

  Outputs all Red/IR/Green values.

  Hardware Connections (Breakoutboard to Arduino):
  -5V = 5V (3.3V is allowed)
  -GND = GND
  -SDA = A4 (or SDA)
  -SCL = A5 (or SCL)
  -INT = Not connected

  The MAX30105 Breakout can handle 5V or 3.3V I2C logic. We recommend powering the board with 5V
  but it will also run at 3.3V.

  This code is released under the [MIT License](http://opensource.org/licenses/MIT).
*/

#include <Wire.h>
#include "MAX30105.h"

MAX30105 particleSensor;

#define debug Serial //Uncomment this line if you're using an Uno or ESP
//#define debug SerialUSB //Uncomment this line if you're using a SAMD21

void setup()
{
  debug.begin(115200);
  debug.println("MAX30105 Basic Readings Example");

  // Initialize sensor
  if (particleSensor.begin() == false)
  {
    debug.println("MAX30105 was not found. Please check wiring/power. ");
    //while (1);
  }

  particleSensor.setup(); //Configure sensor. Use 6.4mA for LED drive
}

void loop()
{
  debug.print(" R[");
  debug.print(particleSensor.getRed());
  debug.print("] IR[");
  debug.print(particleSensor.getIR());
  debug.print("] G[");
  debug.print(particleSensor.getGreen());
  debug.print("]");

  debug.println();
}
```

### Debug Messages:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:10900
load:0x40080400,len:6388
entry 0x400806b4
MAX30105 Basic Readings Example
MAX30105 was not found. Please check wiring/power. 

```

SuGlider, 2021-09-14:
@theotherjenkutler please check PR#5664 that should solve this issue.

SuGlider, 2021-10-02:
Solved by #5679

closed 2021-10-02 by SuGlider