espressif/arduino-esp32 #4594: Deadlock using BluetoothSerial if client is disconnected while high speed data transmission is underway

frankl83, 2020-11-30:
### Hardware:
> Board: Adafruit HUZZAH32- ESP32 Feather
> Core Installation version: current git master
> IDE name: Arduino IDE
> Flash Frequency: 80Mhz
> Upload Speed: 921600
>Computer OS: Windows 10 Pro 64bit 1809
> 
> ### Description:
>I'm using the BluetoothSerial library to send messages to a Bluetooth client (an Android smartphone with the Serial Bluetooth Terminal app). The transmission frequency is about 400Hz. 
I've noticed that if the client is disconnected while data transmission is underway, once the client reconnects to the ESP32, no more datas are received. Transmission is locked and a reset is necessary.
This is due to an already reported bug in the BluetoothSerial library caused by how congestions and the ACK timeout are handled in the original library:
>#3657 

>User:
@jcdevel
has proposed two fixes for this bug targeting both the congestion issue and the ACK timeout issue:
https://github.com/jcdevel/arduino-esp32/commit/9508b8bfffaa84c548c59fb10dc061ebd55cf3b4
and
https://github.com/jcdevel/arduino-esp32/commit/0f7de4de91fab00cfed19699b7eb24e4037380a2
I've implemented both the proposed fixes in the original BluetoothSerial library and based on my tests they are successful in getting rid of the problem.
Since the original issue was closed because it got stale and the pull request ignored, I'd like to open an issue about it again so that if the fixes are validated by others as well they might be integrated in the master release.

This is the example code used to replicate to bug:

**Sketch**
```c++
#include "BluetoothSerial.h"

#if !defined(CONFIG_BT_ENABLED) || !defined(CONFIG_BLUEDROID_ENABLED)
#error Bluetooth is not enabled! Please run `make menuconfig` to and enable it
#endif

BluetoothSerial SerialBT;

void setup() {
  Serial.begin(115200);
  SerialBT.begin("ESP32test"); //Bluetooth device name
  Serial.println("The device started, now you can pair it with bluetooth!");
}

void loop() 
{
  SerialBT.print("Hello!");
  SerialBT.print('\n');
  delay(2.5);
}
```

jcdevel, 2020-12-05:
Hello @frankl83. Yes, I had the PR opened for 9 months. There were nearly 200 PRs pending at that time, so I honestly thought there were no real interest in public contributions, so I gave up and closed mine. Too bad, because the following week most of the PRs were merged (Murphy's law).

Anyway, the patches are available in my fork for everybody.

smarq8, 2021-01-08:
My project also hang after disconnecting.
```
#include "BluetoothSerial.h"
BluetoothSerial BT;

#if !defined(CONFIG_BT_ENABLED) || !defined(CONFIG_BLUEDROID_ENABLED)
#error Bluetooth is not enabled! Please run `make menuconfig` to and enable it
#endif

void setup() {
  Serial.begin(115200);
  BT.begin("ESP32");
}

void loop(){
  BT.println(millis());
  Serial.println(millis());
}
```

closed 2021-07-08 by stale[bot]