espressif/arduino-esp32 #4595: ESP32 AP mode problem

naakhtar, 2020-12-01:
### Hardware:
Board:							FireBeetle-ESP32  (ARDUINO IDE BOARD SELECTION)
Core Installation version:			V1.0.4
IDE name:						ARDUINO IDE V1.8.13
Flash Frequency:					80Mhz
PSRAM enabled:                                    NOT SPECIFIED IN ARDUINO IDE
Upload Speed:					921600
Computer OS:                                        Windows 10

### Description:
Describe your problem here
  We are trying make WEBSERVER using ESP32, on which we are accessing our designed webpages.
   these webpages are written in HTML and CSS.
   - AP MODE used for the ESP32 WEBSERVER.
   - Webpages are not accessible in a web browser after successful connection with Access point of ESP32.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <WiFiClient.h>
#include <ESP32WebServer.h>
#include <WiFi.h>
#include <ESPmDNS.h>
#include "webpage.h"

String defaultAP = "apmode";
String defaultPASS = "11111111";

ESP32WebServer server(80);

void handleRoot() {
//  server.send(200, "text/plain", "hello from ESP32!");
server.send(200, "text/html", userlogin());
}

void handleNotFound(){
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET)?"GET":"POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i=0; i<server.args(); i++){
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
}

void setup(void){
  Serial.begin(9600);
  Serial.println(" ");
  Serial.println("Configuring access point...");
  Deafault_AP_MODE();

  delay(100);

  server.on("/", handleRoot);

  server.on("/inline", [](){
    server.send(200, "text/plain", "this works as well");
  });

  server.onNotFound(handleNotFound);

  server.begin();
  Serial.println("HTTP server started");
}

void loop(void){
  delay(100);
  server.handleClient();
}

void Deafault_AP_MODE()
{
  WiFi.mode(WIFI_AP);delay(500);    
//  WiFi.softAPConfig( IPAddress(defIP0, defIP1, defIP2, defIP3),  IPAddress(192, 168, 2, 2), IPAddress(255, 255, 255, 0) );delay(500); 
  WiFi.softAP( defaultAP.c_str(), defaultPASS.c_str());delay(500); 
  WiFi.persistent(false);
//  defAPFlg=1;
//  Wmode = 0; 
  Serial.print("Default AP Mode Started at ");  Serial.println(WiFi.softAPIP());
}

String userlogin()
{

  String ptr = HTML_headernew("Login", "KVARTech", 20); //Read HTML contents
  ptr += login_setting_array;

  return ptr;
}

String HTML_headernew(String title, String companyname, int page_no)
{
  String ptr = "<!DOCTYPE html>";
  ptr += "<html>";
  ptr += "<head>";
   ptr += "<title>" + title + "</title>";
  ptr += cssforpage;
  ptr += "</head>";
  ptr +="<body>";
  ptr +="<div class=\"header\">";

//  ptr += KVARlogo;
  ptr +="<h1>" + companyname + "</h1>";
  ptr +="<h2>" + title + "</h2>";

  

  ptr +="</div>";

if(page_no > 10){
    return ptr;
}else{
    ptr +="<ul>";
//    if(CID_login.access_level == 0){
  if(page_no == 1) ptr +="<li style=\"float:left\"><a class=\"active\" href=\"/Home\">Home</a></li>"; else ptr +="<li style=\"float:left\"><a href=\"/Home\">Home</a></li>";
  if(page_no == 2) ptr +="<li style=\"float:left\"><a class=\"active\"href=\"/settings\">Log settings</a></li>"; else ptr +="<li style=\"float:left\"><a href=\"/settings\">Log settings</a></li>";
  if(page_no == 3) ptr +="<li style=\"float:left\"><a class=\"active\"href=\"/wifi_settings\">WiFi settings</a></li>"; else ptr +="<li style=\"float:left\"><a href=\"/wifi_settings\">WiFi settings</a></li>";
  if(page_no == 4) ptr +="<li style=\"float:left\"><a class=\"active\"href=\"/download_data\">Download data</a></li>"; else ptr +="<li style=\"float:left\"><a href=\"/download_data\">Download data</a></li>";
  if(page_no == 5) ptr +="<li style=\"float:left\"><a class=\"active\"href=\"/mail_settings\">Mail Settings</a></li>"; else ptr +="<li style=\"float:left\"><a href=\"/mail_settings\">Mail Settings</a></li>";  
//    }
    
  ptr +="<li><a href=\"/logout\">Logout</a></li>";
  ptr +="</ul>";
}
  return ptr;
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here
14:27:22.844 -> dhcps: send_offer>>udp_sendto result 0 
```

lbernstone, 2020-12-01:
Use board "ESP32 Dev Module". Set Core Debug Level to verbose.  This will give you more logging information.
protip: If you want help with your code, it should directly compile.

mkfrey, 2020-12-01:
I think this is a duplicate of #2025, which has not been fixed.

@lbernstone When this is the same issue as #2025, which I think is highly probable, then setting the logging to verbose will actually prevent the issue from appearing.

@naakhtar Wokaroud for this issue:
- Call `WiFi.persistent(false)` before doing anything WiFi related in your code
- Wipe the flash of the devices that have been running code with persistence enabled

solartech77, 2020-12-02:
Hello world

naakhtar, 2020-12-03:
[image]
[image]

lbernstone, 2020-12-03:
So, your issue is only with your Windows 10 PC connecting to the device?  Your android phone is able to browse the web page?  Disconnect your pc from any other networks (even wired), and try again.

naakhtar, 2020-12-04:
As advised tried by disconnecting pc from any other network but still getting same result.

EricMc1289, 2021-01-01:
From your screenshot your are still connected to a wired network. The most os‘s route traffic to the network port that is online even if you use a ip as adress. 
F.e. on ios i got problems getting the website of the esp because the iphone tries to connect to 192.168.4.1 over the mobile data. Try to give your pc only the esp wifi and no other network  connection

closed 2021-07-08 by stale[bot]