espressif/arduino-esp32 #4597: [E][ssl_client.cpp:33] _handle_error(): [start_ssl_client():199]: (-30208) SSL - The own private key or pre-shared key is not set, but needed

radimkohout, 2020-12-01:
### Hardware:
Board:						       ESP32 Dev Module
Core Installation version:		       1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git? ?1.0.2? ?1.0.3?
IDE name:					       Arduino IDE
Flash Frequency:				       80Mhz
PSRAM enabled:                                     no
Upload Speed:				       115200
Computer OS:                                         Windows 10

### Description:
I'm having folowing error:
[E][ssl_client.cpp:33] _handle_error(): [start_ssl_client():199]: (-30208) SSL - The own private key or pre-shared key is not set, but needed
The certificate is self-signed, generated by OpenSSL. What's wrong? I want to communicate through TCP with SSL.

### Sketch:  
```cpp
#include <WiFiClientSecure.h>

const char* ssid     = "mySSID";     // your network SSID (name of wifi network)
const char* password = "MyPassword"; // your network password

const char*  server = "xx.xxx.xxx.xxx";  // Server URL

const char* test_root_ca = "-----BEGIN CERTIFICATE-----\n"\
"mycertificate\n"\
"separatedby\n"\
"lines\n"\
"-----END CERTIFICATE-----\n";

WiFiClientSecure client;

void setup() {
  //Initialize serial and wait for port to open:
  Serial.begin(115200);
  delay(100);

  Serial.print("Attempting to connect to SSID: ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);

  // attempt to connect to Wifi network:
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    // wait 1 second for re-trying
    delay(1000);
  }

  Serial.print("Connected to ");
  Serial.println(ssid);

  client.setCACert(test_root_ca);
  Serial.println("\nStarting connection to server...");
  if (!client.connect(server, 1234))
    Serial.println("Connection failed!");
  else {
    Serial.println("Connected to server!");
    // Make a HTTP request:
    client.println("Hello, world");
    client.println();
...Supressed
```

### Debug Messages:
```
...
23:05:27.734 -> Starting connection to server...
23:05:27.734 -> [V][ssl_client.cpp:56] start_ssl_client(): Free internal heap before TLS 265324
23:05:27.734 -> [V][ssl_client.cpp:58] start_ssl_client(): Starting socket
23:05:27.968 -> [V][ssl_client.cpp:93] start_ssl_client(): Seeding the random number generator
23:05:27.968 -> [V][ssl_client.cpp:102] start_ssl_client(): Setting up the SSL/TLS structure...
23:05:27.968 -> [V][ssl_client.cpp:115] start_ssl_client(): Loading CA cert
23:05:27.968 -> [V][ssl_client.cpp:180] start_ssl_client(): Setting hostname for TLS session...
23:05:27.968 -> [V][ssl_client.cpp:195] start_ssl_client(): Performing the SSL/TLS handshake...
23:05:31.124 -> [E][ssl_client.cpp:33] _handle_error(): [start_ssl_client():199]: (-30208) SSL - The own private key or pre-shared key is not set, but needed
23:05:31.124 -> [E][WiFiClientSecure.cpp:132] connect(): start_ssl_client: -30208
23:05:31.124 -> [V][ssl_client.cpp:248] stop_ssl_socket(): Cleaning SSL connection.
23:05:31.170 -> Connection failed!
```

mkfrey, 2020-12-02:
Does your server require authentication via TLS client certificate or PSK?

radimkohout, 2020-12-02:
I am using this server https://github.com/GuiTeK/Qt-SslServer/tree/master/example/Server/Server. I don't know how to answer your question.

radimkohout, 2020-12-09:
This error was caused by the server, that required PSK, but I'm having another problem:
`
#include <WiFiClientSecure.h>

const char* ssid     = "SSID";     // your network SSID (name of wifi network)
const char* password = "Password"; // your network password

const char*  server = "xxx.xxx.xxx.xxx";  // Server URL

const char* test_client_cert = "-----BEGIN CERTIFICATE-----\n"\
//content of cert
"-----END CERTIFICATE-----\n";

WiFiClientSecure client;

void setup() {
  //Initialize serial and wait for port to open:
  Serial.begin(115200);
  delay(100);

  Serial.print("Attempting to connect to SSID: ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);

  // attempt to connect to Wifi network:
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    // wait 1 second for re-trying
    delay(1000);
  }

  Serial.print("Connected to ");
  Serial.println(ssid);

  //client.setCACert(test_root_ca);
  client.setCertificate(test_client_cert); // for client verification
  //client.setPrivateKey(test_client_cert);	// for client verification

  Serial.println("\nStarting connection to server...");
  if (!client.connect(server, 1234))
    Serial.println("Connection failed!");
  else {
    Serial.println("Connected to server!");
    // Make a HTTP request:
/*    client.println("Hello, world");
    client.println();

    // if there are incoming bytes available
    // from the server, read them and print them:
    while (client.available()) {
      char c = client.read();
      Serial.write(c);
    }

   client.stop();*/
  }
}

void loop() {
  // do nothing
}
`
This one drops this error:
`Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.`
Whole log:
`
Attempting to connect to SSID: SSID
21:44:39.329 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
21:44:39.329 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
21:44:39.329 -> .[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 4 - STA_CONNECTED
21:44:39.560 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
21:44:39.560 -> [D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.68.111, MASK: 255.255.255.0, GW: 192.168.68.1
21:44:40.353 -> Connected to SSID
21:44:40.353 -> 
21:44:40.353 -> Starting connection to server...
21:44:40.353 -> [V][ssl_client.cpp:56] start_ssl_client(): Free internal heap before TLS 265332
21:44:40.353 -> [V][ssl_client.cpp:58] start_ssl_client(): Starting socket
21:44:40.494 -> [V][ssl_client.cpp:93] start_ssl_client(): Seeding the random number generator
21:44:40.494 -> [V][ssl_client.cpp:102] start_ssl_client(): Setting up the SSL/TLS structure...
21:44:40.494 -> [I][ssl_client.cpp:156] start_ssl_client(): WARNING: Use certificates for a more secure communication!
21:44:40.494 -> [V][ssl_client.cpp:180] start_ssl_client(): Setting hostname for TLS session...
21:44:40.541 -> [V][ssl_client.cpp:195] start_ssl_client(): Performing the SSL/TLS handshake...
21:44:42.779 -> [V][ssl_client.cpp:216] start_ssl_client(): Verifying peer X.509 certificate...
21:44:42.779 -> [V][ssl_client.cpp:225] start_ssl_client(): Certificate verified.
21:44:42.826 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
21:44:42.826 -> Core 1 register dump:
21:44:42.826 -> PC      : 0x400eb0bc  PS      : 0x00060a30  A0      : 0x800e2889  A1      : 0x3ffb1c80  
21:44:42.826 -> A2      : 0x3ffbaab0  A3      : 0x40084f34  A4      : 0x3ffb8214  A5      : 0x00000000  
21:44:42.826 -> A6      : 0x00000000  A7      : 0xffffffbc  A8      : 0xfefefefe  A9      : 0x3ffb1c20  
21:44:42.826 -> A10     : 0x3ffb80c0  A11     : 0x3ffd63f4  A12     : 0x00000000  A13     : 0x00000001  
21:44:42.826 -> A14     : 0x00060c20  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
21:44:42.873 -> EXCVADDR: 0xfefeff26  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffb  
21:44:42.873 -> 
21:44:42.873 -> Backtrace: 0x400eb0bc:0x3ffb1c80 0x400e2886:0x3ffb1ca0 0x400d23af:0x3ffb1cc0 0x400d1b15:0x3ffb1f30 0x400d1c61:0x3ffb1f60 0x400d18ed:0x3ffb1f80 0x400d4dd7:0x3ffb1fb0 0x40088b9d:0x3ffb1fd0
21:44:42.873 -> 
21:44:42.873 -> Rebooting...
21:44:42.873 -> ets Jun  8 2016 00:22:57

`
The server is now run by OpenSSL using command:
`
openssl s_server -port 1234 -cert /etc/ssl/private/cert.pem -key /etc/ssl/private/key.pem 
`
The keys are 4096-bit RSA.

mkfrey, 2020-12-10:
You can decode the exception using the [ExceptionDecoder](https://github.com/me-no-dev/EspExceptionDecoder).

The issue might be caused by you setting `client.setCertificate()` and not providing the corresponding private key. `client.setCertificate()` is used to set the the certificate the client uses to authenticate itself to the server, which also requires the private key of this client certificate to be present.

You most likely want to just verify the server certificate, in which case you need to use `client.setCACert()` and use the content of `cert.pem` from the server, if you're using a self signed cert.

closed 2021-06-28 by stale[bot]