espressif/arduino-esp32 #4599: esptool obtained by python get.py on Mac OS X Big Sur fails

clambertus, 2020-12-03:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.5-rc3
IDE name:						Arduino IDE
Flash Frequency:					?40Mhz?
PSRAM enabled:                                    ?no?
Upload Speed:					?115200?
Computer OS:                                        MacOS X Big Sur

### Description:
Both the development and stable Arduino board libraries as documented in https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/boards_manager.md ship with an esptool binary for MacOS X Big Sur with a missing symbol:

./esptool
Traceback (most recent call last):
  File "esptool.py", line 58, in <module>
  File "/Library/Python/2.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 389, in load_module
  File "serial/tools/list_ports.py", line 29, in <module>
  File "/Library/Python/2.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 389, in load_module
  File "serial/tools/list_ports_posix.py", line 31, in <module>
  File "/Library/Python/2.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 389, in load_module
  File "serial/tools/list_ports_osx.py", line 32, in <module>
ValueError: dlsym(RTLD_DEFAULT, kIOMasterPortDefault): symbol not found
Failed to execute script esptool

This appears to be due to this compiled python binary from https://dl.espressif.com/dl/esptool-3.0.0.2-macos.tar.gz using a version of pyserial's list_ports_osx.py which has not had this issue fixed. The code fix in pyserial is:

# kIOMasterPortDefault is no longer exported in BigSur but no biggie, using NULL works just the same
# kIOMasterPortDefault = 0 # WAS: ctypes.c_void_p.in_dll(iokit, "kIOMasterPortDefault")
kCFAllocatorDefault = ctypes.c_void_p.in_dll(cf, "kCFAllocatorDefault")

With this fix, esptool.py works fine, but the Arduino IDE forces the use of the tools/esptool/esptool binary and returns an exec/fork error if esptool.py is specified in platform.txt instead.

lbernstone, 2020-12-03:
Is this a duplicate of #4408 ?

me-no-dev, 2020-12-03:
yup :)

closed 2020-12-03 by me-no-dev