espressif/arduino-esp32 #4601: Multiple TwoWire transmission problems with two MCP3428 ADC Chips

jman-31, 2020-12-03:
### Hardware:
Board:							?ESP32 Dev Module?
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git? ?1.0.2? ?1.0.3?
IDE name:						?Arduino?
Flash Frequency:					?80Mhz?
PSRAM enabled:                                    ?no?
Upload Speed:					?115200?
Computer OS:                                        ?Windows 10? 

### Description:
I am trying to use two MCP3428 ADC's for eight analog inputs. In doing so I am trying to use the Wire library with two TwoWire instances. I can get the ADC's to each work correctly on its own as long as I use  "TwoWire I2Cone = TwoWire(0);" or "TwoWire I2Ctwo = TwoWire(0); but as soon as I add the line "TwoWire I2Cone = TwoWire(1);" I lose contact with the second ADC. I get replicating signals on both function calls from ADC1. I very well could be that I am not understanding how the library works and that my code is completely wrong!

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>
#include <MCP3428.h>
#include <Wire.h>

MCP3428 MCP1(0); // Declaration of MCP3428: A2, A1, A0 bits (000, 0x68)
MCP3428 MCP2(0); // Declaration of MCP3428: A2, A1, A0 bits (000, 0x68)
TwoWire I2Cone = TwoWire(0);
TwoWire I2Ctwo = TwoWire(1);

void setup() {
  Serial.begin(9600);
  I2Cone.begin(19, 32, 100000); // SDA pin 19, SCL pin 32, 100kHz frequency
  I2Ctwo.begin(23, 22, 100000); // SDA pin 23, SCL pin 22, 100kHz frequency
  delay(500);
}

void loop() {
  ADC_I2C0ne();
  Serial.println("        ");
  ADC_I2Ctwo();
  Serial.println("        ");
  delay(500);
}

void ADC_I2C0ne() {
  long Raw_ADC1[4];
  Serial.println("ADC_1");
  I2Cone.beginTransmission(0x68);
  for (int i = 1; i <= 4; i++) {
    // MCP3428 is configured to channel i with 16 bits resolution, continuous mode and gain defined to 1
    // This arrangement of the mentioned paarmeters can be changed as per convenience
    MCP1.SetConfiguration(i, 16, 1, 1);
    // Note that the library waits for a complete conversion
    Raw_ADC1[i - 1] = MCP1.readADC();
    // print results
    Serial.print("Digital value of Analog Input at Channel ");
    Serial.print(i);
    Serial.print(" at ADC_1 : ");
    Serial.println(Raw_ADC1[i - 1]);
    delay(1000);
  }
  I2Cone.endTransmission();
}

void ADC_I2Ctwo() {
  long Raw_ADC2[4];
  Serial.println("ADC_2");
  I2Ctwo.beginTransmission(0x68);
  for (int j = 1; j <= 4; j++) {
    // MCP3428 is configured to channel j with 16 bits resolution, continuous mode and gain defined to 1
    // This arrangement of the mentioned paarmeters can be changed as per convenience
    MCP2.SetConfiguration(j, 16, 1, 1);
    // Note that the library waits for a complete conversion
    Raw_ADC2[j - 1] = MCP2.readADC();
    // print results
    Serial.print("Digital value of Analog Input at Channel ");
    Serial.print(j);
    Serial.print(" at ADC_2 : ");
    Serial.println(Raw_ADC2[j - 1]);
    delay(1000);
  }
  I2Ctwo.endTransmission();
}

```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

lbernstone, 2020-12-03:
https://github.com/espressif/arduino-esp32/blob/master/libraries/Wire/doc/i2c_debugging.md
I2C is not thread safe, and running two devices asynchronously can result in them interrupting each other.  The HAL can only do so much to contain the behavior of libraries that were designed around single threaded bit banging.

jman-31, 2020-12-03:
OK. I think I'll simplify my life and change over to an ADS7828 with eight inputs. Thanks for the feedback!

coratron, 2020-12-04:
@jman-31  I have not used two I2C buses simultaneously on the ESP32 so I can't comment on that, but I am pretty sure your code would work if you used the same bus.

closed 2021-02-16 by stale[bot]