espressif/arduino-esp32 #4602: Opensuse Arduino IDE ESP32

GastonMelo, 2020-12-03:
Im trying to use the esp32 IA thinker with arduino IDE
Im using linux OpenSuse Tumbleweed
Im gettig this error: 
ImportError: No module named serial
Im using the CameraWebServer example on Arduino 1.8.13
The error displayed is:
Show the EXACT error message (it doesn't work is not enough)  
`Arduino:1.8.13 (Linux), Tarjeta:"ESP32 Wrover Module, Huge APP (3MB No OTA/1MB SPIFFS), QIO, 80MHz, 921600, None"

Traceback (most recent call last):
  File "/home/gaston/.arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py", line 37, in <module>
    import serial
ImportError: No module named serial
Se encontraron varias bibliotecas para "WiFi.h"
Usado: /home/gaston/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi
 No usado: /home/gaston/arduino-1.8.13/libraries/WiFi
exit status 1
Error compilando para la tarjeta ESP32 Wrover Module.

Este informe podría contener más información con
"Mostrar salida detallada durante la compilación"
opción habilitada en Archivo -> Preferencias.`

```
Thanks
Gastón

lbernstone, 2020-12-03:
`pip install pyserial`

GastonMelo, 2020-12-03:
Thasnks again for your quick reply. What I tried so far is:
1) pip3 install pyserial
2) pip install pyserial
3) https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/opensuse.md
4) check permission
5) check groups in linux and Im in Dialout group
6) symbolic link python with python3
7) include alias python=python3 in the .bashrc

Nothng works
Im still unable to compile the CameraWebServer on ESP32_CAM

Thanks
Gaston

lbernstone, 2020-12-03:
This is a python problem, not really arduino.  Try running esptool directly- `python ~/.arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py`.  run `python --version` to be sure what version your system is using as default.

GastonMelo, 2020-12-03:
Thanks again for your support and quick reply 
the first command displayed :
<File "/home/gaston/.arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py", line 45, in <module>
    raise ImportError("""
ImportError: 
esptool.py depends on pyserial, but there is a conflict with a currently installed package named 'serial'.

You may be able to work around this by 'pip uninstall serial; pip install pyserial' but this may break other installed Python software that depends on 'serial'.

There is no good fix for this right now, apart from configuring virtualenvs. See https://github.com/espressif/esptool/issues/269#issuecomment-385298196 for discussion of the underlying issue(s).>

the second command displayed:

<Python 3.8.6>

I already tried to uninstall  by using "pip uninstall serial" but it didnt work. 
I can use the esp8266 with arduino without problems but with the new esp32_cam I have this error about "import serial"
How to solve, I already tried a few things (search and read a few forums with this error and a few was solved with the pip install pyserial, someone point to python2 but is deprecated now)
Thanks
Gastón

lbernstone, 2020-12-03:
> I already tried to uninstall by using "pip uninstall serial" but it didnt work.

Please post errors.
python2 is indeed dead. What does pip think the version is? `pip --version`

GastonMelo, 2020-12-03:
pip --version 
`pip 20.2.3 from /usr/lib/python3.8/site-packages/pip (python 3.8)`

GastonMelo, 2020-12-03:
python ~/.arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py

`Traceback (most recent call last):
  File "/home/gaston/.arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool.py", line 45, in <module>
    raise ImportError("""
ImportError: 
esptool.py depends on pyserial, but there is a conflict with a currently installed package named 'serial'.

You may be able to work around this by 'pip uninstall serial; pip install pyserial' but this may break other installed Python software that depends on 'serial'.

There is no good fix for this right now, apart from configuring virtualenvs. See https://github.com/espressif/esptool/issues/269#issuecomment-385298196 for discussion of the underlying issue(s).`

GastonMelo, 2020-12-03:
python --version
`Python 3.8.6`

GastonMelo, 2020-12-03:
this solve my problems
https://www.programmersought.com/article/37265263439/
thanks lbernstone  for your time 
Regards
Gaston

closed 2021-02-16 by stale[bot]