espressif/arduino-esp32 #4603: ESP32S2 - Serial.Flash() is not waiting for transmission to end

DawidDiaco, 2020-12-03:
### Hardware:
Board:							ESP32-S2 Wrover
Core Installation version:			unofficial - branch - esp32s2, commit baaff8480ebbdcf5c8c9214283f538a3fcdba29d and release/4.2-beta1 of ESP-IDF
IDE name:						Visual Studio with VisualGDB
Flash Frequency:					80Mhz
PSRAM enabled:                                    I don't have a clue how to check it
Upload Speed:					115200
Computer OS:                                        Windows 10 Pro

### Description:
I'm using Serial1 for communicating with modbus device. For that I found a great library that relies on Serial.Flash() for ensuring that transmission has ended. What we found out in this issue: https://github.com/emelianov/modbus-esp8266/issues/85, Arduino liblaries are bugged, and Serial.Flush() is not waiting for the end of transmision.

### Sketch:
```cpp
#include <Arduino.h>

void setup() 
{
	Serial1.begin(9600, SERIAL_8E1, GPIO_RXD, GPIO_TXD);
	pinMode(GPIO_RTS, OUTPUT);
}

void loop() 
{
	digitalWrite(GPIO_RTS, HIGH);
	uint8_t data[] = { 0x0, 0x1, 0x2, 0x3, 0x4 };
	Serial1.write(data, 4);
	Serial1.flush();
	digitalWrite(GPIO_RTS, LOW);
	
	delay(1000);
}
```

### Debug Messages:
[image]

lbernstone, 2020-12-03:
Please update to idf-release/v4.2 and retest.  This is beta code and is being actively worked on.

DawidDiaco, 2020-12-05:
I've checkout branch idf-release/v4.2, but I'm unable to compile it with release/4.2 of esp-idf.  I'm getting multiple errors that look like this:

```
FAILED: esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/esp32-hal-time.c.obj 
C:\SysGCC\esp32\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32s2-elf\bin\xtensa-esp32s2-elf-gcc.exe -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -Iconfig -I../../../components/arduino/variants/esp32 -I../../../components/arduino/cores/esp32 -I../../../components/arduino/libraries/ArduinoOTA/src -I../../../components/arduino/libraries/AsyncUDP/src -I../../../components/arduino/libraries/BLE/src -I../../../components/arduino/libraries/BluetoothSerial/src -I../../../components/arduino/libraries/DNSServer/src -I../../../components/arduino/libraries/EEPROM/src -I../../../components/arduino/libraries/ESP32/src -I../../../components/arduino/libraries/ESPmDNS/src -I../../../components/arduino/libraries/FFat/src -I../../../components/arduino/libraries/FS/src -I../../../components/arduino/libraries/HTTPClient/src -I../../../components/arduino/libraries/HTTPUpdate/src -I../../../components/arduino/libraries/LITTLEFS/src -I../../../components/arduino/libraries/NetBIOS/src -I../../../components/arduino/libraries/Preferences/src -I../../../components/arduino/libraries/SD_MMC/src -I../../../components/arduino/libraries/SD/src -I../../../components/arduino/libraries/SimpleBLE/src -I../../../components/arduino/libraries/SPIFFS/src -I../../../components/arduino/libraries/SPI/src -I../../../components/arduino/libraries/Ticker/src -I../../../components/arduino/libraries/Update/src -I../../../components/arduino/libraries/WebServer/src -I../../../components/arduino/libraries/WiFiClientSecure/src -I../../../components/arduino/libraries/WiFi/src -I../../../components/arduino/libraries/WiFiProv/src -I../../../components/arduino/libraries/Wire/src -I../../../components/arduino/cores/esp32/libb64 -IC:/SysGCC/esp32/esp-idf/v4.2/components/newlib/platform_include -IC:/SysGCC/esp32/esp-idf/v4.2/components/freertos/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/freertos/xtensa/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/heap/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/log/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/lwip/include/apps -IC:/SysGCC/esp32/esp-idf/v4.2/components/lwip/include/apps/sntp -IC:/SysGCC/esp32/esp-idf/v4.2/components/lwip/lwip/src/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/lwip/port/esp32/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/lwip/port/esp32/include/arch -IC:/SysGCC/esp32/esp-idf/v4.2/components/soc/src/esp32s2/. -IC:/SysGCC/esp32/esp-idf/v4.2/components/soc/src/esp32s2/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/soc/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/esp_rom/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/esp_common/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/esp_system/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/xtensa/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/xtensa/esp32s2/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/esp32s2/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/driver/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/driver/esp32s2/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/esp_ringbuf/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/efuse/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/efuse/esp32s2/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/espcoredump/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/esp_timer/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/esp_ipc/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/soc/soc/esp32s2/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/soc/soc/esp32s2/../include -IC:/SysGCC/esp32/esp-idf/v4.2/components/soc/soc/esp32s2/private_include -IC:/SysGCC/esp32/esp-idf/v4.2/components/vfs/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/esp_wifi/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/esp_wifi/esp32s2/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/esp_event/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/esp_netif/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/esp_eth/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/tcpip_adapter/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/app_trace/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/spi_flash/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/mbedtls/port/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/mbedtls/mbedtls/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/mbedtls/esp_crt_bundle/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/mdns/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/console -IC:/SysGCC/esp32/esp-idf/v4.2/components/esp_adc_cal/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/fatfs/diskio -IC:/SysGCC/esp32/esp-idf/v4.2/components/fatfs/vfs -IC:/SysGCC/esp32/esp-idf/v4.2/components/fatfs/src -IC:/SysGCC/esp32/esp-idf/v4.2/components/wear_levelling/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/sdmmc/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/nvs_flash/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/app_update/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/bootloader_support/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/spiffs/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/openssl/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/wpa_supplicant/include -IC:/SysGCC/esp32/esp-idf/v4.2/components/wpa_supplicant/port/include -IC:/SysGC [... cut ...]
In file included from ../../../components/arduino/cores/esp32/esp32-hal.h:74,
                 from ../../../components/arduino/cores/esp32/esp32-hal-time.c:15:
../../../components/arduino/cores/esp32/esp32-hal-gpio.h:78:44: error: 'SOC_GPIO_PIN_COUNT' undeclared here (not in a function); did you mean 'GPIO_PIN_COUNT'?
 extern const esp32_gpioMux_t esp32_gpioMux[SOC_GPIO_PIN_COUNT];
                                            ^~~~~~~~~~~~~~~~~~
                                            GPIO_PIN_COUNT
```

Do I need a specific version of esp-idf in order to compile it?

bertmelis, 2020-12-05:
You might want to check out past PRs:
#2029
#4263

cqh963852, 2020-12-13:
https://www.esp32.com/viewtopic.php?p=68894#p68894
Same problem here.

> I've checkout branch idf-release/v4.2, but I'm unable to compile it with release/4.2 of esp-idf. I'm getting multiple errors that look like this:
>
> ```
> FAILED: [...]

closed 2021-06-28 by stale[bot]