espressif/arduino-esp32 #4606: Wifi multi doesn't respond after SYSTEM_EVENT_STA_LOST_IP event

OmarAlkassab, 2020-12-04:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.3
IDE name:						Platform.io
Flash Frequency:					40Mhz
PSRAM enabled:                                     no
Upload Speed:					921600
Computer OS:                                        Windows 10

### Description:
My WiFi router some times loose power (due to several reasons). But after several days (sometimes after some hours) the wifi connection lost, and the WiFiEvent SYSTEM_EVENT_STA_LOST_IP called. then the ESP32 cannot reconnect to the WiFi router again. Note that the other tasks still running correctly but the WiFi status still WL_IDLE_STATUS.
Once I make power reset the code run correctly, then after some time the issue re-happen again.

I putted only the necessary code lines.

```cpp

//Change the code below by your sketch
#include <Arduino.h>
//Necessary includes

void setup() {
wifiMulti.add("SSID","PASS");
}

void loop() {
uint8_t wifi_status = wifiMulti.run();
  if (wifi_status != WL_CONNECTED)
  {
    delay(1000);
    Debug.print(wifi_status);
    Debug.print("+");
    return;
  }
}
```

### Debug Messages:
```
0+0+0+0+0+
```

closed 2021-02-18 by stale[bot]

LeoDJ, 2022-12-20:
seems related to #2269

puterboy, 2024-05-21:
Was this ever fixed? 
I seem to have the same or similar problem.
My esp32 used to run for hundreds of days at a time.
For the past few months, after ~3 days of uptime, it suddenly loses WiFi connection and fails to reconnect even though the core program keeps running. Not sure what is causing this.