espressif/arduino-esp32 #4608: BT stack dies when BLE is scanning and BT classic is writing

aovestdipaperino, 2020-12-06:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Computer OS:                                        Mac OSX

### Description:
Describe your problem here

I am trying to use BT classic as an extra debug channel.
If there are no clients connected to the BT Serial, it works. As soon as a client connects and the Debug function outputs to BT while BLE is scanning, the stack freezes

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

/*
   Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleScan.cpp
   Ported to Arduino ESP32 by Evandro Copercini
*/

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEScan.h>
#include <BLEAdvertisedDevice.h>
#include <BluetoothSerial.h>
#define LOG_LOCAL_LEVEL ESP_LOG_VERBOSE
#include "esp_log.h"
BluetoothSerial SerialBT;

class Debug : public Print {
  public:
    Debug() {};
    virtual size_t write(uint8_t c) {
        //Serial.write(c);
        SerialBT.write(c);
        return 1;
    }
} _debug;

int scanTime = 5; //In seconds
BLEScan* pBLEScan;

class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
    void onResult(BLEAdvertisedDevice advertisedDevice) {
      _debug.printf("Advertised Device: %s \n", advertisedDevice.toString().c_str());
    }
};

void setup() {
  //esp_log_level_set("*", ESP_LOG_ERROR);
  Serial.begin(115200);
    SerialBT.begin("Logger");
  _debug.println("Scanning...");

  BLEDevice::init("");
  pBLEScan = BLEDevice::getScan(); //create new scan
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setActiveScan(true); //active scan uses more power, but get results faster
  pBLEScan->setInterval(100);
  pBLEScan->setWindow(99);  // less or equal setInterval value
}

void loop() {
  // put your main code here, to run repeatedly:
  BLEScanResults foundDevices = pBLEScan->start(scanTime, false);
  _debug.print("Devices found: ");
  _debug.println(foundDevices.getCount());
  _debug.println("Scan done!");
  pBLEScan->clearResults();   // delete results fromBLEScan buffer to release memory
  delay(2000);
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```
It freezes, no more messages.

aovestdipaperino, 2020-12-06:
Of course if I save all the devices in a buffer and print it after the scan is done everything works.

aovestdipaperino, 2020-12-06:
I am suspecting that the BT stack is not re-entrant...

btsimonh, 2020-12-12:
the advert callback is likely called whilst holding a mutex, which is required to call SerialBT, and is not of the nested type.

craiq, 2021-02-19:
I've a similar problem. In my setup I'm streaming data via BT while receiving notifications via BLE.
When the BLE server is connected, the BT connection freezes after some time (5min - 20min) and the BLE does reconnect.
When the BLE server is not connected, the BT connection remains stable over hours.

aovestdipaperino, 2021-03-01:
I will be trying with the latest version of the stack and get back

craiq, 2021-03-01:
On a longer test I did not experience the problem anymore

closed 2021-07-09 by stale[bot]