espressif/arduino-esp32 #4615: Diagnosing: {"RestartReason":"Reset by SLC module, reset digital core"}

jpadie, 2020-12-08:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?
>>toggling relay states causes a CPU reset.  The behaviour is not consistent.

Describe your system( Hardware, computer, O/S, core version, environment).
>>own board with ESP-wroom-32.  Four relays (of which three are in use), 2 DS18b20s and two switches.  There are also ACS712s on the board but I am not using them.

Describe what is failing.
>>the board resets

Show the shortest possible code that will duplicate the error.
>>toggling a relay from the webUI, an external switch or mqtt often (but not aways) causes a CPU reset.

Show the EXACT error message(it doesn't work is not enough).
>>{"RestartReason":"Reset by SLC module, reset digital core"}

All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.
>>this is the third module I've tried on the board. the first was giving guru meditation errors but the crash dump was never fully output in the serial terminal.  the second seemed to be DoA.  The third was left on my lab-bench over night with a script sending mqtt messages every 10 seconds with different power states.  there were no glitches.
Installing it in situ today involved just connecting the relays to their loads (two circulation pumps and an external beefy relay switching an immersion heater).  

I've tried both pre-built binaries (sensor32) and compiling my own.  

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

>> the serial terminal sometimes says guru meditation error but nothing is output for the error. With this module I am not seeing any guru meditation errors.

The power supply is fine.  Tried with others to check. 

as above, on the lab bench before installation the board behaves itself.  there is nothing untoward about the installation location - the relay loads are inductive, but low power.  the switches are connected to dry contacts and are both pulled up with 10k resistors.

compilation and flashing is done with the latest pull from github and using the platformio config with no changes other than baud rate and port.  I've tried with vanilla config and with my usual config with pre-assigned SSID and mqtt data.  Same behaviour.

With pre-built binaries I have used with esptool.py following the instructions.  this was for the first module. I've not tried the pre-built binaries on the current module.  

seemingly only relays 1 and 2 (which are directly connected to pumps) are causing problems at the moment.  Relay 3 is not connected, relay 4 is connected to a contactor and seems to be working ok.  

all relays have flyback diodes.  

an example output from the serial terminal is below

```
12:51:55 CMD: Group 0, Index 2, Command "POWER", ⸮at⸮ets Jun  8 2016 00:22:57

⸮ͱ⸮0x6 (SDIO_RESET),boot:0xqets Jun  8 2016 00:22:57

rst:0x6 (SDIO_RE⸮⸮⸮i\boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10100
ho 0 tail 12 room 4
load:0x40080400,len:5828
entry 0x400806a8
⸮
00:00:00 CFG: Loaded, Count 4741
00:00:00 QPC: Count 1
00:00:00 CFG: CR 408/699, Busy 0
00:00:00 SRC: Restart
00:00:00 Pro⸮⸮⸮t tasmota T⸮⸮⸮⸮⸮a Version 9.1.0.2(sensors)-1_0_4_2(2020-12-07T19:39:32)
00:00:00 DSB: Sensors found 0
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP2 jpadie Channel 11 BSSId 00:27:22:20:10:5C in mode 11n as centralHeating-3561...
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Connected
12:51:58 RTC: UTC 2020-12-08T11:51:58, DST 2020-03-29T02:00:00, STD 2020-10-25T03:00:00
12:51:58 HTP: Web server active on centralHeating-3561 with IP address 192.168.0.78
12:51:58 MQT: Attempting connection...
12:51:59 MQT: Connected
12:51:59 MQT: rab/centralHeating/tele/LWT = Online (retained)
12:51:59 MQT: rab/centralHeating/cmnd/POWER = 
12:51:59 MQT: Subscribe to rab/centralHeating/cmnd/#
12:51:59 MQT: Subscribe to rab/tasmotas/cmnd/#
12:51:59 MQT: Subscribe to cmnd/DVES_034DE9_fb/#
12:51:59 MQT: Unsubscribe from homeassistant/status
12:51:59 MQT: rab/centralHeating/tele/INFO1 = {"Module":"ESP32-DevKit","Version":"9.1.0.2(sensors)","FallbackTopic":"cmnd/DVES_034DE9_fb/","GroupTopic":"rab/tasmotas/cmnd/"}
12:51:59 MQT: rab/centralHeating/tele/INFO2 = {"WebServerMode":"Admin","Hostname":"centralHeating-3561","IPAddress":"192.168.0.78"}
12:51:59 MQT: rab/centralHeating/tele/INFO3 = {"RestartReason":"Reset by SLC module, reset digital core"}

```

relays 1 and 2 are connected to GPIO4 and GPIO17 respectively.  They are NC relays and are set as relay_i in the config page. 

Any help understanding this error message and pointers towards debugging would be appreciated.

thanks
Justin

lbernstone, 2020-12-08:
This forum is for issues with the arduino-esp32 code, general issues should be posted at https://esp32.com.  You have an electrical problem.  Sounds to me like you are trying to drive a relay directly from an mcu pin.  The esp32 can only supply 42mA to each pin, and I would not recommend more than half that for most cases.  Flip a mosfet with your mcu pin, and drive the relay off 5V (through that mosfet).

jpadie, 2020-12-08:
thank you.  this may be an electrical problem but it is unlikely.  the relay coils are being driven by a 'fet and properly snubbed.

jpadie, 2020-12-08:
closing this issue.  I will seek help elsewhere.

closed 2020-12-08 by jpadie