espressif/arduino-esp32 #4618: ESPAsyncWebServer Periodically Rebooting the ESP

danspage, 2020-12-10:
### Hardware:
Board:						      Sparkfun ESP32 WROOM Thing Plus
Core Installation version:		      v3.2.3-14-gd3e562907
IDE name:				              Platform.io
Flash Frequency:				      40 MHz
PSRAM enabled:                                    No
Upload Speed:				      115200
Computer OS:                                        Windows 10

### Description:
First, I should mention that this is the only code in this project that deals with ESPAsyncWebServer, and the problem has only occurred after adding the library to this project, so it's almost definitely something with it. The problem is that occasionally the board will reboot, without any obvious cause other than `abort() was called at PC 0x4014f9e3 on core 1` printing to the console, followed by the stacktrace below. `startDataViewer()` is basically the setup function, and is called by a central class upon startup. Everything else seems to be working fine so I'm at a bit of a loss here. The `sendEvent` function is being run a few times each second, but it works smoothly and does what's expected. If anyone knows what's going on, it would be much appreciated!

Cheers

### Code:
```cpp
#ifndef __DATA_VIEWER_H__
#define __DATA_VIEWER_H__

#include <Arduino.h>
#include <AsyncTCP.h>
#include <DataViewer/SharedVariables.h>
#include <ESPAsyncWebServer.h>
#include <SharedManagers.h>
#include <WiFi.h>
#include <WebDeps/WebDepHeaders.h>
#include <DataViewer/PageManager.h>

AsyncWebServer server(80);
AsyncEventSource events("/events");

void sendEvent(const char* event, const char* sendStr) {
  events.send(sendStr, event);
}

void startDataViewer() {
    WiFi.disconnect();

    WiFi.softAP(settingsManager.getSettingValue("wifiSSID").c_str(),
                settingsManager.getSettingValue("wifiPass").c_str());

    Serial.println();
    Serial.print("IP address: ");
    Serial.println(WiFi.softAPIP());

    // ================================== Web depenencies ==================================
    server.on("/deps/bootstrap.css", HTTP_GET, [](AsyncWebServerRequest* request) {
        AsyncWebServerResponse* response = request->beginResponse_P(
            200, "text/css", BOOTSTRAP_CSS, BOOTSTRAP_CSS_LEN);
        response->addHeader("Content-Encoding", "gzip");
        request->send(response);
    });
    server.on("/deps/bootstrap_bundle.js", HTTP_GET, [](AsyncWebServerRequest* request) {
        AsyncWebServerResponse* response = request->beginResponse_P(
            200, "text/javascript", BOOTSTRAP_BUNDLE_JS, BOOTSTRAP_BUNDLE_JS_LEN);
        response->addHeader("Content-Encoding", "gzip");
        request->send(response);
    });
    server.on("/deps/jquery.js", HTTP_GET, [](AsyncWebServerRequest* request) {
        AsyncWebServerResponse* response = request->beginResponse_P(
            200, "text/javascript", JQUERY_JS, JQUERY_JS_LEN);
        response->addHeader("Content-Encoding", "gzip");
        request->send(response);
    });
    server.on("/deps/chart.js", HTTP_GET, [](AsyncWebServerRequest* request) {
        AsyncWebServerResponse* response = request->beginResponse_P(
            200, "text/javascript", CHART_JS, CHART_JS_LEN);
        response->addHeader("Content-Encoding", "gzip");
        request->send(response);
    });
    server.on("/deps/luxon.js", HTTP_GET, [](AsyncWebServerRequest* request) {
        AsyncWebServerResponse* response = request->beginResponse_P(
            200, "text/javascript", LUXON_JS, LUXON_JS_LEN);
        response->addHeader("Content-Encoding", "gzip");
        request->send(response);
    });
    server.on("/deps/chartjs_luxon_adapter.js", HTTP_GET, [](AsyncWebServerRequest* request) {
        AsyncWebServerResponse* response = request->beginResponse_P(
            200, "text/javascript", CHARTJS_LUXON_ADAPTER_JS, CHARTJS_LUXON_ADAPTER_JS_LEN);
        response->addHeader("Content-Encoding", "gzip");
        request->send(response);
    });
    server.on("/deps/realtime_charts.js", HTTP_GET, [](AsyncWebServerRequest* request) {
        AsyncWebServerResponse* response = request->beginResponse_P(
            200, "text/javascript", REALTIME_CHARTS_JS, REALTIME_CHARTS_JS_LEN);
        response->addHeader("Content-Encoding", "gzip");
        request->send(response);
    });
    server.on("/deps/min_ajax.js", HTTP_GET, [](AsyncWebServerRequest* request) {
        AsyncWebServerResponse* response = request->beginResponse_P(
            200, "text/javascript", MIN_AJAX_JS, MIN_AJAX_JS_LEN);
        response->addHeader("Content-Encoding", "gzip");
        request->send(response);
    });
    // =============================== End of web depenencies ==============================

    dv_initPages();

    server.on("/", HTTP_GET, [](AsyncWebServerRequest* request) {
        request->send(
            200, "text/html", dv_getPageHTML("index"));
    });

    server.addHandler(&events);

    server.begin();

    dataManager.initDataset((char*)"flex", 10, 5);
}
```

### Debug Messages:
```
0x4008c800: rtc_init at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/soc/esp32/rtc_init.c line 64
0x4008ca31: rtc_init at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/soc/esp32/rtc_init.c line 95
0x400d348b: __deregister_frame_info_bases at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/unwind-dw2-fde.c line 216
0x400d3b82: esp_timer_impl_init at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/esp_timer_esp32.c line 369
0x400d3bd4: cpu_configure_region_protection at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/soc/esp32/include/soc/cpu.h line 46
0x400d3be3: main_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/cpu_start.c line 464
0x40088f49: vTaskSwitchContext at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 2816
```

danspage, 2020-12-10:
Forgot to mention, `dv_getPageHTML` just returns a string, so it's nothing interesting.

me-no-dev, 2020-12-10:
seems you have many JS files streamed. If that happens for one page, then you will get more requests than the ESP can handle. Very much possible that you are running out of memory somewhere or that the network gos crazy. Try combining the JS files into one.

danspage, 2020-12-14:
Yep, turns out I was able to move the Server.on's over to core 0 and it fixed the problem. Thanks for the insight!

closed 2020-12-14 by danspage