espressif/arduino-esp32 #4619: Bluetooth pariing problem: BT_BTM: BTM_GetSecurityFlags false

ZweiEuro, 2020-12-10:
Development Kit: ESP32-Wrover-Kit
    Module or chip used: ESP32-WROOM-32
    Build System: platformio
    Compiler version: xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
    Operating System: Linux
    Using an IDE?: Yes VSCode with platformio
    Power Supply: USB

### Description:
running :
https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/examples/BLE_server/BLE_server.ino

and trying to pair will only print out:
BT_BTM: BTM_GetSecurityFlags false

pairing will fail on my phone. 
after this it won't try at all anymore until i reboot.

ZweiEuro, 2021-02-15:
What other info should i provide?

ZweiEuro, 2021-03-08:
I think the issue was my own code which wasn't fully set up, after redoing it all it works fine

closed 2021-03-08 by ZweiEuro

romocitto88, 2022-03-02:
Could you explain what did you miss on your set up? Because I'm facing the same issue.

javiersanzsanchez, 2023-08-17:
Do you have a solution for that?