espressif/arduino-esp32 #4623: WiFi issue in SoftAP and Station Mode

naakhtar, 2020-12-11:
### Hardware:
Board:							ESP32-DOWD
Core Installation version:			1.9
IDE name:						Docklight
Flash Frequency:					
PSRAM enabled:                                    
Upload Speed:					115200bits/sec
Computer OS:                                        WINDOW 7

### Description:
Describe your problem here
> Esp32 module while using WIFI mode in SoftAP and station in both getting below issue.

> After executing (AT+RESTORE) all other command gives -y only.
For example. 
> Restore Command (AT+RESTORE) it gives status Ok. 
> Execute WIFI mode command (AT+CWMODE) receive status  [RX} –y, 
> Execute all other command receive status [RX] - y .  

[image]

lbernstone, 2020-12-11:
If you are running arduino code, please post it.  Otherwise, move your issue to https://esp32.com

naakhtar, 2021-02-22:
“AT+HTTPCLIENT=1,3,"http://103.43.40.23:8581/UPSData/GS19183100001DCB","103.43.40.23:8581",,1” command to test .

closed 2021-02-22 by naakhtar