espressif/arduino-esp32 #4625: ESP32 crashed on boot : SPI RAM not initialized (idf v4.2)

happyme531, 2020-12-11:
### Hardware:
Board:							
Core Installation version:			idf-release/v4.2 master
IDE name:						Platform.io
Flash Frequency:					80Mhz
PSRAM enabled:                                      yes
Upload Speed:				        2000000
Computer OS:                                         Windows 10

### Description:
The esp32 just crashed on start.

### Sketch:  
Since this problem happens so early that even earlier than a \_\_attribute\_\_((constructor)) marked function,
so I don't think pasting a sketch will be helpful....

Here is the platformio config file.
```ini

[env:ttgo-t-watch]
platform = espressif32
board = ttgo-t-watch
framework = arduino
board_build.f_flash = 80000000L
board_build.flash_mode = qio
monitor_speed = 115200
upload_speed = 2000000
monitor_filters = 
	default
	esp32_exception_decoder
build_unflags = -std=gnu++11 -Werror=reorder 
build_flags = 
	-std=gnu++2a	; c++2a->compiling error
	-Dtypeof(x)=decltype(x) 	; no longer useful
	-DCONFIG_ARDUHAL_LOG_COLORS=1 
	-DCORE_DEBUG_LEVEL=4 
	-DBOARD_HAS_PSRAM 
        -mfix-esp32-psram-cache-issue
build_type = release
lib_deps = 
	ESP Async WebServer
	xinyuan-lilygo/TTGO TWatch Library@^1.3.0
	bodmer/TFT_eSPI@^2.2.21
	aharshac/EasyNTPClient@^1.1.0
	ricmoo/QRCode@^0.0.1
src_filter = +<*> -<.git/> -<cparse/test-shunting-yard.cpp>
platform_packages = 
	toolchain-xtensa32 @ 2.80200.200827
	framework-arduinoespressif32 @ http://gitclone.com/github.com/espressif/arduino-esp32.git#idf-release/v4.2

```

### Debug Messages:
```

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4
load:0x3fff0034,len:1576
load:0x40078000,len:12700
load:0x40080400,len:3040
entry 0x400805e4
E (1045) spiram: SPI RAM not initialized

abort() was called at PC 0x40167e0b on core 0

Backtrace:0x4008ef51:0x3ffe3af0 0x4008f8d5:0x3ffe3b10 0x4009525d:0x3ffe3b30 0x40167e0b:0x3ffe3bb0 0x40167f73:0x3ffe3bd0 0x40167cf3:0x3ffe3bf0 0x4020d37b:0x3ffe3c20 0x40083146:0x3ffe3c50 0x40078f67:0x3ffe3c90 |<-CORRUPTED
  #0  0x4008ef51:0x3ffe3af0 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:360
  #1  0x4008f8d5:0x3ffe3b10 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/system_api.c:106
  #2  0x4009525d:0x3ffe3b30 in abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/abort.c:46
  #3  0x40167e0b:0x3ffe3bb0 in esp_spiram_get_chip_size at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp32/spiram.c:132
      (inlined by) esp_spiram_get_chip_size at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp32/spiram.c:128
  #4  0x40167f73:0x3ffe3bd0 in esp_spiram_get_size at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp32/spiram.c:220
  #5  0x40167cf3:0x3ffe3bf0 in esp_himem_init at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp32/esp_himem.c:143
  #6  0x4020d37b:0x3ffe3c20 in do_global_ctors at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/startup.c:145
      (inlined by) start_cpu0_default at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/startup.c:346
  #7  0x40083146:0x3ffe3c50 in call_start_cpu0 at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/port/cpu_start.c:493
  #8  0x40078f67:0x3ffe3c90 in ?? ??:0

ELF file SHA256: 0000000000000000

Rebooting...

```

Note:
In idf v4.0, there is also
```
E (865) spiram: SPI RAM not initialized
E (865) esp_himem: Cannot allocate memory for meta info. Not initializing!

```
in log, but it doesn't aborts.
Something maybe also worth to notice is that my project have used 2660888 bytes of code space(due to the use of serval fonts).

lbernstone, 2020-12-11:
This repo does not provide platformio support. Things have changed significantly with v4.2. https://github.com/platformio/platform-espressif32/

happyme531, 2020-12-12:
Set the core debug level to warn (or -DCORE_DEBUG_LEVEL=2) solved this problem.   
BUT WHY???

happyme531, 2020-12-12:
From my test result, the problem is not related to the difference Arduino or Platformio.

happyme531, 2020-12-21:
Issue solved. It is caused by:
```cpp  
ESP_LOGI("init", "himem free %u\n", esp_himem_get_free_size());
ESP_LOGI("init", "himem phys %u\n", esp_himem_get_phys_size());
ESP_LOGI("init", "himem reserved %u\n", esp_himem_reserved_area_size());
```
But how can a function to cause crash before it is called?

closed 2021-07-08 by stale[bot]