espressif/arduino-esp32 #4629: LITTLEFS library causing release/v4.2 to fail to compile

william-ferguson-au, 2020-12-12:
### Hardware:
Board:							NA
Core Installation version:			idf-release/v4.2
IDE name:						NA
Flash Frequency:					NA
PSRAM enabled:                                     NA
Upload Speed:			 	        NA
Computer OS:                                        Windows 10

### Description:
I'm trying compile arduino-esp32 idf_release/v4.2 as a component in an esp-idf project using Cake.
When I do it fails with a compilation error in the LITLEFS library of arduino-esp32
It is trying to find esp_litlefs.h which does not exist.

### Debug Messages:

Compiler output
```
idf.y build

E:\Source\my_project\components\arduino-esp32\libraries\LITTLEFS\src\LITTLEFS.cpp:25:10: fatal error: esp_littlefs.h: No such file or directory
 #include "esp_littlefs.h"
          ^~~~~~~~~~~~~~~~
```

lbernstone, 2020-12-12:
You either need to include the repo in components, or remark it out of CMakeLists.

william-ferguson-au, 2020-12-12:
I'm sorry, I don't understand.
Include what repo in components or remark what repo out of CMakelists?

I have managed to get arduino-esp32 to build by commenting out 
`
  libraries/LITTLEFS/src/LITTLEFS.cpp
  libraries/LITTLEFS/src
`
in arduino-esp32/CMakeLists.txt

But that is just a hack to get it to arduino-esp32 to build. 
How is arduino-esp32 meant to get built with the LITTLEFS library?

Burnett01, 2020-12-26:
I ran into the same issue whilst building arduino-esp32 with esp-idf. None of the official esp-idf 4.x versions worked right straight out-of-the-box with arduino-esp32 4.x branches, so I read many posts where the authors of arduino-esp32 provided other users with help to make it work. 

Before I managed to get arduino-esp32 to build, it was complaining about ``esp_littlefs.h``.
I quickly searched for that header file on Github and found the official repository, where the author gives advice on how to install it.

Rather than commenting it out in CMakeLists.txt, you can navigate to your esp-idf/components directory and use:

```
git submodule add https://github.com/joltwallet/esp_littlefs.git
git submodule update --init --recursive
```

After that, trigger a new build and it'll work.

Ref: https://github.com/joltwallet/esp_littlefs

closed 2021-07-08 by stale[bot]