espressif/arduino-esp32 #4630: pico32d4 A fatal error occurred: Timed out waiting for packet content

itsieber, 2020-12-12:
Pico32d4 can not programming:

`esptool.py v2.6-beta1
Serial port COM4
Connecting........__
Detecting chip type... ESP32
Chip is ESP32-PICO-D4 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
MAC: 24:a1:60:57:17:70
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0xffffff, SizeID=0xff), defaulting to 4MB
Compressed 8192 bytes to 47...

A fatal error occurred: Timed out waiting for packet content`

Pinout:
1 +3.3
2 LNA
3 +3.3
4 +3.3
9 10k -> +3.3
19 +3.3
23 10k -> +3.3
37 +3.3
39 I2C
40 RXD
41 TXD
42 I2C
43 +3.3
46 +3.3
49 GND

itsieber, 2020-12-15:
Need Help!

lbernstone, 2020-12-15:
https://github.com/espressif/arduino-esp32/blob/master/docs/ISSUE_TEMPLATE.md

lbernstone, 2020-12-27:
This repo is not for esptool or platformio.  Test esptool independently.  If that works, you will need to take this up at [platformio-espressif32](https://github.com/platformio/platform-espressif32).  Also see https://github.com/espressif/esptool/issues/459

itsieber, 2020-12-28:
https://github.com/espressif/esptool/issues/593

closed 2020-12-28 by itsieber