espressif/arduino-esp32 #4632: SimpleTime.ino example don't work with static station address

kendo55, 2020-12-13:
Hi 
we original code runs well!
But if add additional code for work with static station address,  
configTime()  
has no connection to the pool.ntp.org
in other code, like access with http_request, this static config works very well!

```
#include <WiFi.h>
#include "time.h"

#define MODIFICATION_TO_ORIGINAL   1

const char* ssid       = "xxxxxxxx";
const char* password   = "yyyyyyyyy";

const char* ntpServer = "pool.ntp.org";
const long  gmtOffset_sec = 3600;
const int   daylightOffset_sec = 3600;

void printLocalTime()
{
  struct tm timeinfo;
  if(!getLocalTime(&timeinfo)){
    Serial.println("Failed to obtain time");
    return;
  }
  Serial.println(&timeinfo, "%A, %B %d %Y %H:%M:%S");
}

void setup()
{
  Serial.begin(115200);
  
  //connect to WiFi
  Serial.printf("Connecting to %s ", ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.print(".");
  }

#if MODIFICATION_TO_ORIGINAL
  IPAddress ip(192, 168, 1, 160);     // fix station ip.
  IPAddress gateway(192, 168, 1, 4);  // gateway
  IPAddress subnet(255, 255, 255, 0); // set subnet mask to match your network 
  WiFi.config(ip, gateway, subnet);
  Serial.printf("Connected to SSID: %s IP address: %s\n", ssid, WiFi.localIP().toString().c_str());
#endif
  
  Serial.println(" CONNECTED");
  
  //init and get the time
  configTime(gmtOffset_sec, daylightOffset_sec, ntpServer);
  printLocalTime();

  //disconnect WiFi as it's no longer needed
  WiFi.disconnect(true);
  WiFi.mode(WIFI_OFF);
}

void loop()
{
  delay(1000);
  printLocalTime();
}
```
Arduino IDE 1.8.13
Boardmanager: https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json
Version: 1.0.5-rc4

felmue, 2020-12-13:
Hello @kendo55 

I think you're missing the DNS IP address. And I'd set the static IP **before** `Wifi.begin()` else a DHCP address is requested first before the static address is used.
```
  IPAddress ip(192, 168, 1, 160);     // fix station ip.
  IPAddress gateway(192, 168, 1, 4);  // gateway
  IPAddress subnet(255, 255, 255, 0); // set subnet mask to match your network 
  IPAddress dns(192, 168, 1, 4); // assuming your gateway is also your DNS, else try 8.8.8.8 (Google's DNS)
  WiFi.config(ip, gateway, subnet, dns);
```
Thanks
Felix

kendo55, 2020-12-14:
hi Felix
thousand thanks!
 It works! 
Have a nice Day!
Kendo

closed 2020-12-14 by kendo55

felmue, 2020-12-14:
Hello @kendo55 

you are welcome. I am glad to hear it works now.

Thank you! A nice Day to you too.

Cheers
Felix

kendo55, 2020-12-14:
Hello Felix
really nice! 
I have a near equal Problem if I use the IDF with eclipse.
could you please have a look at the issue #6257
https://github.com/espressif/esp-idf/issues/6257
Thanks
Kendo