espressif/arduino-esp32 #4636: Serial Not Available

johnsmakar, 2020-12-16:
### Hardware:
Board:							ESP32 NodeMCU
Core Installation version:			1.0.4
IDE name:						Arduino 1.8.14
Flash Frequency:					80Mhz
PSRAM enabled:                                     Disabled
Upload Speed:				       115200
Computer OS:                                        Windows 10

### Description:
I am trying to Serial.readString() and Serial.write() but it says that the Serial is not available, I isolated the issue to its own sketch in the case that something else was messing with it. I was working fine a couple of days ago, not sure what happened.

### Sketch:  (leave the backquotes for [code formatting]
```cpp

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);         // Init serial port and set baudrate
  while(!Serial);                     // Wait for serial port to connect
  if (!Serial.available())
    Serial.println("Setup Busy");
  else
    Serial.println("Setup Open");
}

void loop() {
  // put your main code here, to run repeatedly:
  if (!Serial.available())
    Serial.println("Serial Busy");
  else
    Serial.println("Serial Open");
  delay(3000);
   if (!Serial.available())
    Serial.println("STIIL Busy");
   else
    Serial.println("STILL Open");
}
```

### Debug Messages:
```
Serial Busy
STIIL Busy
```

lbernstone, 2020-12-16:
You keep using that word.  I do not think it means what you think it means.
https://www.arduino.cc/reference/en/language/functions/communication/serial/available/

johnsmakar, 2020-12-16:
I apologized for any misunderstanding on my part...

My original sketch is:
```
void loop() {
// When program is running stream of data is being written to serial
// iff the data stops terminate the loop.
if(!Serial.available()) return;

String str = Serial.readString(); // Read incoming bytes
    if (str == "I")
    {
      long int lVal = pData;
      byte lBuffer[] = {
        byte(lVal & 0xff),
        byte(lVal >> 8 & 0xff),
        byte(lVal >> 16 & 0xff),
        byte(lVal >> 24 & 0xff)
      };
      Serial.write(lBuffer, 4); // Response
    }
}
```

The program expects a steady stream of incoming data to read... It worked fine last time ran it
I changed the program to `if(Serial.available() > 0 )` to read when there are incoming bytes to read and still have no luck.

closed 2020-12-16 by johnsmakar