espressif/arduino-esp32 #4639: Memory Leak in WiFi.begin() when network is out of range

intakenick, 2020-12-17:
When connecting to client, via wifi.begin(), I receive a guru meditation error but only when in one room of my house. When situated closer to the router, it connects fine. When completely out of range, it times out and fails to connect as expected.  It seems that there is a memory issue in one of the core libraries. Currently using a Node32s. See code and dump below:

`bool WiFiModem::ConnectClient() {
    _logger->SendMessage("WiFi Connecting!");
    _isClientConnected = _wifi.status() == WL_CONNECTED;
    if (_isClientInitialized) {
        WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0); //disable brownout detector
        _logger->SendMessage("Starting connection...");
        _logger->SendMessage(_clientSSID);
        _logger->SendMessage(_clientPassword);
        if (!_isClientConnected) {
            _wifi.disconnect(true);
            _wifi.enableSTA(true);
            _logger->SendMessage("WiFi EnableSTA");
            _logger->SendMessage("WiFi Begin");
            const char* ssid = _clientSSID.c_str();
            const char* password = _clientPassword.c_str();
            _wifi.persistent(false);
            _wifi.setSleep(false);
            delay(10);
            _wifi.begin(ssid, password);
            _logger->SendMessage("WiFi SetSleep");
            
            long start=millis();
            _display->ClearDisplay();
            _display->DisplayLargeMessage(" WAIT");
            while (_wifi.status() != WL_CONNECTED) {
                //_logger->SendMessage("Waiting....."); 
                delay(200);   
                if(millis()%400<200){
                    led.setPixelColor(0,led.Color(0,0,100));
                    led.show();
                }      
                else{
                    led.clear();
                    led.show();
                }      
                if(abs(millis()-start)>10000){
                    _logger->SendMessage("WiFi Failed");  
                    led.clear();
                    led.show(); 
                    break;            
                }
                if(digitalRead(BUTTON)){
                    _display->ClearDisplay();
                    _isClientConnected=false;
                    led.clear();
                    led.show();
                    return 1;
                }
            }
            _logger->SendMessage("WiFi Connected...");
            //delay(50);
            led.clear();
            led.show();
            _display->ClearDisplay();            

        } else {
            _logger->SendMessage("WiFi already connected");
        }
        if (_wifi.status() == WL_CONNECTED) {
            _isClientConnected = true;
            _logger->SendMessage("Client IP Address: " + _wifi.localIP().toString());
        } else {
            _isClientConnected = false;
        }

    } else {
        _logger->SendMessage("WiFi NOT initialized!");
    }`

I receive the following error sometime while waiting in the ` while (_wifi.status() != WL_CONNECTED)` loop (decoded):

`Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Core 1 register dump:
PC      : 0x4000c271  PS      : 0x00060034  A0      : 0x8008bff8  A1      : 0x3ffb32b0  
A2      : 0x3ffb2654  A3      : 0x3f41f15c  A4      : 0x00000014  A5      : 0x3ffc0350  
A6      : 0x3ffc0398  A7      : 0x00000001  A8      : 0x00000001  A9      : 0x3f41f15d  
A10     : 0x000000a5  A11     : 0x3ffbc970  A12     : 0x8008cbaf  A13     : 0x3ffc0320  
A14     : 0x00000008  A15     : 0x00000001  SAR     : 0x00000012  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x4000c271:0x3ffb32b0 0x4008bff5:0x3ffb32d0 0x4008dbcc:0x3ffb32f0 0x4008db82:0xa5a5a5a5

Core 0 register dump:
PC      : 0x401ba072  PS      : 0x00060834  A0      : 0x80167ff6  A1      : 0x3ffbc3b0  
A2      : 0x00000000  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x00000001  
A6      : 0x00060d20  A7      : 0x00000000  A8      : 0x801666e2  A9      : 0x3ffbc380  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x8008b608  A13     : 0x3ffbc2b0  
A14     : 0x00000000  A15     : 0x3ffbc0a0  SAR     : 0x00000000  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x401ba072:0x3ffbc3b0 0x40167ff3:0x3ffbc3d0 0x4008cb9d:0x3ffbc3f0 0x4008b3f6:0x3ffbc410

`

`
PC: 0x4000c271
EXCVADDR: 0x00000000

Decoding stack results
0x4008bff5: vTaskSwitchContext at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 2770

PC: 0x401ba072: esp_pm_impl_waiti at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/pm_esp32.c line 492
EXCVADDR: 0x00000000

Decoding stack results
0x401ba072: esp_pm_impl_waiti at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/pm_esp32.c line 492
0x40167ff3: esp_vApplicationIdleHook at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/freertos_hooks.c line 63
0x4008cb9d: prvIdleTask at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 3382
0x4008b3f6: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
`

lbernstone, 2020-12-17:
Running a tight loop triggers the watchdog.  Put a delay in there so the system has some time to do what you want it to.

me-no-dev, 2020-12-18:
Also seems you decoded the backtrace on Core0 instead of Core1 where the issue shows. Using interrupts? Code for it?

Proto-ventures, 2020-12-18:
Both errors are decided there, I screwed up the code paste formatting somehow. Each starts with "decoding stack results". 

No interrupts during this loop, all that is running is the code you see above. Also tried adding delays into the loop but no improvement.

intakenick, 2020-12-18:
Both errors are decoded above, each start with "decoding stack results," I messed up the code paste formatting somehow. There are no interrupts, just the code above. There is some prior use of spiffs to store two HTML files but those are uslnused here.

Also tried adding delays to the loop but no luck. Any idea why this might work both in and out of range but not in the middle?

closed 2021-06-28 by stale[bot]