espressif/arduino-esp32 #4640: Possible issue in cacheing files with spaces in them

nitko12, 2020-12-17:
### Hardware:
Board:							https://github.com/e-radionicacom/Inkplate-6-hardware
Core Installation version:			1.0.5 RC4
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    yes
Upload Speed:					115200
Computer OS:                                        Mac OSX Big Sur

### Description:
Hi there,

We've been working on a library using your Arduino board definition.

We are using an ESP32 board as specified here. 
https://github.com/e-radionicacom/Inkplate-6-hardware

For some reason the bash script specified in platform.txt fails when theres a space in the file name.

We've diagnosed the issue to be a non escaped space here which causes cp command to fail.
https://github.com/espressif/arduino-esp32/blob/b05bdf690418efd28c8468b5cb8345d95ba77e9f/platform.txt#L69

You can try uploading any sketch as long as there's a space in the name to recreate.

Here's the error:
```
/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/nitkonitkic/Library/Arduino15/packages -hardware /Users/nitkonitkic/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/nitkonitkic/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/nitkonitkic/Documents/Arduino/libraries -fqbn=Croduino_Boards:esp32:nova32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,UploadSpeed=460800,DebugLevel=none -vid-pid=0000_0000 -ide-version=10813 -build-path /var/folders/j3/cv2rp50s14q26k18grs09x540000gn/T/arduino_build_382063 -warnings=none -build-cache /var/folders/j3/cv2rp50s14q26k18grs09x540000gn/T/arduino_cache_248896 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=/Users/nitkonitkic/Library/Arduino15/packages/Croduino_Boards/tools/xtensa-esp32-elf-gcc/1.22.0-96-g2852398-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-96-g2852398-5.2.0.path=/Users/nitkonitkic/Library/Arduino15/packages/Croduino_Boards/tools/xtensa-esp32-elf-gcc/1.22.0-96-g2852398-5.2.0 -prefs=runtime.tools.mkspiffs.path=/Users/nitkonitkic/Library/Arduino15/packages/Croduino_Boards/tools/mkspiffs/0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=/Users/nitkonitkic/Library/Arduino15/packages/Croduino_Boards/tools/mkspiffs/0.2.3 -prefs=runtime.tools.esptool_py.path=/Users/nitkonitkic/Library/Arduino15/packages/Croduino_Boards/tools/esptool_py/3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=/Users/nitkonitkic/Library/Arduino15/packages/Croduino_Boards/tools/esptool_py/3.0.0 -verbose /Users/nitkonitkic/Documents/Arduino/libraries/Inkplate-Arduino-library/examples/Basic Inkplate Functionality/Inkplate_basic_BW/Inkplate_basic_BW.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/nitkonitkic/Library/Arduino15/packages -hardware /Users/nitkonitkic/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/nitkonitkic/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/nitkonitkic/Documents/Arduino/libraries -fqbn=Croduino_Boards:esp32:nova32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,UploadSpeed=460800,DebugLevel=none -vid-pid=0000_0000 -ide-version=10813 -build-path /var/folders/j3/cv2rp50s14q26k18grs09x540000gn/T/arduino_build_382063 -warnings=none -build-cache /var/folders/j3/cv2rp50s14q26k18grs09x540000gn/T/arduino_cache_248896 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=/Users/nitkonitkic/Library/Arduino15/packages/Croduino_Boards/tools/xtensa-esp32-elf-gcc/1.22.0-96-g2852398-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-96-g2852398-5.2.0.path=/Users/nitkonitkic/Library/Arduino15/packages/Croduino_Boards/tools/xtensa-esp32-elf-gcc/1.22.0-96-g2852398-5.2.0 -prefs=runtime.tools.mkspiffs.path=/Users/nitkonitkic/Library/Arduino15/packages/Croduino_Boards/tools/mkspiffs/0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=/Users/nitkonitkic/Library/Arduino15/packages/Croduino_Boards/tools/mkspiffs/0.2.3 -prefs=runtime.tools.esptool_py.path=/Users/nitkonitkic/Library/Arduino15/packages/Croduino_Boards/tools/esptool_py/3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=/Users/nitkonitkic/Library/Arduino15/packages/Croduino_Boards/tools/esptool_py/3.0.0 -verbose /Users/nitkonitkic/Documents/Arduino/libraries/Inkplate-Arduino-library/examples/Basic Inkplate Functionality/Inkplate_basic_BW/Inkplate_basic_BW.ino
Using board 'nova32' from platform in folder: /Users/nitkonitkic/Library/Arduino15/packages/Croduino_Boards/hardware/esp32/1.0.1
Using core 'esp32' from platform in folder: /Users/nitkonitkic/Library/Arduino15/packages/Croduino_Boards/hardware/esp32/1.0.1
bash -c "[ ! -f /Users/nitkonitkic/Documents/Arduino/libraries/Inkplate-Arduino-library/examples/Basic Inkplate Functionality/Inkplate_basic_BW/partitions.csv ] || cp -f /Users/nitkonitkic/Documents/Arduino/libraries/Inkplate-Arduino-library/examples/Basic Inkplate Functionality/Inkplate_basic_BW/partitions.csv /var/folders/j3/cv2rp50s14q26k18grs09x540000gn/T/arduino_build_382063/partitions.csv"
bash: line 0: [: too many arguments
usage: cp [-R [-H | -L | -P]] [-fi | -n] [-apvXc] source_file target_file
       cp [-R [-H | -L | -P]] [-fi | -n] [-apvXc] source_file ... target_directory
exit status 64
/Applications/Arduino.app/Contents/Java/arduino-builder returned 64
Error compiling for board Croduino NOVA32 (ESP32).
```

Please ignore that it says NOVA32 and not Inkplate because it works the same for both.

I've tried adding single quotes where double quotes are now, and adding double quotes around the file names, but it gives this error then:

```
panic: runtime error: index out of range [-1]

goroutine 1 [running]:
github.com/arduino/arduino-cli/legacy/builder/utils.ParseCommandLine(0xc000164500, 0x134, 0x199c9c0, 0x1eead88, 0xc0000c6fc0, 0xc000164500, 0xc00014b6b0, 0xc000494000, 0x20a6d98)
	/Users/jenkins/go/pkg/mod/github.com/arduino/arduino-cli@v0.0.0-20200612152641-f7a6d63d208a/legacy/builder/utils/utils.go:63 +0x5d6
github.com/arduino/arduino-cli/legacy/builder/utils.PrepareCommandFilteredArgs(0xc000164500, 0x134, 0x18ada10, 0x199c9c0, 0x1eead88, 0x0, 0x0, 0x11, 0x10124a2, 0xc00014b788)
	/Users/jenkins/go/pkg/mod/github.com/arduino/arduino-cli@v0.0.0-20200612152641-f7a6d63d208a/legacy/builder/utils/utils.go:195 +0x67
github.com/arduino/arduino-cli/legacy/builder/utils.PrepareCommand(...)
	/Users/jenkins/go/pkg/mod/github.com/arduino/arduino-cli@v0.0.0-20200612152641-f7a6d63d208a/legacy/builder/utils/utils.go:233
github.com/arduino/arduino-cli/legacy/builder/builder_utils.PrepareCommandForRecipe(0xc000065c00, 0xc00014b8e0, 0xc00039da2d, 0x1f, 0x7ffeefbffb00, 0x23, 0x7ffeefbffbd9, 0x54)
	/Users/jenkins/go/pkg/mod/github.com/arduino/arduino-cli@v0.0.0-20200612152641-f7a6d63d208a/legacy/builder/builder_utils/utils.go:522 +0x226
github.com/arduino/arduino-cli/legacy/builder/builder_utils.ExecRecipe(0xc000065c00, 0xc00014b8e0, 0xc00039da2d, 0x1f, 0xc00014b800, 0x2, 0x1, 0x2, 0x1743ea0, 0xc0005420e0, ...)
	/Users/jenkins/go/pkg/mod/github.com/arduino/arduino-cli@v0.0.0-20200612152641-f7a6d63d208a/legacy/builder/builder_utils/utils.go:493 +0x65
github.com/arduino/arduino-cli/legacy/builder.(*RecipeByPrefixSuffixRunner).Run(0xc0001dab20, 0xc000065c00, 0xc0001dab20, 0x0)
	/Users/jenkins/go/pkg/mod/github.com/arduino/arduino-cli@v0.0.0-20200612152641-f7a6d63d208a/legacy/builder/recipe_runner.go:50 +0x2bc
github.com/arduino/arduino-cli/legacy/builder.runCommands(0xc000065c00, 0xc00014bae0, 0x20, 0x20, 0x0, 0x0)
	/Users/jenkins/go/pkg/mod/github.com/arduino/arduino-cli@v0.0.0-20200612152641-f7a6d63d208a/legacy/builder/builder.go:192 +0x19f
github.com/arduino/arduino-cli/legacy/builder.(*Builder).Run(0xc00014fd40, 0xc000065c00, 0x1eeac70, 0xc000279550)
	/Users/jenkins/go/pkg/mod/github.com/arduino/arduino-cli@v0.0.0-20200612152641-f7a6d63d208a/legacy/builder/builder.go:102 +0xb90
github.com/arduino/arduino-cli/legacy/builder.RunBuilder(...)
	/Users/jenkins/go/pkg/mod/github.com/arduino/arduino-cli@v0.0.0-20200612152641-f7a6d63d208a/legacy/builder/builder.go:210
main.main()
	/Users/jenkins/Jenkins/workspace/arduino-builder-mac-catalina/main.go:400 +0x7cd
/Applications/Arduino.app/Contents/Java/arduino-builder returned 2
Error compiling for board Croduino NOVA32 (ESP32).
```

Any skech with space in name:
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

me-no-dev, 2020-12-18:
So you tried everything that could be tried and it did not work... that is maybe because the IDE escapes things itself, other IDEs do as well. How do you propose we fix this? I propose you do not have spaces in the path

nitko12, 2020-12-18:
Okay,

that's exactly what I've done for now.

You could maybe start an issue on Arduino's repo about that.

laserir, 2021-01-27:
Could you tell us how we have to change the line to fix this?

aslobodskoy, 2021-02-01:
As far as I can see there is no one line solution to fix this issue. Here is patch example for another platform

https://github.com/stm32duino/Arduino_Core_STM32/pull/965/files
https://github.com/stm32duino/Arduino_Core_STM32/issues/957

closed 2021-07-08 by stale[bot]