espressif/arduino-esp32 #4643: Documentation issue: which esp-idf versions are supported?

laolux, 2020-12-18:
### Hardware:
Board:							Any ESP32 board
Core Installation version:			idf-release/v4.2
IDE name:						IDF component
Flash Frequency:					any
PSRAM enabled:					any
Upload Speed:					any
Computer OS:						Fedora

### Description:
Using arduino-esp32 as a component for the esp-idf does currently not work when using esp-idf v4.2.
Problems of this kind are documented in [docs/esp-idf_component.md](https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md). The document says that in case of compilation issues, one should "roll esp-idf back to a known good version". Well, how to find a known good version? I think it would be great to document which versions of the esp-idf are currently supported with which version/git commit of arduino-esp32. Maybe start by commenting known good version combinations here.

me-no-dev, 2020-12-18:
Try IDF master branch. I have not actually switched to 4.2 yet

lbernstone, 2020-12-18:
You can see the IDF git commit in [sdkconfig.h](https://github.com/espressif/arduino-esp32/blob/idf-release/v4.2/tools/sdk/esp32/include/config/sdkconfig.h#L620)

laolux, 2020-12-18:
@me-no-dev unfortunately fails with a different problem. So I need to use an old version of the esp-idf. That's not a big problem for me, as I use the docker containers anyways, so changing versions is quick. Just need to know which ones work.
@lbernstone Thanks, thats helpful. Maybe that info could be listed in `docs/esp-idf_component.md`? That might save someone else the work to search and wonder.

judge2005, 2020-12-18:
Which exact IDF release is each 1.0.5 release candidate delivered with? For example 1.0.5-rc3 says 'last release on IDF 3.3, but which version of 3.3? e.g. 3.3.2? Then 1.0.5-rc4 doesn't mention the IDF release at all.

laolux, 2020-12-21:
arduino-esp32: 1.0.5-rc4, esp-idf commit: b4c075169, from December 3, just before release v4.2b4c075169
arduino-esp32: 1.0.5-rc3, esp-idf commit: c33fc7821, from November 23, does not seem to be a tagged release
arduino-esp32: 1.0.4, esp-idf commit: d3e562907, from Sepember 20, before release v3.3.4
So, it seems as if this does not align with esp-idf releases.
In any case, I got all the information by looking at tools/sdk/include/config/sdkconfig.h, so it should be easy to find once documented.

closed 2021-07-11 by stale[bot]