espressif/arduino-esp32 #4644: BLE doesn't start advertise automatically after disconnection.

JLightMedia, 2020-12-18:
Hi everyone! 
I worked in 1.0.4 version, but after update MAC OS to BigSur I had to update to 1.0.5 library because the old version didn't work more.
But after that I have noticed one issue with advertise function. In setup function I start advertise BLE service, but when client disconnected from BLE server, ESP32 don't start advertising again. I fixed that added addition line in Callbacks service 

```cpp
     void onDisconnect(BLEServer* pServer) {
      pServer->getAdvertising()->start();
    }
```

Why BLE library doesn't start Advertising again, when client disconnected?

chegewara, 2020-12-18:
Its not starting advertising after disconnect because its not always expected or required.

raphael-bmec-co, 2021-03-08:
Seeing the same issue.

chegewara, 2021-03-08:
Not an issue, it is intended.

raphael-bmec-co, 2021-03-08:
Okay thank you. Can you confirm this is a change from 1.0.4?

chegewara, 2021-03-08:
Its not a change, it was from beginning.

raphael-bmec-co, 2021-03-08:
Good to know. Thanks again. Much appreciated.

closed 2021-06-28 by stale[bot]