espressif/arduino-esp32 #4646: HTTPClient redirection with different host does not replace the host

thegreatdane6, 2020-12-19:
After enabling the HTTPClient redirection, i kept running into a 404 Status Code. So I did some investigation and noticed that the host does not get changed if the host in the location header different than the previous one. 
I assume the issue must be somewhere within the "beginInternal" since i cannot see any changes related to this in #4240.
For everyone that cannot wait for a fix, I implemented a workaround for myself which you can check out [here](https://github.com/MaibornWolff/clean-air/pull/77/files)

"https://github.com/MaibornWolff/clean-air/releases/download/LATEST"
with response location
"Location: https://github.com/MaibornWolff/clean-air/releases/download/v1.1.0/LATEST"
leads to requesting
"https://github.com/MaibornWolff/clean-air/releases/download/v1.1.0/LATEST"

but 
"https://github.com/MaibornWolff/clean-air/releases/download/v1.1.0/LATEST"
with response location
"Location: https://github-production-release-asset-2e65be.s3.amazonaws.com/more/stuff?&soon"
leads to requesting
"https://github.com/more/stuff?&soon"

Batwam, 2021-02-22:
@thegreatdane6  did you find a solution to this issue?

me-no-dev, 2021-02-22:
it is fixed in the latest RC/master

closed 2021-02-22 by me-no-dev