espressif/arduino-esp32 #4647: MQTT - Guru Meditation Error: Core  1 panic'ed (LoadProhibited)

petrkoc, 2020-12-20:
Hello.
I am trying to use mqtt_client.h but receive Guru Meditation Error: Core  1 panic'ed (LoadProhibited) when trying to create MQTT client and connect to broker.

```cpp
#include <WiFi.h>
#include <mqtt_client.h>

void connectToWifi() {
// not shown there for clarity - works as expected
}

void WiFiEvent(WiFiEvent_t event) {
// not shown there for clarity - works as expected
}

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.println();

  Serial.println("Connecting to WiFi.");
  WiFi.onEvent(WiFiEvent);
  connectToWifi();
  delay(5000);
  
  esp_mqtt_client_config_t mqtt_cfg;
  mqtt_cfg.uri = "mqtt://mqtt.eclipse.org";

  esp_mqtt_client_handle_t mqtt_client = esp_mqtt_client_init(&mqtt_cfg);
  esp_mqtt_client_start(mqtt_client);
}

void loop() {
}
```

### Debug Messages:
```
Guru Meditation Error: Core  1 panic'ed (LoadStoreError). Exception was unhandled.
Core 1 register dump:
PC      : 0x4000c320  PS      : 0x00060c30  A0      : 0x8000146e  A1      : 0x3ffb1e70  
A2      : 0x3ffb6368  A3      : 0x40088b98  A4      : 0x00000003  A5      : 0x3ffb6368  
A6      : 0x00ff0000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffb1e40  
A10     : 0x3ffb6368  A11     : 0x00000003  A12     : 0x00060820  A13     : 0x00000001  
A14     : 0x00060a20  A15     : 0x00000000  SAR     : 0x00000018  EXCCAUSE: 0x00000003  
EXCVADDR: 0x40088b98  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x4000c320:0x3ffb1e70 0x4000146b:0x3ffb1e80 0x40001445:0x3ffb1ea0 0x400fd84b:0x3ffb1ec0 0x400fdf88:0x3ffb1ef0 0x400d12c8:0x3ffb1f20 0x400d27e3:0x3ffb1fb0 0x40088b9d:0x3ffb1fd0

```

PC: 0x4000c320
EXCVADDR: 0x40088b98: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 142

Decoding stack results
0x400fd84b: esp_mqtt_set_config at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c line 144
0x400fdf88: esp_mqtt_client_init at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c line 338
0x400d12c8: setup() at C:\Arduino\ESP32\ESP32_MQTT/ESP32_MQTT.ino line 105
0x400d27e3: loopTask(void*) at C:\Users\petrk\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\main.cpp line 14
0x40088b9d: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

lbernstone, 2020-12-21:
You will need to null out or fill all the parameters of the config structure.  C++ doesn't do this for you.

petrkoc, 2020-12-21:
You are right! After some testing I discovered following parameters have to be set:
.event_handle - this can be NULL
.host
.uri

Other are not required. Great, thank you! I would really welcome if there was an example showing how to use functionality.

closed 2020-12-21 by petrkoc

chegewara, 2020-12-21:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/mqtt.html#application-example

petrkoc, 2020-12-21:
> https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/mqtt.html#application-example

Well... the "Minimum configuration" on the link above sets only the .uri parameter and that is not enough in arduino-esp32 core as demonstrated above. Besides, the example uses esp_mqtt_client_register_event which is not present in arduino-esp32 core.

chegewara, 2020-12-21:
There is one good rule in C++ (read arduino). When you have data structure and you wont init all variables in that struct then you should init struct with:
`variable_struct = {0};`
It may save you many problems.

lbernstone, 2020-12-22:
events will be available in 1.0.5, which is right around the corner.  If you write and submit an example, I'm sure it will get merged.

WebDust21, 2023-05-20:
> There is one good rule in C++ (read arduino). When you have data structure and you wont init all variables in that struct then you should init struct with: `variable_struct = {0};` It may save you many problems.

Now that's a true lifesaver!  Thanks a million.