espressif/arduino-esp32 #4652: FS.h .open() return value

LocoMatt, 2020-12-22:
I'm working on app which uses SD card. I need to make exceptions if something failed to inform user. I noticed that Arduino-level software doesn't return error that I could easily handle. However vfs_api.h returns that something went wrong. Below is sample piece of code.
```cpp
File my_file = SD.open(file_name, "a");
   
if (!my_file)
{
  error = true; 
}
```
It returns 
```
 [E][vfs_api.cpp:265] VFSFileImpl(): fopen(/sd/file.txt) failed
```
and error isn't set to true

When I just open file with no mode pointed and there is no such file/card is unmounted, statement is fulfilled and I'm able to trigger error. Error sets it's value to true.
```cpp
File my_file = SD.open(file_name);
   
if (!my_file)
{
  error = true; 
}
```
And it returns 
```
[E][vfs_api.cpp:64] open(): /sd/file.txt does not exist
```

I'm unsure is it desired way it to work or not.

neutralvibes, 2021-01-31:
Still no solution?

fabiuz7, 2021-04-06:
Had you tried with the latest version (1.0.6)? Some updates regarding SD were released since your question.

LocoMatt, 2021-04-06:
> Had you tried with the latest version (1.0.6)? Some updates regarding SD were released since your question.

Will check it in near future witch nearest firmware update.

closed 2021-07-11 by stale[bot]