espressif/arduino-esp32 #4655: Extremely poor performance on ESP32-CAM board

UnforeseenOcean, 2020-12-23:
### Hardware:
Board: ESP32-CAM (Generic, using ESP32-S module)
Core Installation version: 1.0.4
IDE name: Arduino IDE
Flash Frequency:  Default
PSRAM enabled: Unsure (most likely yes, IPS6404LSO)
Upload Speed: 115200
Computer OS: Windows 10 / Android 10

### Description:
- Extremely poor performance on any setting
- Black screen after few seconds
- Many dropped frames
- Long delays between frames
- Stream does not even reach 15FPS
- Stream gets stuck
- Sluggish response 

### Sketch:
```cpp
#include "esp_camera.h"
#include <WiFi.h>

#define CAMERA_MODEL_AI_THINKER

#include "camera_pins.h"

const char* ssid = "ESP32CAM";
const char* password = "88888888";

void startCameraServer();

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  Serial.println();

  camera_config_t config;
  config.ledc_channel = LEDC_CHANNEL_0;
  config.ledc_timer = LEDC_TIMER_0;
  config.pin_d0 = Y2_GPIO_NUM;
  config.pin_d1 = Y3_GPIO_NUM;
  config.pin_d2 = Y4_GPIO_NUM;
  config.pin_d3 = Y5_GPIO_NUM;
  config.pin_d4 = Y6_GPIO_NUM;
  config.pin_d5 = Y7_GPIO_NUM;
  config.pin_d6 = Y8_GPIO_NUM;
  config.pin_d7 = Y9_GPIO_NUM;
  config.pin_xclk = XCLK_GPIO_NUM;
  config.pin_pclk = PCLK_GPIO_NUM;
  config.pin_vsync = VSYNC_GPIO_NUM;
  config.pin_href = HREF_GPIO_NUM;
  config.pin_sscb_sda = SIOD_GPIO_NUM;
  config.pin_sscb_scl = SIOC_GPIO_NUM;
  config.pin_pwdn = PWDN_GPIO_NUM;
  config.pin_reset = RESET_GPIO_NUM;
  config.xclk_freq_hz = 20000000;
  config.pixel_format = PIXFORMAT_JPEG;
  //init with high specs to pre-allocate larger buffers
  if(psramFound()){
    config.frame_size = FRAMESIZE_UXGA;
    config.jpeg_quality = 10;
    config.fb_count = 2;
  } else {
    config.frame_size = FRAMESIZE_SVGA;
    config.jpeg_quality = 12;
    config.fb_count = 1;
  }

#if defined(CAMERA_MODEL_ESP_EYE)
  pinMode(13, INPUT_PULLUP);
  pinMode(14, INPUT_PULLUP);
#endif

  // camera init
  esp_err_t err = esp_camera_init(&config);
  if (err != ESP_OK) {
    Serial.printf("Camera init failed with error 0x%x", err);
    return;
  }

  sensor_t * s = esp_camera_sensor_get();
  //initial sensors are flipped vertically and colors are a bit saturated
  if (s->id.PID == OV3660_PID) {
    s->set_vflip(s, 1);//flip it back
    s->set_brightness(s, 1);//up the blightness just a bit
    s->set_saturation(s, -2);//lower the saturation
  }
  //drop down frame size for higher initial frame rate
  s->set_framesize(s, FRAMESIZE_QVGA);

#if defined(CAMERA_MODEL_M5STACK_WIDE)
  s->set_vflip(s, 1);
  s->set_hmirror(s, 1);
#endif

  WiFi.setHostname("ESP32_CAM");
  delay(500);
  
  WiFi.begin(ssid, password);

  // Start AP mode
  WiFi.softAP(ssid, password);

  startCameraServer();
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(10000);
}
```

### Debug Messages:
[image]
[image]

Jason2866, 2020-12-24:
This is a known behaviour when power supply is to weak and/or cooling of the ESP32 is not good enough

raduprv, 2020-12-28:
A lot of people including me are having this problem. It's a hardware issue. Good power makes no difference. Cooling shouldn't be an issue.

Jason2866, 2020-12-29:
@raduprv  How should software solve a hardware issue?

raduprv, 2020-12-29:
No idea how it could solve it in this case, since no one seems to know what the hardware issue is. Some hardware issues can be fixed in software, some not.

MJavad, 2020-12-29:
I also have this problem with TTGO T-Camera Plus.

It also shows signs of temperature dependence and I sometimes get good results when switching from station to AP. It’s worth mentioning that I had to disable WiFi persistence (#2025) for AP to work.

I tried with esp-idf iperf example and had bad performance (1 to 5mbit) with station mode. AP had good speeds (15 to 30mbit).

raduprv, 2020-12-29:
I have a m5stack timer camera, and the wifi works perfectly (not so the USB drivers for the pc though).
My room temperature is constant, around 21-25c, and the board's wifi doesn't work properly even right after I start it.

MJavad, 2020-12-29:
Also there seems to be some correlation (not sure) between the performance and the temperature of the board at the startup. Could be some calibration problem (I think with persistence disabled, ESP33 does full calibration at startup)

MJavad, 2020-12-29:
@raduprv yes I also have bad performance when I start the board cool, but in my tests I sometimes had better performance after cooling when board is hot but just marginal.

I’m going to continue testing and check if there is really good evidence for this.

raduprv, 2020-12-29:
Some people reported (and I can confirm with the esp32 cam board) that when you stream video with the default test program, if you hold your fingers on certain parts of the board (towards the antenna) it gets much better.

MJavad, 2020-12-29:
Then I will test an external antenna soon. Can you post link of the “finger method”? :)

raduprv, 2020-12-29:
https://www.reddit.com/r/esp32/comments/c4iy1j/aithinker_esp32_cam_slow_video_stream_try/
https://forum.sparkfun.com/viewtopic.php?t=46563

Someone reported somewhere else that they still have wifi issues even with an external antenna, but if you are willing to try that, please let me know if it works. Don't forget to reposition the zero ohm resistor to the correct position.

MJavad, 2020-12-29:
Thanks I will.

FedericoBusero, 2020-12-30:
When I try to run the example code ESP32 > Camera > CameraWebServer it runs perfectly on my ESP32-CAM when choosing

`#define CAMERA_MODEL_AI_THINKER`

But you seem to have changed some lines of code in the setup, to run it in softAP mode. When I replace the wifi-code in the setup with

```
  WiFi.setHostname("ESP32_CAM");
  delay(500);

  // Start AP mode
  WiFi.softAP(ssid, password);

  startCameraServer();

  Serial.print("Camera Ready! Use 'http://");
  Serial.println(WiFi.softAPIP());
  Serial.println("' to connect");
```
it runs perfectly on 25 fps. I am running esp32 version 1.0.4 and using the module ESP32-CAM-MB.

Please notice that I removed following line of code:
`  WiFi.begin(ssid, password);`
as this is not necessary when running in softAP-mode.

But when I insert the line, in order to reproduce your problem, I indeed notice freezing issues, althoug not as bad as yours.

MJavad, 2020-12-30:
I'm currently testing with an external antenna, and I have great results.

I get 50 fps on lower resolutions (with xclk set to 10mhz), 25 fps on SVGA and 6 fps on UXGA without any jitter!

There is one issue that when I move the antenna cable over some components on the board, it seems to revert back to bad performance (tests with iperf show 1 to 3mbit maximum data rate) which is very similar to the on-board antenna.

With this in mind, I assume there is either some unexpected interference problem within hardware, or they just didn't properly tune the on-board antenna when designing the boards.

raduprv, 2020-12-31:
That's good to know. I don't think the antenna design is a problem, because not everyone is having this issue.

Can you test a ftp program to upload images to a ftp server, by any chance?
If not, can you try to NOT put the board in AP mode, and try the station mode?

MJavad, 2020-12-31:
There results were in station mode.

I had good results with AP and on-board antenna but I suspect that was because board and the computer were near each other. Right now the distance between board and router is about 2 meters with a wall in-between (signal strength reported by the board range from -52 to -44dbm, signal strength report from router range from -62 to -57dbm) .

I suspect ftp results would be similar to my iperf speed tests, which was 15mbit average.

raduprv, 2020-12-31:
Then it might be a different problem. Mine doesn't work properly even if it's at ~1 meter from the router, direct line of sight.
The m5stack ones works fine though.
The signal reported by the esp32 cam is ~-52db and on the m5stack timer cam it's around -36 db. Even so, -52 db is pretty strong.

MJavad, 2020-12-31:
That reminds me I also moved the board near the router and still had this problem. I'm inclined to agree with you that it's not directly related to just antenna.

But there were only two factors that changed between my tests: 
* Repositioning the antenna resistor and connecting external antenna
* Other tests were two weeks ago and there were updates
My soldering iron broke so I can't change back to on-board antenna anytime soon, but it would be good if you could test external antenna.

Just disconnecting the antenna makes it bad again.

Mungbeanz, 2021-01-20:
all your guys hard work help me come up with a simple solution.  I had some aluminium foil tape.  Just stuck about 4 layers ontop of the circuit board antenna and its solved.  On my second module I just hot glued on thin piece of the antistatic bag it came in.  Fixed.

drunkscientist, 2021-03-19:
so im dealing with the same issues as @UnforeseenOcean, but with some important details: it doesnt seem to make any difference what video size/resoloution/quality i select, result is the same(1-15seconds of flawless streaming before losing picture and displaying little x), temp makes no difference whatsoever, and most interesting: the board does NOT reset when it drops the stream. nor does it drop connection, so it is definitely not related to power supply, and im not sure is a hardware problem at all. im curious how the antenna plays into things, @Mungbeanz would you be willing to be more descriptive of what exactly you did? maybe a picture?

Mungbeanz, 2021-03-20:
On the side with the pins, just below the antenna is the circuit pcb antenna.  I attached a strip of antistatic bag and foil tape.
[image]

drunkscientist, 2021-03-28:
@Mungbeanz thanks. that didnt help but i appreciate the clarity. for anyone who arrives here from google like i did: the problem is the example provided by expressif sends its data in chunks/batches, this is also why it wont work in vlc or with blynk. i tried with a [different sketch](https://github.com/techiesms/esp32-cam-mjpeg) and it works as perfect as can be hoped for: the stream continues even if it drops a frame or 2 here and there

gwisp, 2021-06-18:
Bit stale, but I had issues with one board that seems to have a wide range of issue however in the run of it, the external antenna near the CPU case causes issues.

The FB buffer I set to 3. (Most sketch are 2)

From what I gather in semi tech knowledge, is that the frames are around 120kb, the core 0 handles the WiFi side and the Sketch runs on core 1.
With a buffer and interrupt this will slow the process.
The math 120k X 30(FPS) is 3.6Mb. doesn't leave much and therefore the ram refresh timing with a degradation from multicore use/access will show a slow down.

Try quality setting of 10 and see how this works, but there are two other issues, one being heat.
The other being power, the amps need to be higher than what the board uses. I get spikes around 820mh.(using 5v) it is often prudent to put in a 470 ohm cap to the power to ensure the spikes are "lowered".

I cleaned up the scripts and can run for 7-8 hours without issue, however cooling is a issue on the CPU and the ams1117 I soldered a Nickel tab to this to add extra cooling there due to the heat and the actual tab being very small. I think the tcase is 53, which it was easy getting too.

Hope that helps, 
Ps I think this is why the tape was working as it improved PCB cooling.

MJavad, 2021-06-18:
I suspect most issues come from interference between CPU, camera and antenna.
Since switching to external antenna I didn’t have any problem but if I move the antenna over the cpu or ram it will cause some issues.

Maybe the tape was working because it significantly reduced the interference.
I have already modified my board but it would be good to test if moving the camera away from the pcb may help.

andreondra, 2021-06-19:
@Mungbeanz Thank you for the tip! I just put a small piece of an aluminum foil on the antenna and it worked like a charm. Very weird issue.

TD-er, 2021-06-21:
Found this issue while searching for a solution with a completely unrelated board, but the issue does look very similar to what is reported here.

I do have a so called "Wemos" board (one I never seen before, therefore the quotes) running an ESP8266.
The board is not able to keep a WiFi connection, unless I press one end of the antenna between thumb and finger.
The antenna also does feel warm.
So this does sound a lot like what is described here and for this board it surely has nothing to do with weak power supply.
It does look like the antenna is not tuned to the right frequency as you can easily offset the tuned frequency of an antenna by holding it in your hands. Also the significant heat (hand warm) is a good indicator for this.
And the reported 'fix' to add some anti static foil next to the antenna does also proof this point.

The really strange part is that this specific board does seem to work with fimware versions of over 2 years old, so to me it sounds like there might still be something possible to fine tune using the RF calibration.
Has anyone dug into triggering this RF calibration?
I also cleared the flash etc. but at least the default calibration run does seem to produce the same (not working) calibration.
I know that there is a big difference between ESP8266 and ESP32, but the problem here is the same; the antenna tuning is wrong.

gwisp, 2021-06-21:
Id have to agree with the tune issue.
After my post I used a wif analyser to track the signals strength.
I've a "pole type" antenna, which looks to be the old design that has hung about since WiFi came out and would be around 2db.
I found that the signal was excessively weak by around 20db and I removed the external antenna and got a boost by 20db. The WiFi seems to be a short cast area, and the form is odd. This is to say I went further away and got a better signal (30feet approx).

When I made antenna many years ago, is say the PCB antenna is just off/right maybe. I'd have to look up the tech spec again, but the antenna is too narrowly spaced.

The antenna didn't work well, and I suspect the resistance may be too high to make for a good signal.

If someone wants to try, if you get copper wire and wrap it around your biggest finger and spaced correctly. And then testing it out.

Also checking the channel (default 1) for nearby users and interference, and therefore changing the channel.
Also is there a way to check the output power?

Also noticed the ping response was around 350ms (AVG) 
With spikes upto 580 and Low of around 64ms.
So there is a issue, that could be related to power or CPU time to address the WiFi requirements?

I'm struggling to get a good communication at 15 feet. But I may have to get out my old wrt 2w WiFi ap, pre 2007 WiFi being more powerful that today's and after when they limited the power outputs.

Also may help the WiFi white paper said 20db sensitivity. I'm getting about 32 reported by my router at a good day, but like I said it was and has been running fine. I'm wondering if the now out of date/missing WiFi.h  may have some impact??

strider222, 2021-07-15:
I use a wifi analyzer on my phone to test the signal strength of esp boards and find significant variations.
I run a web server demo and check the db of the signals. Some boards are unusable with wifi so I put them aside for other tasks.

sohtamei, 2021-11-09:
I fixed this issue, please see following report.
https://github.com/espressif/arduino-esp32/issues/5834

I released the firmware for ESP32-CAM which fixed this issue on above thread.

aguaviva, 2021-12-03:
Is lowering clocks affecting the performance in any way?

mosazahany, 2022-03-03:
Strange problems in adjusting the saturationand the brightness in ov2640
These bugs do not actually exist and are a problem of the available documents
When you execute the following commands
set_brightness(s,2);
set_contrast(s,2);
set_brightness(s,2);
The last command is always active and the previous commands are canceled.
To solve this problem, you must activate the desired effect and this is done with the following command

Run the following command at the end of all settings.

unsigned char enable_effect=	1(saturation and hue enable)+
				2(aditional saturation enable)+
				4(brighnes and contrast enable)+
				8(spital effect1 enable green -> blue)+
				16(spital effect1 enable gray -> read)+
				32(remove (UV) in YUV color system)+
				64(enable negative effect)+
				128(remove (Y) in YUV color system)  ;
write_SCCB(0xff, 0x00);
write_SCCB(0x7c, 0x00);
write_SCCB(0x7d,enable_effect);

command 
write_SCCB(0x7c, ADRESS);
Preparing the address for data injection

ofter that use
The following command to inject data
write_SCCB(0x7d,byte_0);
write_SCCB(0x7d,byte_1);
write_SCCB(0x7d,byte_2);
.
.
.

The memory structure is as follows
byte_0 =enable_effect  
         byte_0->bit0  =saturation and hue 
         byte_0->bit1  =saturation enable
         byte_0->bit2  =brighnes and contrast enable
         byte_0->bit3  =green -> blue spitial effect (Antique and blunish and greenish and readdish and b&w) enable
         byte_0->bit4  =gray  -> read spitial effect (Antique and blunish and greenish and readdish and b&w) enable
         byte_0->bit5  =remove (UV) in YUV color system
         byte_0->bit6  =negative enable
         byte_0->bit7  =remove (Y) in YUV color system
byte_1 =saturation1  0-255
byte_2 =hue 0-255
byte_3 =saturation2  0-255
byte_4 =reenter saturation2 in documents
byte_5 =spital effect green -> blue   0-255
byte_6 =spital effect gray  -> read   0-255
byte_7 =contrast uper  eadge  0-255
byte_8 =contrast lower eadge  0-255
byte_9 =brighnes 0-255
byte_10= if byte_10==4 contrast effective 

mosazahany@gmail.com

با تشکر

VojtechBartoska, 2022-03-30:
Hello, adding this to Roadmap so we will investigate this more and come with some resolution.

VojtechBartoska, 2022-04-07:
Hello, did you try to test your code on some Espressif device? It really looks like a HW issue and we are not able to help to replicate all thrid party boards issues.

Derek-K, 2022-04-23:
I just googled the same issue like many other has, and yes, I agree it is somewhat a hardware related issue. But wrapping the cam connector doesn't seem to be an ideal solution because it also blocks the WiFi signal as well, I do think somewhere it is the high-frequency signal/clock having too much interference.

rogerdahl, 2022-05-18:
@Mungbeanz @andreondra When you put aluminum foil on the onboard antenna, do you have to connect an external antenna in order to connect with WiFi?

aguaviva, 2022-05-18:
It is a HW issue, one of the pins carries a high frequency clock signal that is causing intereferences with everything. I tried to shield it putting some foil and it worked. Sorry for the lame description :)

VojtechBartoska, 2022-05-18:
I'm closing this one as HW issue, if needed please reopen it.

closed 2022-08-09 by VojtechBartoska

xam-ps, 2023-01-07:
Setting the clock speed (xclk) to 8 Mhz as described by @sohtamei did the job for me. Now getting constantly good framerates.

MortyFleem, 2023-04-18:
I just moved my board closer to the router. Voila! Strongly suspect signal issue.

krupis, 2023-07-11:
I have tested the ESP32 Cam with the webserver on ESP32-S2-Kaluga-1 and I am having really poor performance. 
Please note that I am using esp-idf and already creating esp-idf git ticket for this issue:
https://github.com/espressif/esp-idf/issues/11849

I am getting about 0.5 - 1fps when trying to stream video to the webserver and cannot figure out what could be an issue.

krupis, 2023-07-11:
> I recommend reading the last messages in this thread.

I did. Setting anything other than 16MHZ will not work.
```
E (2105) cam_hal: cam_dma_config(315): DMA buffer 32000 Byte malloc failed, the current largest free block:31744 Byte
E (2115) cam_hal: cam_config(384): cam_dma_config failed
E (2125) camera: Camera config failed with error 0xffffffff
E (2125) esp32-cam-test: Camera Init Failed
```

esp-idf suggests setting it to 16MHZ to enable EDMA.

With 16MHZ set, I am getting crazy bad performance:
```
I (2025) camera: Detected camera at address=0x30
I (2025) camera: Detected OV2640 camera
I (2025) camera: Camera PID=0x26 VER=0x42 MIDL=0x7f MIDH=0xa2
I (2085) s2 ll_cam: node_size: 3200, nodes_per_line: 1, lines_per_node: 2
I (2085) cam_hal: buffer_size: 960000, half_buffer_size: 480000, node_buffer_size: 3200, node_cnt: 300, total_cnt: 2
I (2085) cam_hal: Allocating 960064 Byte frame buffer in PSRAM
I (2095) cam_hal: Frame[0]: Offset: 64, Addr: 0x3F8008C0
I (2105) cam_hal: Allocating 960064 Byte frame buffer in PSRAM
I (2105) cam_hal: Frame[1]: Offset: 32, Addr: 0x3F8EAF00
I (2115) cam_hal: cam config ok
I (2115) ov2640: Set PLL: clk_2x: 1, clk_div: 7, pclk_auto: 1, pclk_div: 8
I (2175) esp32-cam-test: ESP32 CAM Web Server is up and running

I (5725) esp32-cam-test: MJPG: 47KB 981ms (1.0fps)
I (6655) esp32-cam-test: MJPG: 35KB 926ms (1.1fps)
I (9455) esp32-cam-test: MJPG: 101KB 2803ms (0.4fps)
I (16095) esp32-cam-test: MJPG: 80KB 6638ms (0.2fps)
I (18705) esp32-cam-test: MJPG: 85KB 2609ms (0.4fps)
```