espressif/arduino-esp32 #4660: unused parameter in BLERemoteService::getCharacteristics()

cyberman54, 2020-12-26:
My compiler complains on a unused, but set, parameter in `BLERemoteService::getCharacteristics()`:

`*pCharacteristicMap = m_characteristicMapByHandle; `
ist set, but neither used in the function, nor returned by it.

Maybe shall this function have a return value, other than void()?

/cc @kolban

chegewara, 2020-12-26:
I dont remember now why there is 2 `getCharacteristics()`, but you can use this one:
https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLERemoteService.cpp#L217

cyberman54, 2020-12-28:
@chegewara thanks for the hint! Would be nice to delete the redundant function, if not needed (where?), to get rid of the compiler warning.

chegewara, 2020-12-28:
Feel free to make PR, for example add this in function with warning:
`(void)pCharacteristicMap`

cyberman54, 2020-12-28:
I am careful to make changes here, as long as i don't know sure what i am doing in this code.

chegewara, 2020-12-28:
It is easy change, you can try it in your code. That change does nothing in code except is fooling compiler.

cyberman54, 2020-12-29:
issue is now adressed by PR #4669

closed 2021-07-08 by stale[bot]