espressif/arduino-esp32 #4666: Regression issue with WiFiClientSecure (timeout related?) with large files

ambanmba, 2020-12-29:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.5rc-4
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                     No
Upload Speed:					921600
Computer OS:                                         macOS 11 (Big Sur)

### Description:
Needed to upgrade to 1.05.rc-4 to fix the Big Sur compatibility issue with esptool.py, but then Universal Telegram Bot started failing when sending larger (> 100K) images to Telegram. The debug trace showed the upload failing with:

[V][ssl_client.cpp:289] send_ssl_data(): Writing HTTP request with 0 bytes....
[V][ssl_client.cpp:293] send_ssl_data(): Handling error -26752
[V][ssl_client.cpp:36] _handle_error(): [send_ssl_data():294]: (-26752) SSL - Connection requires a write call
[V][ssl_client.cpp:259] stop_ssl_socket(): Cleaning SSL connection.

Reverting back to the version of WiFiClientSecure that came with 1.0.4 (dated 2-Oct-19) restores functionality and sending images no longer fails.

### Sketch:  (

Refer the sendPhotoByBinary function here with photos that are > ~100k https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot

### Debug Messages:

timkoers, 2021-01-12:
I've got the same problem, not with the same code, but with PubSubClient.
The issue is only present when I decrease the logging level to something that is not verbose. When I set the logging level to verbose (-DCORE_DEBUG_LEVEL=5), the issue is not here.

FBMinis, 2021-01-31:
**Hardware:**
Board: ESP32-Cam
Core Installation version: 1.0.5rc-4
IDE name: Arduino IDE 1.8.5
Computer OS: Win10

**Description:**

My Arduino sketches that involved communication between the ESP-32Cam and Telegram stopped working once I began using dev 1.05.rc-6. 
They worked perfectly with 1.04 stable.

I noticed they were failing to connect with telegram api, so I tried adding `client.setInsecure();` , whis is present in 1.05.rc-6 version of WifiClientSecure . This partially solved the problem: I was able to turn On a GPIO through Telegram but I couldnt retrieve a photo (it would never arrive even after 'done sending' message).

**_Reverting back to the version of WiFiClientSecure that came with 1.0.4 (dated 2-Oct-19) restores functionality and sending images no longer fails._** 

This worked as far as running the ESP32-Cam example in https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot

However, the code found in this tutorial still doesnt work, Im not able to comunicate with the board by sending requests from Telegram (again, the same setup works fine in 1.04) https://randomnerdtutorials.com/telegram-esp32-cam-photo-arduino/

They should do the same thing: text /flash to light the LED, text /photo to request a photo from the AI Thinker ESP32-cam.

This the the code from Universal Telegram Bot:
```
/*******************************************************************

   A Telegram bot for taking a photo with an ESP32Cam

   Parts used:
   ESP32-CAM module* - http://s.click.aliexpress.com/e/bnXR1eYs

    = Affiliate Links

   Note:
   - Make sure that you have either selected ESP32 Wrover Module,
           or another board which has PSRAM enabled
   - Choose "Huge App" partion scheme

   Some of the camera code comes from Rui Santos:
   https://randomnerdtutorials.com/esp32-cam-take-photo-save-microsd-card/

   Written by Brian Lough
    YouTube: https://www.youtube.com/brianlough
    Tindie: https://www.tindie.com/stores/brianlough/
    Twitter: https://twitter.com/witnessmenow
*******************************************************************/

// ----------------------------
// Standard Libraries - Already Installed if you have ESP32 set up
// ----------------------------

#include <WiFi.h>
#include <WiFiClientSecure.h>
#include <UniversalTelegramBot.h>
#include "esp_camera.h"
#include <ArduinoJson.h>
//#include "camera_pins.h"
//#include "camera_code.h"

//define CAMERA_MODEL_AI_THINKER
#define PWDN_GPIO_NUM 32
#define RESET_GPIO_NUM -1
#define XCLK_GPIO_NUM 0
#define SIOD_GPIO_NUM 26
#define SIOC_GPIO_NUM 27
#define Y9_GPIO_NUM 35
#define Y8_GPIO_NUM 34
#define Y7_GPIO_NUM 39
#define Y6_GPIO_NUM 36
#define Y5_GPIO_NUM 21
#define Y4_GPIO_NUM 19
[... 482 lines cut ...]
    Serial.print(".");
    delay(500);
  }
  Serial.println();
  Serial.print("ESP32-CAM IP Address: ");
  Serial.println(WiFi.localIP()); 
}

void loop() {
  if (sendPhoto) {
    Serial.println("Preparing photo");
    sendPhotoTelegram(); 
    sendPhoto = false; 
  }
  if (millis() > lastTimeBotRan + botRequestDelay)  {
    int numNewMessages = bot.getUpdates(bot.last_message_received + 1);
    while (numNewMessages) {
      Serial.println("got response");
      handleNewMessages(numNewMessages);
      numNewMessages = bot.getUpdates(bot.last_message_received + 1);
    }
    lastTimeBotRan = millis();
  }
}
```

johnwalicki, 2021-03-13:
My ESP32 is also reporting this error with WiFiClientSecure and PubSubClient sending MQTT.  
```[ssl_client.cpp:36] _handle_error(): [send_ssl_data():294]: (-26752) SSL - Connection requires a write call```

With the recent version of ssl_client.cpp, it works for small buffers, but fails on larger buffers.  Prior version of ssl_client.cpp  works with all the buffers sizes I give it.

The regression seems to be in ssl_client.cpp int send_ssl_data()
The while() loop works, the if() fails.

```
$ diff ssl_client.cpp.prior ssl_client.cpp.latest
293,296c293,297
<     while ((ret = mbedtls_ssl_write(&ssl_client->ssl_ctx, data, len)) <= 0) {
<         if (ret != MBEDTLS_ERR_SSL_WANT_READ && ret != MBEDTLS_ERR_SSL_WANT_WRITE) {
<             return handle_error(ret);
<         }
---
>     if ((ret = mbedtls_ssl_write(&ssl_client->ssl_ctx, data, len)) <= 0){
>         log_v("Handling error %d", ret); //for low level debug
>         return handle_error(ret);
>     } else{
>         log_v("Returning with %d bytes written", ret); //for low level debug
456a458
> 
```

johnwalicki, 2021-03-13:
Seems to fixed in this commit
https://github.com/espressif/arduino-esp32/commit/560c0f45f58b907f0d699f65408b87fe54650854#diff-1294bfb831e132940639f2b2768d1444395a9d982ff45c2e7b4b5f73e4e99d6f

jameszah, 2021-03-27:
This might be the same issue with mbedtls_ssl_write() -- not necessarily a bug, but different behavior.

client->println(F(""));   which used to work, needs to be changed to:   client->println();

Its purpose was to send another cr-lf (2 bytes), but what it was actually doing was sending a 0 byte message, then the 2 byte message, but ssl_client.cpp, which calls mbedtls_ssl_write(), declared a fatal error after the 0 byte message.

19:06:57.685 -> *[V][ssl_client.cpp:295] send_ssl_data(): Writing HTTP request with 0 bytes...
19:06:59.279 -> [V][ssl_client.cpp:300] send_ssl_data(): Handling error -80
19:06:59.279 -> [E][ssl_client.cpp:36] _handle_error(): [send_ssl_data():301]: (-80) UNKNOWN ERROR CODE (0050)
19:06:59.279 -> [V][ssl_client.cpp:265] stop_ssl_socket(): Cleaning SSL connection.
19:06:59.314 -> **Start request: --------------------------b8f610217e83e29b

https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot/issues/235#issue-842397567

closed 2021-07-11 by stale[bot]