espressif/arduino-esp32 #4667: Problems with flashing - reboot loop and different stability issues with same code on different ESP32s

DiscrafT, 2020-12-29:
### Hardware:
I'm using ESP32 Wrover, which is the MCU for my own custom Nixie Clock PCB, so no breakout boards. For programming, I'm on windows and using Arduino IDE. For programming, I'm using an FDTI breakout along with one button for GPIO 0 and one for Reset (both have 10k pull-up resistors) With that setup I'm able to flash the ESP without any problems.
[image]

### Description:
I have recently received my new boards for the Nixie Tube clock which I'm making. So I wrote a code which was working for some time but after a time I started to receive a Guru meditation error which I would like to decode with EspExceptionDecoder but I cannot right now because I'm stuck with a reboot loop and I'm not able to flash any code to ESP32 because it reboots before the uploading is done. Anyway, the code was working for some time totally fine (let's say about 1 hour and multiple restarts) but after that, I started receiving error so I have put on that ESP an easy piece of code for diagnostics and after some time I ended up with this reboot loop. I took another ESP32 brand new and I have put both codes on it and they work totally fine. For that Guru meditation error I was able to find by removing parts of code and libs that the problem is in NTPClient.h while setup during initialization: timeClient.begin();
After some troubleshooting, I have also found and Improved NTP library for ESP32 but it was doing the same thing. But everything was working fine on that other ESP32 as I said before.
This is not the first time I have problems with let's say consistent and stable development on ESP32 boards. So I would really like to know what I'm doing wrong or what should I change to make it work properly and most important stable.

Is there any way how can I flash at least BareMinimum example to it or anything else? 
Thank you in advance.

PS: I'm sorry if this issue will be not correctly formatted or posted incorrectly but this is the first time I'm writing on GitHub issues page. 

### Here I tried to put ESP32 into programm mode by pressing the prog button even before I connected the power to the board:
```
06:56:05.806 -> 
06:56:05.806 -> rst:0x10 (RTCWDT_RTC_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
06:56:05.806 -> flash read err, 1000
06:56:05.806 -> ets_main.c 371 
06:56:06.144 -> ets Jun  8 2016 00:22:57
06:56:06.144 -> 
06:56:06.144 -> rst:0x10 (RTCWDT_RTC_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
06:56:06.144 -> flash read err, 1000
06:56:06.144 -> ets_main.c 371 
06:56:06.482 -> ets Jun  8 2016 00:22:57
06:56:06.482 -> 
06:56:06.482 -> rst:0x10 (RTCWDT_RTC_RESET),boot:0x23 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
06:56:06.482 -> waiting for download
06:56:35.220 -> ⸮
```
### Arduino IDE output while uploading:
```
Sketch uses 206609 bytes (15%) of program storage space. Maximum is 1310720 bytes.
Global variables use 14564 bytes (4%) of dynamic memory, leaving 313116 bytes for local variables. Maximum is 327680 bytes.
esptool.py v2.6
Serial port COM7
Connecting......
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, VRef calibration in efuse, BLK3 partially reserved, Coding Scheme 3/4
MAC: 30:ae:a4:6a:f4:d0
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0x0, SizeID=0x0), defaulting to 4MB
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 4096.0 kbit/s)...

A fatal error occurred: Timed out waiting for packet header
A fatal error occurred: Timed out waiting for packet header

```
### Working code output on another ESP32:
[image]

chegewara, 2020-12-29:
Please try with lower flash speed, optimal is 115200 in case you have problems with flash.

DiscrafT, 2020-12-29:
Thank you @chegewara for your response but even with a lower upload speed, I have the same problem. The ESP32 reboots while I'm trying to write a code to it and it fails I think.
[image]

chegewara, 2020-12-29:
Ok, that looks to me like issue with bootstrap pin that is responsible for flash voltage.
https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection

DiscrafT, 2020-12-29:
Thank you for that interesting link. I spend quite a while reading it and also checking my ESP32 but unfortunately, I should have connected everything correctly. 
**I have this error message on boot:**
`06:56:05.806 -> rst:0x10 (RTCWDT_RTC_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)`
So according to your link, it should be the 0x10 - GPIO0 which during the normal execution mode should be VCC/HIGH/3.3V

And also 
> GPIO0 has an internal pullup resistor, so if it is left unconnected then it will pull high.
So this should not make any problems, but anyway I have a bigger 10k ohm resistor connected to GPIO 0 and 3.3V

In this photo, you can see the voltage while running on the GPIO 0 pin
[image]

This is the voltage while pressing the program button
[image]

I also tried to erase flash with esptool.py according to this page [https://github.com/espressif/esptool#erase-flash-erase_flash--erase-region](url)
But also with the same fatal error...
[image]

chegewara, 2020-12-29:
GPIO12
>If driven High, flash voltage (VDD_SDIO) is 1.8V not default 3.3V. Has internal pull-down, so unconnected = Low = 3.3V. May prevent flashing and/or booting if 3.3V flash is used and this pin is pulled high, causing the flash to brownout. See the ESP32 datasheet for more details.

chegewara, 2020-12-29:
in your case important is `boot:0x33 (SPI_FAST_FLASH_BOOT)`:
https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection#boot-mode-message

chegewara, 2020-12-29:
Now ignore my last 2 posts. You are using standalone module, so you have to enter download mode with buttons, since you dont have hardware that will do it auto-magically.

DiscrafT, 2020-12-29:
I just tried to connect GPIO 12 to 3.3V and also to GND and boot from it but in both cases, it fails. Also according to my boot message, `SPI_FAST_FLASH_BOOT` ESP should be starting normally, but there is still a problem with the reboot loop and also that I'm not able to flash anything into it or erase its memory :/

chegewara, 2020-12-29:
> 06:56:05.806 -> flash read err, 1000
> 06:56:05.806 -> ets_main.c 371 
> 06:56:06.144 -> ets Jun  8 2016 00:22:57

This is because flash is empty, so boot loop occurring.

> You are using standalone module, so you have to enter download mode with buttons, since you dont have hardware that will do it auto-magically.

This is your issue i think.

DiscrafT, 2020-12-29:
> You are using standalone module, so you have to enter download mode with buttons, since you dont have hardware that will do it auto-magically.

I know, thats the reason why I have put those buttons on the board while I was designing it. So when I want to flash it I have to release reset while loading prog. button. And this procedure was working many times.
If I do so, the rebooting loop stop and there is that download message
`06:56:06.482 -> rst:0x10 (RTCWDT_RTC_RESET),boot:0x23 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2)) 06:56:06.482 -> waiting for download`
But even when it is waiting for download the upload process will fail as I wrote in the first post.

> This is because flash is empty, so boot loop occurring.

You are maybe right because that means that bootloader which loads the 0x1000 from memmory doesnt have anything to read but what can I do about it when I cannot put anything on it?

PS Thank you for your suggestions :)

chegewara, 2020-12-29:
Please try one more thing. I am having few devkit boards that should enter download mode when i am using arduino ide, but some just does not. In that case i just have to keep pressed IO0 button until flashing starts. It is annoying but after some time i just used to it.

kubo6472, 2020-12-29:
> that should, but some just does not

So you think this may be a manufacturing error?
> the code was working for some time totally fine, but after that, I started receiving error

So it worked, and then stopped working for no apparent reason, other than changes in code, right?

DiscrafT, 2020-12-29:
As you can see here, while I'm holding it during `Connecting......` part, it successfully starts a flash process as you can see in the lines below. If the FTDI was not able to communicate with ESP32 I would just end up with timeout after the long `Connecting......` part
[image]
Here you can also see that I have succesfully started the upload 
[image]
Anyway I had no problem with uploading the code into ESP, just running it then without errors.

kubo6472, 2021-06-23:
I assume this is still not fixed @DiscrafT

kubo6472, 2021-08-28:
I assume this is still not fixed

VojtechBartoska, 2022-04-13:
Hello, 

as there was no answer in last 14 days, I'm **closing the issue as expired** to keep our backlog manageable. If it's still needed, please reopen the issue.

Thanks for understanding.

closed 2022-04-13 by VojtechBartoska

kubo6472, 2022-04-13:
Apparently some workaround was used. Thanks.