espressif/arduino-esp32 #4668: Sketch using both BLE and WiFi exceeds available space

tko, 2020-12-29:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Flash Frequency:					80MHz
PSRAM enabled:                                    no
Upload Speed:					921600
Computer OS:                                        macOS Mojave 10.14.6

### Description:

I'm trying to build a thing to forward BLE announcements to an HTTP endpoint in WiFi network. However, just combining minimal BLE and WiFi code already exceeds available space. Compiling fails with error:
```
Sketch uses 1311266 bytes (100%) of program storage space. Maximum is 1310720 bytes.
text section exceeds available space in board
Global variables use 57716 bytes (17%) of dynamic memory, leaving 269964 bytes for local variables. Maximum is 327680 bytes.
Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.
Error compiling for board ESP32 Dev Module.
```

(My first attempt in firmware / arduino, not too familiar how restrictive the environment really is but this is feeling like hitting the limits much too quickly.)

### Sketch:
```cpp
#include <BLEAdvertisedDevice.h>
#include <BLEDevice.h>
#include <BLEScan.h>
#include <BLEUtils.h>
#include <WiFi.h>

void setup() {
  Serial.begin(115200);
  WiFi.begin("", "");
  BLEDevice::init("");
  auto scan = BLEDevice::getScan();
  auto result = scan->start(0);
}
void loop() {
}
```

chegewara, 2020-12-29:
https://github.com/espressif/arduino-esp32/issues?q=is%3Aissue+of+program+storage+space.+Maximum+is+is%3Aclosed+ble+wifi

h2zero, 2021-01-05:
As an alternative solution I welcome you to checkout my library here:

https://github.com/h2zero/NimBLE-Arduino

It uses half of the flash and memory of the built-in BLE library.

Very few changes to make to your code as the classes are all aliased to the built in BLE library. For the most part it's a just a change to the include header.

closed 2021-07-08 by stale[bot]

Rhodion, 2022-01-20:
https://www.youtube.com/watch?v=5VoXNloOwZE&ab_channel=XTronical
case you are on mac (caso esteja no mac) /Users/$(whoami)/Projects/bt4ml/data_manager_page