espressif/arduino-esp32 #4670: How to set security mode and level

sasodoma, 2020-12-29:
I am trying to read the HID report map from my wireless mouse. I have previously opened issue #4351 about this, but while waiting for my nRF dongle to arrive, the stale bot closed that issue. I have now received the dongle. I installed nRF Connect on Windows and connected to my mouse. When trying to read the report map I get "BLE_GATT_STATUS_ATTERR_INSUF_AUTHENTICATION". If I then pair the mouse, the security gets updated to mode 1 and level 2. After that I can read the report map sucessfully. I'm guessing that in order to read the report map with my ESP I need to somehow raise the security level to 2. Is that something that can be done?

chegewara, 2020-12-29:
`BLE_GATT_STATUS_ATTERR_INSUF_AUTHENTICATION` - that means you have to connect with secure/authenticated connection.

sasodoma, 2020-12-29:
Apparently mode 1, level 2 (Unauthenticated pairing with encryption) is enough, because after updating security to level 2 nRF Connect can read the report. How do I mirror this level of security with ESP32 BLE?

chegewara, 2020-12-29:
No idea, maybe ask on esp-idf github.

sasodoma, 2020-12-29:
Ok I will ask there and report back if I find a solution.

chegewara, 2020-12-29:
If i remember i could read report map with esp32, but it was long time ago, when ive been trying to write simple HID client app (but i may be wrong now). At least its been working with 2 x esp32, because i could not find ble keyboard.

sasodoma, 2020-12-29:
I think that this is a security feature of the mouse. You need to pair to it and setup an encrypted connection so that others can't eavesdrop and record your actions. Is there an example on how to use BLESecurity from the Arduino library? I feel like that would be a good starting point.

chegewara, 2020-12-29:
https://github.com/nkolban/esp32-snippets/tree/master/cpp_utils/tests/BLETests/Arduino/security

sasodoma, 2020-12-29:
Thank you! It turns out that the only thing I needed was to add `BLEDevice::setEncryptionLevel(ESP_BLE_SEC_ENCRYPT);` before the createClient command. I am now able to read the report map correctly. I will also add a comment to the other issue in case anyone finds it.

closed 2020-12-29 by sasodoma