espressif/arduino-esp32 #4675: What is the maximum file size of Esp32?

miracle0419, 2020-12-30:
Hi,
I using ESP32 / 4Mb(32Mb) : No OTA (2MB APP/2MB SPIFFS).
I trying that files over 1.5M are not saved.  Since the file was formatted before saving, there is more than 1.8M space available.

-------------------------------------------------------------------------
one more question please let me know if you have any idea.
I can manually modify the partition scheme?
Currently, our application binary size is about 1.5M  and I would like to implement OTA.

Thanks, ck

lbernstone, 2020-12-30:
This would be filesystem dependent.  However, both SPIFFS and FAT12 can handle 32MB files. SPIFFS.totalBytes() - SPIFFS.usedBytes() = free space.  Verbose logging will tell you if the disk runs out of space.  Writing 1.5MB will take over a minute, so the watchdog timer will be triggered unless you turn it off.
https://github.com/espressif/arduino-esp32/issues/4551#issuecomment-729902674

miracle0419, 2020-12-30:
I not sure,  only 1.5 can be saved. flash wite value keeps returning 0.

[30/12/20 - 11:26:33:685] V: SPIFFS DATA USAGE TOTAL : [1860161], USAGE : [1512024]
[30/12/20 - 11:26:33:689] V: Listing directory:
[30/12/20 - 11:26:33:893] V:   FILE: /esp_fw.bin        SIZE:1499520

closed 2021-07-08 by stale[bot]