espressif/arduino-esp32 #4676: Calling touch_pad_set_meas_time() function makes the core restart

saeedbala, 2020-12-30:
### Hardware:
|||||||
|:---|---|---|---|---|---|
|<B>Board</B>|node32|
|<B>Version/Date</B>|1.0.4|
|<B>IDE name</B>|Arduino IDE|
|<B>Flash Frequency</B>|80Mhz|
|<B>PSRAM enabled</B>|yes|no|
|<B>Upload Speed</B>|115200|
|<B>Computer OS</B>|Windows 10|

### Description:
Hi guys,
I am trying to make adjustments to the touch function parameters by using touch_pad_set_meas_time() function, but calling the function makes the core SWRESTART. It seems feeding the function deferent values, including the values you get by touch_pad_get_meas_time makes no defference.(256 for sleep,4096 for measure time).

### Sketch:  
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup()
{
  Serial.begin(115200);
  delay(1000); // give me time to bring up serial monitor
  Serial.println("ESP32 Touch Test");
  uint16_t sleep_cycle;
  uint16_t meas_cycle;
  touch_pad_get_meas_time(&sleep_cycle,&meas_cycle);

  Serial.println("sleep_cycle:");
  Serial.println(sleep_cycle);
  Serial.println("meas_cycle:");
  Serial.println(meas_cycle);
  touch_pad_set_meas_time(sleep_cycle,meas_cycle);

  
}

void loop()
{
  Serial.println(touchRead(T0));  // get value using T0
  delay(200);
}
```

### Debug Messages:
```
rst:0xc (SW_CPU_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
ESP32 Touch Test
sleep_cycle:
256
meas_cycle:
4096
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:1442 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x40087f59 on core 1

Backtrace: 0x4008b5b8:0x3ffb1ed0 0x4008b7e5:0x3ffb1ef0 0x40087f59:0x3ffb1f10 0x400e754f:0x3ffb1f50 0x400d0c3e:0x3ffb1f70 0x400d15c7:0x3ffb1fb0 0x4008826d:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

```

lbernstone, 2020-12-30:
https://github.com/espressif/arduino-esp32#decoding-exceptions
https://docs.espressif.com/projects/esp-idf/en/release-v3.3/api-reference/peripherals/touch_pad.html

saeedbala, 2020-12-30:
> https://github.com/espressif/arduino-esp32#decoding-exceptions
> https://docs.espressif.com/projects/esp-idf/en/release-v3.3/api-reference/peripherals/touch_pad.html

Updated main post with the stack-trace

lbernstone, 2020-12-30:
You need to init before you can change the setting.

saeedbala, 2020-12-30:
> You need to init before you can change the setting.
That was it! Thanks!

closed 2020-12-30 by saeedbala