espressif/arduino-esp32 #4677: Ultra slow downloads from dl.espressif.com

naster01, 2020-12-30:
I'm trying to install the board manager for ESP32 from 
`https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json` but all tools point to `dl.espressif.com` witch is ultra slow. It will take for ever to install the board manager.

Could you please upload all files that points to  `dl.espressif.com` to GitHub?
Example of files:
```
https://dl.espressif.com/dl/esptool-2.6.1-windows.zip
https://dl.espressif.com/dl/esptool-2.6.1-macos.tar.gz
https://dl.espressif.com/dl/esptool-2.6.1-linux.tar.gz
```

ChR-iSz, 2021-01-11:
Same here. Its a nightmare! 4kb/s !

ChrisAFRcrew, 2021-01-18:
Arduino is giving an error message about this site "https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json " 'Error Download' 
 When I clicked on the link "https://github.com/espressif/arduino-esp32/releases/download/1.0.5-rc4/esp32-1.0.5-rc4.zip" nothing happens

closed 2021-07-08 by stale[bot]