espressif/arduino-esp32 #4679: ESP32-cam

iabesadze, 2020-12-31:
Hello,

I using this tutorial [https://robotzero.one/esp32-face-door-entry/](url)

I flash camerawebserver

```esptool.py v2.6
Serial port COM10
Connecting........_____....._____....._____....._
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 10:52:1c:5c:a6:a8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 4096.0 kbit/s)...
Hash of data verified.
Compressed 17392 bytes to 11186...
Wrote 17392 bytes (11186 compressed) at 0x00001000 in 0.1 seconds (effective 966.2 kbit/s)...
Hash of data verified.
Compressed 2108368 bytes to 1665844...
Wrote 2108368 bytes (1665844 compressed) at 0x00010000 in 22.6 seconds (effective 745.8 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 149...
Wrote 3072 bytes (149 compressed) at 0x00008000 in 0.0 seconds (effective 1638.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

After flash

I get error in terminal

```ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled

[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
.[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.100.9, MASK: 255.255.255.0, GW: 192.168.100.1
.
WiFi connected

assertion "l->size == size" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/esp-face/face_recognition/fr_flash.c", line 85, function: read_face_id_from_flash
abort() was called at PC 0x40109b83 on core 1

Backtrace: 0x40091448:0x3ffb1db0 0x40091679:0x3ffb1dd0 0x40109b83:0x3ffb1df0 0x401199f6:0x3ffb1e20 0x400d2ac8:0x3ffb1e60 0x400d18fd:0x3ffb1f10 0x400d4d73:0x3ffb1fb0 0x4008db91:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57
```

if i remove `read_face_id_from_flash(&id_list);` line in app_httpd.cpp sketch work fine.

Please help
Sorry bad english

iabesadze, 2020-12-31:
Very sorry friends
I chosen bad board
my true board is ai-thinker

closed 2021-07-08 by stale[bot]

zubairadil, 2023-02-15:
I am also having this same issue, any idea on how to solve this?

I have tried all possible solutions, but still not solved.