espressif/arduino-esp32 #4690: FEATHERS2 - cannot get LITTLEFS_test example working

mcc94704, 2021-01-06:
### Hardware:
Board:							FeatherS2 - ESP32-S2 Feather Development Board
Core Installation version:			?
IDE name:						Arduino IDE
Flash Frequency:					?
PSRAM enabled:                                    yes
Upload Speed:					921600
Computer OS:                                        Windows 10

### Description:
I could not get the LITTLEFS_test example working with the FeatherS2 - ESP32-S2 Feather Development Board in the Arduino IDE. It just says 'LITTLEFS Mount Failed'. It works fine on the original Adafruit HUZZAH32 Feather.  I can run other test code on the Feather S2 board.

I don't know what my core version is.  However, I am using the "idf-release/v4.2" from the git repository, per instructions from:
https://learn.adafruit.com/adafruit-metro-esp32-s2/arduino-ide-setup
for a similar ESP32-S2 board.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#if !( defined(ESP32) )
  #error This code is intended to run on the ESP32 platform! Please check your Tools->Board setting.
#endif
#include <FS.h>
#include <LITTLEFS.h> 
#define FileFS        LITTLEFS
#define FS_Name       "LittleFS"
FS* filesystem =      &LITTLEFS;

void setup() {
  // put your setup code here, to run once:
  delay(2000);

  FileFS.begin();
  FileFS.format();
  Serial.print(FS_Name);
  if (FileFS.begin(true)) {
    Serial.println(F(" successfully formatted. [1]"));
  }
  else {
    Serial.println(F(" failed formatting. [2]"));
  }
  while (true);
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

### Debug Messages:
I cannot get any messages from the Serial Monitor of Arduino IDE.  I can monitor serial port using PuTTY, which only gives the message:
```
LittleFS failed formatting. [2]
```

lbernstone, 2021-01-06:
Make sure to choose the default 4MB with spiffs partition scheme.

mcc94704, 2021-01-06:
Thank you very much @lbernstone . LITTLEFS_test example works with this setting.

closed 2021-01-06 by mcc94704