espressif/arduino-esp32 #4691: Arduino esp32 core cannot be used by other platform

vroland, 2021-01-06:
### Hardware:
Board:							other
Core Installation version:			1.0.4/master
IDE name:						Arduino IDE
Flash Frequency:					-
PSRAM enabled:                                    -
Upload Speed:					-
Computer OS:                                        All

### Description:
I'd like to specify custom boards. According to the [Arduino platform specification](https://arduino.github.io/arduino-cli/latest/platform-specification/), I should be able to do refer to the esp32 platform for the build instructions (platform.txt) and only add additional board definitions (board.txt).
However, the platform.txt file of the arduino-esp32 core assumes `{runtime.platform.path}` its its own path instead of a derived platform. See also a similar issue with the esp8266 core: https://github.com/esp8266/Arduino/issues/5769.
Would using `{build.core.path}/../..` instead of `{runtime.platform.path}` be a possibility here?

### Sketch:  not relevant

solartech77, 2021-01-09:
I want to build voltmeter using esp32 to display on my phone or a web browser?

settinger, 2021-07-05:
The inability to define custom boards is a real nuisance. Is the issue blocked by this bug in arduino-cli? https://github.com/arduino/arduino-cli/issues/1031

jandrassy, 2021-07-05:
> The inability to define custom boards is a real nuisance. Is the issue blocked by this bug in arduino-cli? [arduino/arduino-cli#1031](https://github.com/arduino/arduino-cli/issues/1031)

in classic Arduino IDE the upload is still done by the IDE, not by the CLI. and the IDE doesn't know the paths which should be used in the esp32 (and esp8266) upload tool location to support referencing platforms or custom boards..

in IDE 2.00 custom boards definitions are not supported yet.

settinger, 2021-07-06:
> in classic Arduino IDE the upload is still done by the IDE, not by the CLI. and the IDE doesn't know the paths which should be used in the esp32 (and esp8266) upload tool location to support referencing platforms or custom boards..

Thank you for the info! I am unclear why, in issue https://github.com/esp8266/Arduino/issues/5769 , your PR was not sufficient. Specifically:
> the solution was to use `{build.core.path}/../..` instead of `{runtime.platform.path}` but in `git version` not all platform.txt keys can work with {build.core.path}.

I'm not sure what `git version` means here, or how it can trip up on certain platform.txt keys. Does the same issue affect the ESP32 platform.txt?

------

In case anyone else comes here looking for a workaround, here is how I have managed to add custom boards to the IDE boards menu. Tested with Arduino IDE v1.8.13 on Windows 10.

1) Ensure the ESP32-arduino platform is installed: https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/boards_manager.md
1) Locate the ESP32 platform in the Arduino installation folder. In my case it was located at `C:/Users/{name}/AppData/Local/Arduino15/packages/esp32/hardware/esp32/1.0.6`.
1) Create a new folder for your platform in the Arduino sketches folder as `Hardware/{your platform name}/esp32/`.
1) Create your `boards.txt` file in your platform folder. Be sure to set the following lines to reference the ESP32 platform: `{your board}.upload.tool=esp32:esptool_py`, `{your board}.build.core=esp32:esp32`. If you are reusing an existing variant's pinout, set the variant as `{your board}.build.variant=esp32:{variant}`. If you are not reusing an existing variant's pinout, you will need to define them in your platform folder as `variants/{your variant}/pins_arduino.h`.
1) Copy `platform.txt` from the ESP32 platform folder to your platform folder. Change the following lines:
    1) Change every instance of `{runtime.platform.path}` to `{build.core.path}\..\..`. On Windows it is necessary to use backslashes, not forward slashes, or else `recipe.hooks.prebuild.2.pattern.windows` will fail.
    1) Change the line `name=ESP32 Arduino` to your platform name.
1) I have not figured out a way to change the arguments in `tools.esptool_py.upload.pattern`. Therefore, copy the folders `tools/sdk/bin` and `tools/partitions` from the ESP32 platform folder to your platform folder. If you know which bootloader binaries and partition files you need, you can delete the unnecessary ones; if you're unsure, don't delete any of the folder contents.
1) Restart the Arduino IDE.

At that point, I was able to select my custom boards in the IDE board manager, compile, and upload without trouble. I hope someone finds this helpful.

jandrassy, 2021-07-06:
at the time of my PR in esp8266 Arduino, they changed to esptool.py. until then the esptool.exe was installed in external tools folder (arduino15/packages/esp8266/tools). the esptool.py then couldn't be configured for referring platform as I proposed in the PR and I closed the PR. 
esp32 arduino has esptool.py installed as external tool (arduino15/packages/esp32/tools/esptool_py).

jandrassy, 2021-09-07:
it is still a valid issue

JAndrassy, 2023-03-30:
> Is this still a valid issue? What would it take to resolve it?

yes it is still valid. 

The problem is that {runtime.platform.path} is evaluated to board platform so `{runtime.platform.path}/tools` doesn't work for referring package. 
A solution is to use `{build.core.path}/../..tools` in platform.txt. 
A clean solution would be to rename the `tools` folder to `system` and use `{build.system.path}` instead of  `{runtime.platform.path}/tools`.

https://arduino.github.io/arduino-cli/0.31/platform-specification/#referencing-another-core-variant-or-tool

egnor, 2024-05-28:
Is there any strong reason replacing `{runtime.platform.path} with `{build.core.path}/../..` wouldn't fix this? Would such a PR be accepted? Clearly it would be a version bump, and it would be incompatible with old versions of Arduino, but those versions are fairly old now I think? Or is there still an issue with tool execution?

JAndrassy, 2024-05-28:
@egnor I think espota.py would not work at leas in the IDE 1, because it is in tools and not installed as a tool
https://github.com/espressif/arduino-esp32/issues/4691#issuecomment-874458374

egnor, 2024-05-28:
And I take it IDE 1 compatibility is needed, even for arduino-esp32 3+?

me-no-dev, 2024-05-28:
We get requests to support 32bit Windows and old Linux, so what is left for IDE1 :) We need to support it all, even in 3.0. There are works being done to streamline this and maybe in near future we will make things better, but not yet possible

egnor, 2024-05-28:
OK I made a [specific proposal to change arduino-cli (and thus the IDE)](https://github.com/arduino/arduino-cli/issues/2616#issuecomment-2135866008) that SHOULD solve this problem without requiring any platforms sacrifice backward compatibility??

Parsaabasi, 2025-01-16:
Hello, 

Due to the overwhelming volume of issues currently being addressed, we have decided to close the previously received tickets. If you still require assistance or if the issue persists, please don't hesitate to reopen the ticket.

Thanks.

closed 2025-01-16 by Parsaabasi

JAndrassy, 2025-01-16:
this one is still valid

me-no-dev, 2025-01-16:
@JAndrassy change needs to come from the Arduino.cc team. We are also waiting to be able to do this