espressif/arduino-esp32 #4693: MBEDTLS doing Zeropadding by default, and not PKCS7

ortegafernando, 2021-01-06:
### Hardware:
Board:							LOLIN32
Core Installation version:			1.0.4
IDE name:						ArduinoIDE
Flash Frequency:					80Mhz
PSRAM enabled:                                     no
Upload Speed:					921600
Computer OS:                                         Windows 10

### Description:
Describe your problem here

What you are trying to do?:      Decrypt information recived by an MQTT payload

Describe what is failing. :    In ESP32 I have checked that mbedtls_aes_crypt_cbc use ZERO PADDING by DEFAULT. I don't know how to change it.

Show the shortest possible code that will duplicate the error.
I can managed to encrypt and decrypt the code inside ESP32 without any problem. But I have problems decrypting recieved data in the MQTT payload, So I have been testing all stuff.
I have even tried defining these inside my code (no solution with or without them): 
  #define MBEDTLS_CIPHER_MODE_WITH_PADDING 
  #define MBEDTLS_CIPHER_PADDING_PKCS7    

If we use, for example, this web page (https://the-x.cn/en-us/cryptography/Aes.aspx) or a nodered with nodejs and crypto-js library I get this:
1. ESP32 encrypted hex string is the same ONLY if in NodeJS or in that web page select ZERO PADDING.
2. ESP32 encrypted hex string is DIFFERENT if I use PKCS7 (both javascript (nodejs) and web page)

How could I change the padding to PKCS7 in my ESP32 ???

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```
  #define MBEDTLS_CIPHER_MODE_WITH_PADDING 
  #define MBEDTLS_CIPHER_PADDING_PKCS7  
  #include "mbedtls/aes.h";

 //ONLY PARTIAL CODE:
    mbedtls_aes_context aesCon;
    unsigned char *input = (unsigned char*) "XXXXXXXXX";
    unsigned char output[128] = {0};
    unsigned char output2[128] = {0};
    int resp;
    resp = mbedtls_aes_setkey_enc( &aes, aes_key, 16*8 );
    resp = mbedtls_aes_crypt_cbc( &aes, MBEDTLS_AES_ENCRYPT, strlen((const char*)input), iv, input, output );
    resp = mbedtls_aes_setkey_dec( &aes2, aes_key, 16*8 );
    resp = mbedtls_aes_crypt_cbc( &aes2, MBEDTLS_AES_DECRYPT, strlen((const char*)output), iv2, output, output2 );
    Serial.println((char *)input);
    printHexUC(input, strlen((const char*)input));   //this only prints in HEX format
    Serial.dprint("Output (");    Serial.print(sizeof(output));    Serial.print(") ;");
    printHex(output, sizeof(output)); //this only prints in HEX format
    Serial.println((char *)output2);
```
Output2 and input are equal
I compare output with the output of nodejs and web page I said. They are EQUAL if I select ZERO PADDING, they are different if I select PKCS7 padding.

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

ortegafernando, 2021-01-07:
Sorry, closed. Misunderstanding. But be carefull with strings, '\0' final char, and so on.

closed 2021-01-07 by ortegafernando

d1nz-g33k, 2021-08-08:
Hi, I'm facing the same issue, can you please tell me what you meant by 
> be carefull with strings, '\0' final char, and so on.