espressif/arduino-esp32 #4694: Weird behavior of Serial.println()

hmhuang0501, 2021-01-07:
I was writing two tasks both pinned to core 0 with TTGO LoRa ESP32. When I upload the sketch, the ESP32 started rebooting repeatedly....
The code is quite simple but I cannot figure out what's going on with my code. Here is the task code which might cause problem:

```cpp

xTaskCreatePinnedToCore( logWiFiDeviceData,                           // Task function
                                           "TaskForLoggingWiFiDeviceData",    // Task name
                                           10000,                                                // Stack size in words
                                           NULL,                                                 // Parameter passed as input to task
                                           0,                                                       // Task priority (0 is the lowest priority)
                                           NULL,                                                // Task handle
                                           0 );                                                    // Core ID
``` 
```cpp
void logWiFiDeviceData( void* parameter )
{
  
  combinedWifiDeviceData combinedData;
  std::vector<String> RSSIc(g_espConfigData.number_of_slaves, "");  // Initialize each element to a null string.
  for ( size_t i = 0; i < g_espConfigData.number_of_slaves; ++i )
  {   
    Serial.println( "RSSIc[" + String(i) + "] = " + '"' + RSSIc[i] + '"' );
  }

  wifiDevicePayload payload;

  for(;;)
  {
    for ( size_t i = 0; i < cg_queueSizeBetweenReceiveAndLog; ++i )
    {
      xQueueReceive( g_queueBetweenReceiveAndLog, &payload, portMAX_DELAY );

      //Serial.println( "1.Slave ID = " + String(payload.slaveID) );

      if ( payload.slaveID == 4 )  // check if the RSSI value received by respective slave has been filled in.      
      {                                                           
        Serial.println( "Slave ID = " + payload.slaveID ); 
      }

    vTaskDelay( 1 / portTICK_PERIOD_MS );
  }

} // logWiFiDeviceData
```

The struct is defined as:
```cpp
typedef struct
{
  char slaveID;

} __attribute__((packed)) wifiDevicePayload;
```

### Debug Messages:
```
RSSIc[0] = ""
RSSIc[1] = ""
RSSIc[2] = ""
RSSIc[3] = ""
MAC matched!
e ID =
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40001277  PS      : 0x00060b30  A0      : 0x800dbc14  A1      : 0x3ffd0d70
A2      : 0x0943463a  A3      : 0x3f401600  A4      : 0x00000007  A5      : 0x0000ff00
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x39353731  A9      : 0x00000000
A10     : 0x00000003  A11     : 0x3f404df9  A12     : 0x00000002  A13     : 0x0000ff00
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0943463a  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffd

Backtrace: 0x40001277:0x3ffd0d70 0x400dbc11:0x3ffd0d80 0x400d7713:0x3ffd0da0 0x400d79d3:0x3ffd0dc0 0x400895dd:0x3ffd0e90
  #0  0x40001277:0x3ffd0d70 in ?? ??:0
  #1  0x400dbc11:0x3ffd0d80 in String::equals(char const*) const at C:\Users\USER\.platformio\packages\framework-arduinoespressif32\cores\esp32/WString.cpp:857
  #2  0x400d7713:0x3ffd0da0 in String::operator==(char const*) const at C:/Users/USER/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:191
      (inlined by) bool __gnu_cxx::__ops::_Iter_equals_val<char const [1]>::operator()<__gnu_cxx::__normal_iterator<String*, std::vector<String, std::allocator<String> > > >(__gnu_cxx::__normal_iterator<String*, std::vector<String, std::allocator<String> > >) at c:\users\user\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits/predefined_ops.h:194
      (inlined by) __gnu_cxx::__normal_iterator<String*, std::vector<String, std::allocator<String> > > std::__find_if<__gnu_cxx::__normal_iterator<String*, std::vector<String, std::allocator<String> > >, __gnu_cxx::__ops::_Iter_equals_val<char const [1]> >(__gnu_cxx::__normal_iterator<String*, std::vector<String, std::allocator<String> > >, __gnu_cxx::__normal_iterator<String*, std::vector<String, std::allocator<String> > >, __gnu_cxx::__ops::_Iter_equals_val<char const [1]>, std::random_access_iterator_tag) at c:\users\user\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits/stl_algo.h:128
  #3  0x400d79d3:0x3ffd0dc0 in __gnu_cxx::__normal_iterator<String*, std::vector<String, std::allocator<String> > > std::__find_if<__gnu_cxx::__normal_iterator<String*, std::vector<String, std::allocator<String> > >, __gnu_cxx::__ops::_Iter_equals_val<char const [1]> >(__gnu_cxx::__normal_iterator<String*, std::vector<String, std::allocator<String> > >, __gnu_cxx::__normal_iterator<String*, std::vector<String, std::allocator<String> > >, __gnu_cxx::__ops::_Iter_equals_val<char const [1]>) at c:\users\user\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits/stl_algo.h:162
      (inlined by) __gnu_cxx::__normal_iterator<String*, std::vector<String, std::allocator<String> > > std::find<__gnu_cxx::__normal_iterator<String*, std::vector<String, std::allocator<String>      (inlined by) logWiFiDeviceData(void*) at src/taskFunctions.cpp:222
  #4  0x400895dd:0x3ffd0e90 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
ets Jun  8 2016 00:22:57
```

As you can see, the serial only print "e ID = " (supposed to be "Slave ID = 4", though.) once and crashed subsequently....
Another point is that if I uncomment the line: ```Serial.println( "1.Slave ID = " + String(payload.slaveID) );```
I could get
```
RSSIc[0] = ""
RSSIc[1] = ""
RSSIc[2] = ""
RSSIc[3] = ""
MAC matched!
1.Slave ID = 4
e ID =
Guru Meditation Error: .... (same as above)
```

It seemed the Serial.println worked well before the if-condition statement, but ESP32 still crashed/rebooted afterwards.

I am new to ESP32 and helps anyone could help me to debug.
Any help will be greatly appreciated! Thanks!

DougArmstrong, 2021-01-07:
This code appears to be crashing with serial data still in the output queue.  I recommend you put some delay statements inline and allow it to execute slowly until you may see what line is actually crashing the system.  If it works then there may be some type of timing issue.

jandrassy, 2021-01-07:
what is `"Slave ID = " + payload.slaveID`?  it is a pointer to "Slave ID = " plus ASCII code of the char slaveId, which is a pointer into random memory location.

hmhuang0501, 2021-01-07:
> This code appears to be crashing with serial data still in the output queue. I recommend you put some delay statements inline and allow it to execute slowly until you may see what line is actually crashing the system. If it works then there may be some type of timing issue.

But why the ```Serial.println( "1.Slave ID = " + String(payload.slaveID) );``` works well if the data still in the output queue?

hmhuang0501, 2021-01-07:
> what is `"Slave ID = " + payload.slaveID`? it is a pointer to char array "Slave ID = " plus the value of char slaveId. if slaveId is 4 it is a pointer to "e ID =". if slaveId is larger then the length of the "Slave ID = " then it is a pointer to some random memory content

Why it is a pointer to "e ID =" , not to the first element of "Slave ID = " + the value of char slaveID if slaveID is 4?

jandrassy, 2021-01-07:
String has operator + as concatenation.
on char array pointer + advances the pointer 4 characters

hmhuang0501, 2021-01-08:
I change the block
```cpp
if ( payload.slaveID == 4 )  // check if the RSSI value received by respective slave has been filled in.      
{                                                           
  Serial.println( "Slave ID = " + payload.slaveID ); 
}
```

to

```cpp
if ( payload.slaveID == 4 )  // check if the RSSI value received by respective slave has been filled in.      
{                                                           
  Serial.print( "Slave ID = " );
  Serial.println( payload.slaveID ); 
}
```

The output is ```Slave ID =  ```, shouldn't it be ```Slave ID = 4``` in this case?
Why the information is lost ```payload.slaveID``` after the if-condition statement? Or it still exist but not shown on the serial terminal? (But why.......)

jandrassy, 2021-01-09:
char with ascii code 4 is invisible control character

closed 2021-07-08 by stale[bot]