espressif/arduino-esp32 #4695: Possible memory issue in BLE when receiving long value

Sonnbc, 2021-01-07:
### Hardware:
Board:							NodeMCU-32 v1.2
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                      no
Upload Speed:					115200
Computer OS:                                           Mac OSX

### Description:
I'm creating a BLE server on my ESP dev module with 1 characteristic. Writing a long value to this characteristic (for example, with the *nRF Connect* app on my phone) seems to cause memory corruption on the board. 

#### Reproduce:
1. Compile and upload the code below  
2. Use *nRF Connect* on a phone (or any app that supports sending characteristic values) to send the string (ASCII) `aaaaaaaaaaaaaaaaaaaabbbbbbbbbbbbbbbbbbbbccccccccccccccccccccdddddddddddddddddddd` (20 characters of `a`, `b`, `c`, `d` each) to the only characteristic on the BLE server.
3. Arduino IDE console on the server shows wrong output:
```
onWrite onWrite aaaaaaaaaaaabbbbbbbbbbbbbbbbbbbbccccccccccccccccccccdddddddddddddddddddd
```
the string `onWrite ` (8 bytes) is duplicated and written into our long string. Only 12 `a`s left.
4. Replace line 11 with the following fixes the problem and the correct value is printed: 
```cpp
Serial.printf("onWrite %s\n", pCharacteristic->getValue().c_str());
```
Output:
```
onWrite aaaaaaaaaaaaaaaaaaaabbbbbbbbbbbbbbbbbbbbccccccccccccccccccccdddddddddddddddddddd
```
 

### Sketch: 
```cpp
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>

#define SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"

class BLEManager : public BLECharacteristicCallbacks {
  void onWrite(BLECharacteristic *pCharacteristic) {
    const char* value = pCharacteristic->getValue().c_str();
    Serial.printf("onWrite %s\n", value);
  }
};

void setup() {
  Serial.begin(115200);
  Serial.println("Starting BLE work!");

  BLEDevice::init("Long name works now");
  BLEServer *pServer = BLEDevice::createServer();
  BLEService *pService = pServer->createService(SERVICE_UUID);
  BLECharacteristic *pCharacteristic = pService->createCharacteristic(
                                         CHARACTERISTIC_UUID,
                                         BLECharacteristic::PROPERTY_READ |
                                         BLECharacteristic::PROPERTY_WRITE
                                       );

  pCharacteristic->setValue("Hello World says Neil");
  pCharacteristic->setCallbacks(new BLEManager());
  
  pService->start();
  BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
  pAdvertising->addServiceUUID(SERVICE_UUID);
  pAdvertising->setScanResponse(true);
  pAdvertising->setMinPreferred(0x06);  // functions that help with iPhone connections issue
  pAdvertising->setMinPreferred(0x12);
  BLEDevice::startAdvertising();
  Serial.println("Characteristic defined! Now you can read it in your phone!");
[... 132 lines cut ...]
21:50:07.208 -> [V][BLEServer.cpp:271] handleGATTServerEvent(): << handleGATTServerEvent
21:50:22.360 -> [D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
21:50:22.360 -> [V][BLEUtils.cpp:1530] dumpGattServerEvent(): GATT ServerEvent: Unknown
21:50:22.398 -> [V][BLEUtils.cpp:1720] dumpGattServerEvent(): dumpGattServerEvent: *** NOT CODED ***
21:50:22.398 -> [V][BLEServer.cpp:144] handleGATTServerEvent(): >> handleGATTServerEvent: Unknown
21:50:22.398 -> [V][BLECharacteristic.cpp:198] handleGATTServerEvent(): >> handleGATTServerEvent: Unknown
21:50:22.398 -> [V][BLECharacteristic.cpp:646] setValue(): >> setValue: length=80, data=6161616161616161616161616161616161616161626262626262626262626262626262626262626263636363636363636363636363636363636363636464646464646464646464646464646464646464, characteristic UUID=beb5483e-36e1-4688-b7f5-ea07361b26a8
21:50:22.434 -> [D][FreeRTOS.cpp:189] take(): Semaphore taking: name: SetValue (0x3ffdddbc), owner: <N/A> for <Unknown>
21:50:22.434 -> [D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: SetValue (0x3ffdddbc), owner: <Unknown>
21:50:22.434 -> [V][FreeRTOS.cpp:143] give(): Semaphore giving: name: SetValue (0x3ffdddbc), owner: <Unknown>
21:50:22.472 -> [V][BLECharacteristic.cpp:655] setValue(): << setValue
21:50:22.472 -> [D][BLECharacteristic.cpp:285] handleGATTServerEvent():  - Response to write event: New value: handle: 2a, uuid: beb5483e-36e1-4688-b7f5-ea07361b26a8
21:50:22.472 -> [D][BLECharacteristic.cpp:288] handleGATTServerEvent():  - Data: length: 80, data: 6161616161616161616161616161616161616161626262626262626262626262626262626262626263636363636363636363636363636363636363636464646464646464646464646464646464646464
21:50:22.510 -> onWrite onWrite aaaaaaaaaaaabbbbbbbbbbbbbbbbbbbbccccccccccccccccccccdddddddddddddddddddd
21:50:22.510 -> [V][BLECharacteristic.cpp:452] handleGATTServerEvent(): << handleGATTServerEvent
21:50:22.510 -> [V][BLEServer.cpp:271] handleGATTServerEvent(): << handleGATTServerEvent
21:50:22.510 -> [D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
21:50:22.510 -> [V][BLEUtils.cpp:1530] dumpGattServerEvent(): GATT ServerEvent: Unknown
21:50:22.547 -> [V][BLEUtils.cpp:1720] dumpGattServerEvent(): dumpGattServerEvent: *** NOT CODED ***
21:50:22.547 -> [V][BLEServer.cpp:144] handleGATTServerEvent(): >> handleGATTServerEvent: Unknown
21:50:22.547 -> [V][BLECharacteristic.cpp:198] handleGATTServerEvent(): >> handleGATTServerEvent: Unknown
21:50:22.547 -> [V][BLECharacteristic.cpp:452] handleGATTServerEvent(): << handleGATTServerEvent
21:50:22.547 -> [V][BLEServer.cpp:271] handleGATTServerEvent(): << handleGATTServerEvent

```

chegewara, 2021-01-07:
Sorry, but i dont see what problem you may have. Logs seems to be correct, no crash, what do you mean by long value?

Sonnbc, 2021-01-08:
> Sorry, but i dont see what problem you may have. Logs seems to be correct, no crash, what do you mean by long value?

Hi @chegewara, 

By long value I mean I send a string that is relatively large in size (80 characters) to the characteristic. You can see in the sketch, the `Serial.printf` statement should print
```
onWrite aaaaaaaaaaaaaaaaaaaabbbbbbbbbbbbbbbbbbbbccccccccccccccccccccdddddddddddddddddddd
```
But instead, it printed
```
onWrite onWrite aaaaaaaaaaaabbbbbbbbbbbbbbbbbbbbccccccccccccccccccccdddddddddddddddddddd
```
which is incorrect. 

As I described in the **reproduce** part, changing the `Serial.printf` line fixes the problem but the two statements should be equivalent and this probably indicates that there is some memory issue here.

chegewara, 2021-01-08:
I dont think it is issue with ble, it looks like some issue with Serial.printf. To confirm it you can try for loop and print every single character from value. It is strange, because i never got issues like this.

Sonnbc, 2021-01-08:
The value is definitely overwritten. I changed the code to:
```cpp
class BLEManager : public BLECharacteristicCallbacks {
  void onWrite(BLECharacteristic *pCharacteristic) {
    const char* value = pCharacteristic->getValue().c_str();
    Serial.printf("onWrite %s\n", value);
    Serial.print("char printing:");
    for (int i = 0; i < strlen(value); i++) {
      Serial.print(value[i]);
    }
    Serial.println();
  }
};
```
And here is the log:
```
onWrite onWrite aaaaaaaaaaaabbbbbbbbbbbbbbbbbbbbccccccccccccccccccccdddddddddddddddddddd
char printing:onWrite onWrite aaaaaaaaaaaabbbbbbbbbbbbbbbbbbbbccccccccccccccccccccdddddddddddddddddddd
```

chegewara, 2021-01-09:
I dont know what to say or how to explain this. I am using BLE in many projects, but never got something like this.
Even in this line it looks like 2 `onWrite` is appended in front, but your actual string is still the same (wrong) length:
`char printing:onWrite onWrite `

closed 2021-07-08 by stale[bot]