espressif/arduino-esp32 #4696: Task watchdog got triggered. The following tasks did not reset the watchdog in time:

tegusg, 2021-01-07:
Hi, i am new in this environment of collaboration but i'll try to be exact, also my english it's poor 😢 

### Hardware:
|||||||
|:---|---|---|---|---|---|
|<B>Board</B>|ESP32 Dev Module|
|<B>Version/Date</B>|1.0.4|
|<B>IDE name</B>|Arduino IDE|
|<B>Flash Frequency</B>|80Mhz|
|<B>Upload Speed</B>|115200|
|<B>Computer OS</B>|Windows 10|

i'm using the ESP32 to take audio signal by the adc and send it through UDP to Python code in my pc.
The signal it's ok, i'll improve with adc module for audio, because the quality it's not good.
Everything works while i use only one core (1) by the typical loop with Arduino IDE, but i need more speed to less lag, 
When i try to implement the second core (0) i have a problem because i got the next error:

task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (17446) task_wdt:  - IDLE0 (CPU 0)
E (17446) task_wdt: Tasks currently running:
E (17446) task_wdt: CPU 0: loop2
E (17446) task_wdt: CPU 1: loop
E (17446) task_wdt: Aborting.
abort() was called at PC 0x400e37d3 on core 0

i've seen this problem to others here, the solution it's add a delay(10) or vTaskDelay (10) in task of core(0),
that's works but at the same time add more lag to audio transmision, this is bad.

Other solution [Here](https://github.com/espressif/esp-idf/issues/1646) also add lag like [This](https://github.com/espressif/arduino-esp32/issues/3871)

so can you help me to implement this faster as possible using two cores or better optimized? 
i've seen a increible projects with this ESP32 and i'm sure that it's possible to get more 
processing power

don't worry by the fails things that i've done it, in every task of every core; the problem it's not there

### Sketch:  
```cpp
void setup() {
  
  //Serial.begin(115200);
  //Serial.println();
  //WiFi.mode(WIFI_STA);              **no important this for the question, i guest**
  //WiFi.begin(ssid, password);
  //while (WiFi.status() != WL_CONNECTED) {
    //Serial.print(".");
  //delay(500);
  //}

 
  xTaskCreatePinnedToCore(codeForTask2, "loop2", 1000, NULL, 1, &Task2, 0);
  delay(500);
  
}

void loop(  )
{
    
 //ready1=false;
  Udp.beginPacket("myIp", 4321);

  for(int i=0; i<730;i++){ 
     int old=micros();
     
     Udp.write(uint8_t(analog)); 
     Udp.write(uint8_t(analog>>8));
     
     while(micros()-old<20); //22uSec = 1Sec/44100Hz - 2uSec 
    }
     //ready1=true;
  //while(ready1==true);
  Udp.endPacket();
  
    //Serial.print("0, 2048, 4096, ");
    //Serial.println(analog);
   
  }

void codeForTask2( void * parameter )
{
  for (;;) {
      
      delay(10);
//    if(ready1==true)
//    {
//      Udp.endPacket();
//      ready1=false;
//      }
    analog = analogRead(34);
    Serial.print("0, 2048, 4096, ");
    Serial.println(analog);
    //vTaskDelay (10);
  }
  
}
```

dumarjo, 2021-01-08:
Since your loop function is higher priority than IDLE, you need to exit the loop function quicker or yield, or kick the watchdog.

tegusg, 2021-01-08:
Thanks u, can you give an idea how implement that you say, sounds good!
Sorry i'm a noob

dumarjo, 2021-01-08:
you can fiid the watch dog like this
```
TIMERG0.wdt_wprotect=TIMG_WDT_WKEY_VALUE;
TIMERG0.wdt_feed=1;
TIMERG0.wdt_wprotect=0;
```
You add this in your for loop in your case. This is bad to do this normaly, but sometime it's the only solution. 

 you will nee to add those headers too I think
```
#include "soc/timer_group_struct.h"
#include "soc/timer_group_reg.h"
```

tegusg, 2021-01-09:
really very thank u, it´s working the lag was reduce, but i have a new problem because like you know i'm working with audio signal and now i listen a few moments where the signal transmission is disrupted like a shut down (so short but annoying), 
i don't know if the solution add dead time or breaks.

closed 2021-01-27 by tegusg