espressif/arduino-esp32 #4697: Data loss when receiving from the Queue

hmhuang0501, 2021-01-08:
Hi, I am doing a project with writing two tasks both pinned to core 0 with TTGO LoRa ESP32 as a server device. One task (say, Task 1) for sending data (which is received from other client esps via LoRa radio) to the queue, one (say, Task 2) for receiving data from the queue. 

The code for Task 2 is below:
```cpp
 xTaskCreatePinnedToCore( logWiFiDeviceData,                        // Task function
                                         "TaskForLoggingWiFiDeviceData",    // Task name
                                         10000,                                           // Stack size in words
                                         NULL,                                            // Parameter passed as input to task
                                         0,                                                  // Task priority (0 is the lowest priority)
                                         NULL,                                            // Task handle
                                         0 );                                                // Core ID
```

```cpp
void logWiFiDeviceData( void* parameter )
{
  wifiDevicePayload payload;

  for(;;)
  {
    for ( size_t i = 0; i < cg_queueSizeBetweenReceiveAndLog; ++i )
    {
      xQueueReceive( g_queueBetweenReceiveAndLog, &payload, portMAX_DELAY );
      
     if ( strcmp( payload.mac, DESIRED_DEVICE_MAC_ADDRESS ) == 0 )
      {
        Serial.println( "MAC matched!" );
        Serial.print( "Slave ID = " );
        Serial.println( payload.slaveID );
        Serial.print( "MAC = " );
        Serial.println( payload.mac );
        Serial.print( "RSSI = " );
        Serial.println( payload.rssi );
        Serial.print( "Packet ID = " );
        Serial.println( payload.packetID ); 
      }
     }
     vTaskDelay( 1 / portTICK_PERIOD_MS );
  }

} // logWiFiDeviceData
```

The struct wifiDevicePayload is defined as:
```cpp
typedef struct
{
  char mac[100] = {0};
  uint32_t packetID;
  int8_t rssi;                           
  int8_t slaveID;

} __attribute__((packed)) wifiDevicePayload;
```

The output is as follows:
[attachment=0]2021-01-09 00 10 38.jpg[/attachment]](url
[image]
)

My question is: why are the information of payload.slaveID, payload.rssi, payload.packetID (which are type of int8_t, int8_t, uint32_t, respectively) got lost when transmitted by the Queue?
I am sure that the three members have non-zero values before sending into the Queue. They should be something like:

```
MAC matched!
Slave ID = 4
MAC = 38:10:D5:5C:58:62
RSSI = -76
packetID = 3465
```

I really can't figure out why I got this result and have nothing to do with this weird issue.....
Please help me if any ideas...
Any help is greatly appreciated!

chegewara, 2021-01-09:
This is only half code. We dont see xQueueSend part, so hard to say.

hmhuang0501, 2021-01-09:
@chegewara Thanks for reminding me! I added them into the post.

chegewara, 2021-01-09:
> WiFiDevicePayload.rssi = (int8_t) incomingPayload.substring(pos2 + 1, pos3).toInt(); 

Did you confirm this value is correct? I think it should be `pos3 - 1`. Either way, did you print this value to logs to make sure it is expected value and different than received by xQueueReceive?

> WiFiDevicePayload.packetID = (uint32_t) strtoul( incomingPayload.substring(0, pos1).c_str(), NULL, 0 ); 

Same question here.

hmhuang0501, 2021-01-09:
> Did you confirm this value is correct? I think it should be `pos3 - 1`. Either way, did you print this value to logs to make sure it is expected value and different than received by xQueueReceive? [...]

@4695  Yes, I did compare the values before and after the Queue. The output is below:
[image]

As you can see, the payload.mac didn't lose but the other three did. I only come up with the data type of each of them may be related to this problem. Really strange....

chegewara, 2021-01-09:
I didnt see Serial logs from received packets, thats why i asked.
QueueSend/Receive code looks good to me.

btw, in my opinion you should simplify packets format:
`incomingPayload format: packetID + '/' + MAC address + '\t' + RSSI + '\n'`
to
`incomingPayload format: packetID + RSSI + MAC address` - packetID is uint32_t, RSSI is int8_t, so it is 5 bytes, remaining is MAC, which is easier to parse.

hmhuang0501, 2021-01-13:
@chegewara Thank you very much for your help! I solved this issue by moving char mac[100] = {0}; as the last member in the struct as you said. I never come up with this problem could be from the order of struct members. Could you give me some explanations why this is the case? Thanks!

chegewara, 2021-01-13:
I dont have good explanation for that, just i thought when mac is shorter than 100 bytes, then for some reason rssi and slaveID are taken from mac, which is all 0s. Maybe its some weird compiler optimization or something else i dont know yet.

closed 2021-07-08 by stale[bot]