espressif/arduino-esp32 #4700: Cannot turn off WIFi

philbowles, 2021-01-09:
Core 1.0.4, multiple different boards.

I'm trying to determine what state the WiFi is in immediately after boot
Calling printDiag first thing causes guru meditation

```
Decoding 10 results
0x400d1c99: Print::write(char const*) at C:\Users\phil\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32/Print.cpp line 234
0x400d1ce1: Print::print(char const*) at C:\Users\phil\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32/Print.cpp line 234
:  (inlined by) Print::println(char const*) at C:\Users\phil\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32/Print.cpp line 182
0x400d0f73: WiFiClass::printDiag(Print&) at C:\Users\phil\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi\src/WiFi.cpp line 60
0x400d0e75: setup() at C:\Users\phil\Desktop\ESP32_WIFI_BUG/ESP32_WIFI_BUG.ino line 9
0x400d21df: loopTask(void*) at C:\Users\phil\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32/main.cpp line 14
0x40088b7d: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 355 (discriminator 1)
```

Q1 why has #1088 never been fixed?

Q2 Both getMode() and SSID() report incorrect values if called first thing. What is  the "correct" way to determine intial state / settings without first calling some API that will change them, e.g. WiFi.mode(...) or WiFi.disconnect?

Q3 What is  the "correct" way to turn off WiFi? `WiFi.softAPdisconnect(true)` does not do it*, WiFi.mode(WIFI_OFF) does not do it*

Q4 What is the "correct" way to clear any stored credentials as in the ESP8266 equivalent of `ESP.eraseConfig()`?  I want to do a full "factory reset" and the IDE does not have "total flash erase" tools options for either of my two boards: WEMOS LOLIN32 and ESP32 Dev Module

 # MVCE

```cpp
#include <WiFi.h>
void setup()
{
    Serial.begin(115200);
    // WiFi.printDiag(Serial); // guru meditation
    Serial.printf("INITIAL MODE=%d\n",WiFi.getMode()); // always reports 0
    Serial.printf("STORED  SSID=%s\n",WiFi.SSID().c_str()); // not true
    WiFi.softAP("BOGUS");
    Serial.printf("\nAfter softAP\n");
    WiFi.printDiag(Serial); // now shows ACTUAL stored SSID
    // keywords is missing softAPdisconnect: not syntax colored
    WiFi.softAPdisconnect(true); // true -> does NOT turn off WiFi!
    Serial.printf("\nAfter softAPdisconnect(true)\n");
    WiFi.printDiag(Serial);
    WiFi.mode(WIFI_OFF);
    Serial.printf("\nAfter WiFi.mode(WIFI_OFF)\n");
    WiFi.printDiag(Serial);
}
void loop(){}
```

# Output WEMOS LOLIN32 (no diag options available)
```
14:53:45.283 -> ets Jun  8 2016 00:22:57
14:53:45.283 -> 
14:53:45.283 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
14:53:45.283 -> configsip: 0, SPIWP:0xee
14:53:45.283 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
14:53:45.283 -> mode:DIO, clock div:1
14:53:45.283 -> load:0x3fff0018,len:4
14:53:45.283 -> load:0x3fff001c,len:1044
14:53:45.283 -> load:0x40078000,len:8896
14:53:45.283 -> load:0x40080400,len:5816
14:53:45.283 -> entry 0x400806ac
14:53:45.520 -> INITIAL MODE=0
14:53:45.520 -> STORED  SSID=
14:53:45.621 -> 
14:53:45.621 -> After softAP
14:53:45.621 -> Mode: AP
14:53:45.621 -> Channel: 1
14:53:45.621 -> SSID (8): XXXXXXXX
14:53:45.621 -> Passphrase (8): XXXXXXXX
14:53:45.621 -> BSSID set: 0
14:53:45.655 -> 
14:53:45.655 -> After softAPdisconnect(true)
14:53:45.655 -> Mode: AP
14:53:45.655 -> Channel: 1
14:53:45.655 -> SSID (8): XXXXXXXX
14:53:45.655 -> Passphrase (8): XXXXXXXX
14:53:45.655 -> BSSID set: 0
14:53:45.655 -> 
14:53:45.655 -> After WiFi.mode(WIFI_OFF)
14:53:45.655 -> Mode: AP
14:53:45.655 -> Channel: 1
14:53:45.655 -> SSID (8): XXXXXXXX
14:53:45.655 -> Passphrase (8): XXXXXXXX
14:53:45.655 -> BSSID set: 0

```
# ESP32 Dev Board diag=verbose
```
15:01:50.857 -> ets Jun  8 2016 00:22:57
15:01:50.857 -> 
15:01:50.857 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
15:01:50.857 -> configsip: 0, SPIWP:0xee
15:01:50.857 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
15:01:50.857 -> mode:DIO, clock div:1
15:01:50.857 -> load:0x3fff0018,len:4
15:01:50.892 -> load:0x3fff001c,len:1216
15:01:50.892 -> ho 0 tail 12 room 4
15:01:50.892 -> load:0x40078000,len:9720
15:01:50.892 -> ho 0 tail 12 room 4
15:01:50.892 -> load:0x40080400,len:6352
15:01:50.892 -> entry 0x400806b8
15:01:51.101 -> INITIAL MODE=0
15:01:51.101 -> STORED  SSID=
15:01:51.196 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
15:01:51.196 -> 
15:01:51.196 -> After softAP
15:01:51.196 -> Mode: AP
15:01:51.196 -> Channel: 1
15:01:51.196 -> SSID (8): XXXXXXXX
15:01:51.196 -> Passphrase (8): XXXXXXXX
15:01:51.243 -> BSSID set: 0
15:01:51.243 -> 
15:01:51.243 -> After softAPdisconnect(true)
15:01:51.243 -> Mode: AP
15:01:51.243 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 15 - AP_STACONNECTED
15:01:51.243 -> Channel: 1
15:01:51.243 -> SSID ([D][WiFiGeneric.cpp:337] _eventCallback(): Event: 14 - AP_STOP
15:01:51.243 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 15 - AP_STACONNECTED
15:01:51.243 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 15 - AP_STACONNECTED
15:01:51.243 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 15 - AP_STACONNECTED
15:01:51.243 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 15 - AP_STACONNECTED
15:01:51.290 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 15 - AP_STACONNECTED
15:01:51.290 -> 
15:01:51.290 -> BSSID set: 0
15:01:51.290 -> 
15:01:51.290 -> After WiFi.mode(WIFI_OFF)
15:01:51.290 -> Mode: AP
15:01:51.290 -> Channel: 1
15:01:51.290 -> SSID (8): XXXXXXXX
15:01:51.290 -> Passphrase (8): XXXXXXXX
15:01:51.290 -> BSSID set: 0
```
Looking at that event trace - maybe this is why printDiag fails - because some background has not yet started WiFi ? Maybe all users need to monitor WiFi events and not call anything until after event 0? I'm guessing this is also why getmode and ssid report incorrect values... Is there anything in the docs I have missed that make this clear? Also does the WiFi have to be in a certain state before any of the OFF methods work? What am I doing wrong?

philbowles, 2021-01-09:
I can confirm that printDiag behaves as expected if called after  SYSTEM_EVENT_WIFI_READY. Long term solution is to use totally event-driven methodology. Short term is to change documentation / examples to require inefficient arbitrary delay( ? ) before any WiFi call to allow for SYSTEM_EVENT_WIFI_READY to have occurred.

Similarly, getMode() returns expected value after SYSTEM_EVENT_WIFI_READY , however SSID() still reports empty string even though printDiag clearly shows a value. Does SSID() only work inSTA mode ?

```cpp
    case SYSTEM_EVENT_WIFI_READY:
      Serial.printf("WIFI READY!\n");
      WiFi.printDiag(Serial); // guru meditation
      Serial.printf("INITIAL MODE=%d\n",WiFi.getMode()); // always reports 0
      Serial.printf("STORED  SSID=%s\n",WiFi.SSID().c_str()); // not true
      break;
```

shows:

```
16:25:11.158 -> WIFI READY!
16:25:11.158 -> Mode: AP
16:25:11.158 -> Channel: 1
16:25:11.158 -> SSID (8): XXXXXXXX
16:25:11.192 -> Passphrase (8): XXXXXXXX[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 14 - AP_STOP
16:25:11.192 -> INITIAL MODE=2
16:25:11.192 -> STORED  SSID=
16:25:11.192 -> 
```

but  no change to inability to tun off

pvint, 2021-01-09:
I can confirm that experience each of these same issues (Tested on a custom WROOM32 device usind DOWD-Q6 chip.

pvint, 2021-01-09:
A quick test seems to show that it is actually turning the radio off - I used the sketch above and moved the actions into `loop()` and added a 5000ms delay after turning on/off, and I'm seeing the current draw on the board change from ~130mA when "on" to ~45mA when "off"

philbowles, 2021-01-09:
AHA seems like this is the SAME unresolved, unfixed bug that been around since Apr 2020 (see #3707). I feel sorry for the poor guy who reported THAT one and STILL hasn't got a fix...

philbowles, 2021-01-09:
Finally I see ***ALL*** the issues I have here are the result of failure to address earlier issues... I just found this one re the blank SSID() #1743. So even though "officially" I ought to close this as I now know the reasons for (and have answers to, or workarounds for) all of the issues raised...I'm going to leave it open so that hopefully other users can see just how poorly this firmware is supported and how little attention and actual "support" is given to users who devite their own free time and tear their hair out to a) find problems b) suggest good, working fixes, only to have them completely ignored :(

I'm also going to remember this number #4700 and every few weeks I'm going to come back in here and raise another issue, referencing this one #4700 to see if anything has been done yet about any of them. The more people who do that, the more likely it is that we will either: get them fixed AT LAST or at least hear some kind of cogent argument why "the team" continues to fail / refuse to fix them. For what is otherwise a GREAT product, this lack of decent support is such a huge shame. :(

closed 2021-07-08 by stale[bot]