espressif/arduino-esp32 #4706: any possibility to merge BLE disconnect fix?

tonton81, 2021-01-11:
Windows 8, esp32 wroom, 1.0.5-rc4, arduino IDE

Any way to merge in a BLE disconnect fix?
https://github.com/espressif/esp32-bt-lib/commit/04bb2cd254b4a13c7979276160a59430e2335fb3

this issue is affecting all BLE users and causes lockups or issues without workarounds (user forced esp32 resets by code or host resets of ble controller). Issue can be replicated by setting disconnects on a timer and letting the BLE reconnect, over time (couple hours) the esp32 will either freeze or crash & reboot

lbernstone, 2021-01-11:
me-no-dev merged esp-idf v3.3 changes into master this morning, so if it that fix is in idf, then you can test that branch.  It is unlikely that he will be adding anything outside of idf into 1.0.5 after 4 release candidates.

h2zero, 2021-01-11:
>me-no-dev merged esp-idf v3.3 changes into master this morning

I can confirm that this merge includes the referenced commit in the OP.

btsimonh, 2021-01-16:
thanks for this....  but... using this RC6 version, is anyone else getting a stack canary on ipc0 task related to BLE?:

```
10:26:55.827 readOneSensor - already active reading 2
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (ipc0) 
Core 0 register dump:
PC      : 0x40094c83  PS      : 0x00060036  A0      : 0x40084a64  A1      : 0x3ffb9a20  
A2      : 0x00000000  A3      : 0x00060023  A4      : 0x00060020  A5      : 0x00000001
A6      : 0x00000020  A7      : 0x00000000  A8      : 0x801c5842  A9      : 0x3ffb9bb0  
A10     : 0x00000008  A11     : 0x40082f30  A12     : 0x3ffb61b4  A13     : 0x00000000
A14     : 0x3ffb616c  A15     : 0x00060b23  SAR     : 0x00000008  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff

ELF file SHA256: 0000000000000000

Backtrace: 
0x40094c83:0x3ffb9a20 
0x40084a61:0x3ffb9af0 _xt_user_exit
0x40095657:0x3ffb9b10 
0x40095dae:0x3ffb9b30 multi_heap_malloc
0x40084535:0x3ffb9b50 
0x40084566:0x3ffb9b70 
0x40082f35:0x3ffb9b90 _malloc_r
0x4000beaf:0x3ffb9bb0 
0x401c583f:0x3ffb9bd0 btdm_task_post
0x401c58b9:0x3ffb9c00 
0x40181b7d:0x3ffb9c30 async_wakeup_request
0x40181e2d:0x3ffb9c50 esp_vhci_host_send_packet
0x4011eb8c:0x3ffb9c70 ble_hci_trans_hs_acl_tx_on_core_0
0x40087473:0x3ffb9da0 libesp32.a(ipc.o)
0x4009226a:0x3ffb9dc0 (port.o)

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
```

closed 2021-07-08 by stale[bot]