espressif/arduino-esp32 #4707: Does arduino for esp32s2 not support i2s ？

tumuyan, 2021-01-11:
I try to build and download the  i2s example (ESP32\examples\I2S\HiFreq_ADC)，but it shows build error.  
then I find this folder has a file named as .skip.esp32s2  
Does arduino for esp32s2 not support i2s?  

  
### Hardware:
Board:							ESP32-S2-SAOLA 
Computer OS:                                        Windows 10

### Debug Messages:
```
Arduino:1.8.13 Hourly Build 2020/05/14 06:33 (Windows 10), 开发板："ESP32S2 Dev Module, UART0, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi), QIO, 80MHz, 4MB (32Mb), 921600, None"

C:\Users\Yazii\Documents\Arduino\HiFreq_ADC_\HiFreq_ADC_.ino: In function 'void i2sInit()':

C:\Users\Yazii\Documents\Arduino\HiFreq_ADC_\HiFreq_ADC_.ino:27:29: warning: 'I2S_COMM_FORMAT_I2S_MSB' is deprecated [-Wdeprecated-declarations]

     .communication_format = I2S_COMM_FORMAT_I2S_MSB,

                             ^~~~~~~~~~~~~~~~~~~~~~~

In file included from C:\Users\Yazii\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/esp32s2/include/hal/esp32s2/include/hal/i2s_ll.h:27,

                 from C:\Users\Yazii\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/esp32s2/include/hal/include/hal/i2s_hal.h:28,

                 from C:\Users\Yazii\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/esp32s2/include/driver/include/driver/i2s.h:24,

                 from C:\Users\Yazii\Documents\Arduino\HiFreq_ADC_\HiFreq_ADC_.ino:6:

C:\Users\Yazii\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/esp32s2/include/hal/include/hal/i2s_types.h:71:5: note: declared here

     I2S_COMM_FORMAT_I2S_MSB   __attribute__((deprecated)) = 0x01, /*!< I2S format MSB, (I2S_COMM_FORMAT_I2S |I2S_COMM_FORMAT_I2S_MSB) correspond to `I2S_COMM_FORMAT_STAND_I2S`*/

     ^~~~~~~~~~~~~~~~~~~~~~~

HiFreq_ADC_:36:21: error: 'ADC_UNIT_1' was not declared in this scope

    i2s_set_adc_mode(ADC_UNIT_1, ADC_INPUT);

                     ^~~~~~~~~~

C:\Users\Yazii\Documents\Arduino\HiFreq_ADC_\HiFreq_ADC_.ino:36:21: note: suggested alternative: 'ADC_INPUT'

    i2s_set_adc_mode(ADC_UNIT_1, ADC_INPUT);

                     ^~~~~~~~~~

                     ADC_INPUT

HiFreq_ADC_:9:19: error: 'ADC1_CHANNEL_4' was not declared in this scope

 #define ADC_INPUT ADC1_CHANNEL_4 //pin 32

                   ^~~~~~~~~~~~~~

C:\Users\Yazii\Documents\Arduino\HiFreq_ADC_\HiFreq_ADC_.ino:36:33: note: in expansion of macro 'ADC_INPUT'

    i2s_set_adc_mode(ADC_UNIT_1, ADC_INPUT);

                                 ^~~~~~~~~

C:\Users\Yazii\Documents\Arduino\HiFreq_ADC_\HiFreq_ADC_.ino:9:19: note: suggested alternative: 'I2S_CHANNEL_MONO'

 #define ADC_INPUT ADC1_CHANNEL_4 //pin 32

                   ^~~~~~~~~~~~~~

C:\Users\Yazii\Documents\Arduino\HiFreq_ADC_\HiFreq_ADC_.ino:36:33: note: in expansion of macro 'ADC_INPUT'

    i2s_set_adc_mode(ADC_UNIT_1, ADC_INPUT);

                                 ^~~~~~~~~

HiFreq_ADC_:36:4: error: 'i2s_set_adc_mode' was not declared in this scope

    i2s_set_adc_mode(ADC_UNIT_1, ADC_INPUT);

    ^~~~~~~~~~~~~~~~

C:\Users\Yazii\Documents\Arduino\HiFreq_ADC_\HiFreq_ADC_.ino:36:4: note: suggested alternative: 'i2s_set_dac_mode'

    i2s_set_adc_mode(ADC_UNIT_1, ADC_INPUT);

    ^~~~~~~~~~~~~~~~

    i2s_set_dac_mode

HiFreq_ADC_:37:4: error: 'i2s_adc_enable' was not declared in this scope

    i2s_adc_enable(I2S_NUM_0);

    ^~~~~~~~~~~~~~

C:\Users\Yazii\Documents\Arduino\HiFreq_ADC_\HiFreq_ADC_.ino:37:4: note: suggested alternative: 'i2s_ll_enable_dma'

    i2s_adc_enable(I2S_NUM_0);

    ^~~~~~~~~~~~~~

    i2s_ll_enable_dma

C:\Users\Yazii\Documents\Arduino\HiFreq_ADC_\HiFreq_ADC_.ino: In function 'void reader(void*)':

HiFreq_ADC_:9:19: error: 'ADC1_CHANNEL_4' was not declared in this scope

 #define ADC_INPUT ADC1_CHANNEL_4 //pin 32

                   ^~~~~~~~~~~~~~

C:\Users\Yazii\Documents\Arduino\HiFreq_ADC_\HiFreq_ADC_.ino:44:26: note: in expansion of macro 'ADC_INPUT'

   uint16_t offset = (int)ADC_INPUT * 0x1000 + 0xFFF;

                          ^~~~~~~~~

C:\Users\Yazii\Documents\Arduino\HiFreq_ADC_\HiFreq_ADC_.ino:9:19: note: suggested alternative: 'I2S_CHANNEL_MONO'

 #define ADC_INPUT ADC1_CHANNEL_4 //pin 32

                   ^~~~~~~~~~~~~~

C:\Users\Yazii\Documents\Arduino\HiFreq_ADC_\HiFreq_ADC_.ino:44:26: note: in expansion of macro 'ADC_INPUT'

   uint16_t offset = (int)ADC_INPUT * 0x1000 + 0xFFF;

                          ^~~~~~~~~

HiFreq_ADC_:63:7: error: 'i2s_adc_disable' was not declared in this scope

       i2s_adc_disable(I2S_NUM_0);

       ^~~~~~~~~~~~~~~

C:\Users\Yazii\Documents\Arduino\HiFreq_ADC_\HiFreq_ADC_.ino:63:7: note: suggested alternative: 'esp_intr_disable'

       i2s_adc_disable(I2S_NUM_0);

       ^~~~~~~~~~~~~~~

       esp_intr_disable

HiFreq_ADC_:65:7: error: 'i2s_adc_enable' was not declared in this scope

       i2s_adc_enable(I2S_NUM_0);

       ^~~~~~~~~~~~~~

C:\Users\Yazii\Documents\Arduino\HiFreq_ADC_\HiFreq_ADC_.ino:65:7: note: suggested alternative: 'i2s_ll_enable_dma'

       i2s_adc_enable(I2S_NUM_0);

       ^~~~~~~~~~~~~~

       i2s_ll_enable_dma

exit status 1

'ADC_UNIT_1' was not declared in this scope

 
```

lbernstone, 2021-01-11:
It supports I2S, but the peripheral is not connected to the ADC SAR.  SOC_I2S_SUPPORTS_ADC_DAC is only defined for esp32, so it probably won't be available in s3 either.

tumuyan, 2021-01-13:
> It supports I2S, but the peripheral is not connected to the ADC SAR. SOC_I2S_SUPPORTS_ADC_DAC is only defined for esp32, so it probably won't be available in s3 either.
> The replacement is the ADC DIG driver, which should support even faster sampling and DMA. https://github.com/espressif/esp-idf/blob/master/components/driver/esp32s2/include/driver/adc.h

Thanks for your reply, but I find it's really hard for me to use  ADC driver to get and decode data

lbernstone, 2021-01-13:
This should allow sampling up to a theoretical max of 400kS/sec on esp32s2 fed to DMA (which can then be written to psram).
The HAL is still cooking in the oven.  https://github.com/espressif/esp-idf/blob/master/components/hal/esp32s2/include/hal/adc_hal.h
Ping this issue in a couple months and I will take another look.  Hopefully espressif will have an example in idf by then.

closed 2021-07-08 by stale[bot]

summerfind, 2023-03-10:
hello, is it too later to ask for this same topic? I got compiling error: 'i2s_adc_enable' was not declared in this scope when compiling: Simple 1Msps ESP32 Oscilloscope - DEMO VIDEO ?
thanks.