espressif/arduino-esp32 #4709: Changing board throws Error Compiling for ESP32/ESP32S2 Dev Module

harshmangukiya, 2021-01-12:
I am programming and testing the same code firmware for ESP32 and ESP32S2 but when I change the from one to another it throws Error Compiling for Board ESP32 Dev Module.

### Hardware:
Board:							ESP32 Dev Module/ESP32S2 Dev Module
Core Installation version:			1.0.4 (idf-release/v4.2)
IDE name:						Arduino IDE
Computer OS:                                        Mac OSX

### Description:
Code file is completely working file and uploading the code to Dev Board but when I switch from one board to another board like ESP32 to ESP32S2 or vice versa it is throwing me error as given below but when I copy the same whole code and make new file and then compile for other board it is working completely fine. Again in new file when I switch the board then it throws the same error.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp
I am uploading same code file to both ESP32 and ESP32S2
```

### Debug Messages:
```
bash: line 0: [: too many arguments
usage: cp [-R [-H | -L | -P]] [-fi | -n] [-apvXc] source_file target_file
       cp [-R [-H | -L | -P]] [-fi | -n] [-apvXc] source_file ... target_directory
exit status 64
/Applications/Arduino.app/Contents/Java/arduino-builder returned 64
Error compiling for board ESP32 Dev Module.
```

chegewara, 2021-02-10:
But you understand that v1.0.4 does not support S2?

I am using idf-release/v4.2 (esp32s2 branch earlier) for over 8 months and no issues ith switching between esp32 and S2 on ubuntu 20.04.

oscarmoutinho, 2021-02-16:
Same issue as Harsh Mangukiya, in MAC Big Sur, when I try with board: "AI Thinker ESP32-CAM".

I'm using "https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json" v1.0.5-rc7, to solve bug ...symbol missing... in Big Sur.

OK in Windows using standard "https://dl.espressif.com/dl/package_esp32_index.json" v 1.0.4.

It compiles OK if I open Arduino IDE and paste my sketch to the opened editor window.

If I open my sketch from the File menu, it opens a new editor window and, when compiling, I receive the error.

oscarmoutinho, 2021-02-16:
I think I found the issue.
This error throws when the directory path has spaces.
OK: [desktop]/**x_x**/mySketch/mySketch.ino
NOK: [desktop]/**x x**/mySketch/mySketch.ino

**It seems an IDE issue.**

closed 2021-07-08 by stale[bot]