espressif/arduino-esp32 #4710: setCpuFrequencyMhz(80) causes unexpected results and crashes (Exceptions)

alorbach, 2021-01-12:
I am on ESP-IDF release 3.3 with arduino-esp32 master branch build with https://github.com/espressif/esp32-arduino-lib-builder

So I discovered an issue when using setCpuFrequencyMhz to switch from 240hz to 80hz. 
In my code, it appears to cause instability like dropping WLAN when I do the following: 

- Switch to 80hz with setCpuFrequencyMhz(80);
- send UDP 512KB Paket after the Change

What happens is that WIFI dopped with "bcn_timout,ap_probe_send_start" but recovers due my retry code after a few seconds.

So I went to sdkconfig and enabled heap tracing as I have experienced issues like this before because stack or mem was corrupted at some point. 
```
CONFIG_HEAP_POISONING_COMPREHENSIVE=y
CONFIG_HEAP_TRACING=y
CONFIG_HEAP_TRACING_STACK_DEPTH=2
```

So before and after using **setCpuFrequencyMhz**() I called **heap_caps_check_integrity_all**(true); now, and it crashes with this exception: 
**Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)**

Stack trace: 
> 0x40097a5e: verify_fill_pattern at /esp-idf/components/heap/multi_heap_poisoning.c line 160
> 0x40097d35: multi_heap_internal_check_block_poisoning at /esp-idf/components/heap/multi_heap_poisoning.c line 358
> 0x400977c9: multi_heap_check at /esp-idf/components/heap/multi_heap.c line 657
> 0x401290ad: heap_caps_check_integrity at /esp-idf/components/heap/heap_caps.c line 416
> 0x401290d1: heap_caps_check_integrity_all at /esp-idf/components/heap/heap_caps.c line 425

So I have disabled using setCpuFrequencyMhz for now, but this appears to be a real bad issue.

me-no-dev, 2021-01-12:
`WiFi.setSleep(false);`?

alorbach, 2021-01-13:
@me-no-dev : nope thats not the problem, but thanks for the hint I have added this to my code.

me-no-dev, 2021-01-13:
What if you set the firmware to run at 80MHz from the beginning? Like using the Board menu in ArduinoIDE? Maybe the issue is only when the radio is already running?

alorbach, 2021-01-15:
Well my basic idea was to lower power consumption when the device is idle. My goal is to use a esp32 camera module running on usb powerbank as long as possible.

me-no-dev, 2021-01-15:
I would not recommend you to run the camera at 80MHz at all. On the ESP32 there is a fair amount of processing in ISR which will only suffer if your clock is that low. Also, the camera itself uses quite a bit of power.

lbernstone, 2021-01-15:
I think OP is implying that he will turn it up when the camera is running, and then try to get as low as possible to "hibernate".  I have not tested their exact config, but it is certainly possible to use that function as a throttle. I do it in almost every setup function I write.  Perhaps something in your system is getting starved at 80MHz, so it triggers the watchdog?  Try to make sure all your processes are idled (or even end them) before you switch.
Ultimately, wifi is power hungry and eats far more of your power than the cpu.

alorbach, 2021-01-19:
@lbernstone: I am open for alternative solutions to keep power usage low. 
Does it make sense to turned off Wifi and turned it on like every few seconds to check if somebody send data to it?

closed 2021-07-08 by stale[bot]