espressif/arduino-esp32 #4714: Platformio-build.py is missing in pre release packages

Jason2866, 2021-01-13:
It looks like the copy command for this file is missing in this part:
https://github.com/espressif/arduino-esp32/blob/08f4665775067c4778445709f382d2a649b61311/.github/scripts/on-release.sh#L179

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module? ?node32? ?ttgo_lora?
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git? ?1.0.2? ?1.0.3?
IDE name:						?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					?115200?
Computer OS:                                        ?Windows 10? ?Mac OSX? ?Ubuntu?

### Description:
(Pre) Release package does not work with PlatformIO because Platformio-build.py is missing

me-no-dev, 2021-01-13:
PIO does not use the packaged versions. They are only for ArduinoIDE

closed 2021-01-13 by me-no-dev

Jason2866, 2021-01-13:
Mhh, with adding this file PIO can use the packaged version.
I add just this missing file and it works as expected and supported from PlatformIO.
```
platform                    = espressif32 @ 2.1.0
platform_packages           = framework-arduinoespressif32 @ https://github.com/Jason2866/arduino-esp32/releases/download/1.0.5-rc6/esp32-1.0.5-rc6.zip
                              toolchain-xtensa32 @ https://github.com/Jason2866/platform-espressif32/releases/download/5.2.0.97/xtensa-esp32-elf-win32-1.22.0-97-gc752ad5-5.2.0.zip
                              platformio/tool-mklittlefs @ ~1.203.200522
```

me-no-dev, 2021-01-14:
I was not aware of this functionality. Something new? Will add the script so next version has it.

me-no-dev, 2021-01-14:
Done :) https://github.com/espressif/arduino-esp32/commit/d2530850a38af18ada5e805f0f4539e0048c43b3

Jason2866, 2021-01-14:
THX! To use the build chain packages only a valid package.json would be needed in the tarball ;-)
for example:
```
  "name": "toolchain-xtensa32",
  "version": "2.50200.97",
  "description": "GCC Toolchain for Xtensa32 processor",
  "system": [
    "windows",
    "windows_amd64",
    "windows_x86"
  ],
  "repository": {
    "type": "git",
    "url": "https://github.com/espressif/esp-idf"
  }
}
```

me-no-dev, 2021-01-14:
can you give me an exact package.json for the current version? need to support more than Windows

Jason2866, 2021-01-14:
This one are working https://github.com/Jason2866/platform-espressif32/releases/tag/5.2.0.97

MAC
```
{
  "name": "toolchain-xtensa32",
  "version": "2.50200.97",
  "description": "GCC Toolchain for Xtensa32 processor",
  "system": [ "darwin_x86_64" ],
  "repository": {
    "type": "git",
    "url": "https://github.com/espressif/esp-idf"
  }
}
```
Linux
```
  "name": "toolchain-xtensa32",
  "version": "2.50200.97",
  "description": "GCC Toolchain for Xtensa32 processor",
  "system": [
    "linux_x86_64"
  ],
  "repository": {
    "type": "git",
    "url": "https://github.com/espressif/esp-idf"
  }
}
```
More examples are here (Arduino ESP8266 crew added the package.json too)
https://github.com/earlephilhower/esp-quick-toolchain/releases/tag/3.0.0-newlib4.0.0-gnu21

me-no-dev, 2021-01-14:
where is the package.json that you mention? I do not see it

me-no-dev, 2021-01-14:
ahhh in the tarball... this you should take to https://github.com/espressif/crosstool-NG ;) toolchain comes from there and it only makes sense to be implemented there so it is included in future builds